saki4510t/UVCCamera #369: Open and Closing Camera

xwilus, 2018-06-04:
Hi,

Thank you once again for your awesome library. We used it initially normally with a SurfaceView but the specific camera the client was fitted upside down so we needed  to flip orientation. We did this by using your UVCCameraTextureView. This all worked awesomely until we updated the native libraries with a few fixes. We are using the IFrameCallBack to catch each frame and analysing it. This also works fine until we stop preview. The code for this is :

    public void stopPreview() {
        Log.i(TAG, "stopPreview: Entering");
        synchronized (mSync) {
            streamingStopped = true;
            Log.i(TAG, "stopPreview: Checking USB Monitor");
            if (mUSBMonitor != null) {
                Log.i(TAG, "stopPreview: Unregistering USB Monitor");
                mUSBMonitor.unregister();
            }
            Log.i(TAG, "stopPreview: Checking Texture View");
            if (mUVCCameraView != null) {
                Log.i(TAG, "stopPreview: ");
                mUVCCameraView.onPause();
            }

        }
    }

This then calls the UVCCameraTextureView's onDestroy in which we call the following code:

public boolean onSurfaceTextureDestroyed(SurfaceTexture surface) {

            Log.i(TAG, "Surface Texture Destroyed");
            synchronized (mSync) {
                streamingStopped = true;
                if (mUVCCamera != null) {
                    mUVCCamera.stopPreview();
                }
                try {
                    Thread.sleep(2000);
                } catch (InterruptedException e)
                {
                    Log.e(TAG, "onSurfaceTextureDestroyed: Interruption Exception", e);
                }
                isPreview = false;
            }
            mPreviewSurface = null;
            return true;
        }

Note: The Thread.sleep was brought in to try and circumvent our current issues.

This works fine randomly. Sometimes when the stoppreview has been called the IFrameCallback it seems is still processing and then we get an native error as below. You will see the IFrameCallback executes, then code in between executes in which time the stoppreview etc. And when everything is done the last step in IFramcallback and  then the native error occurs. I am a bit lost as how to get this resolved. Any advice please ? Logcat is below:

06-04 16:28:22.830 2868-5740/? I/UVC Camera Manager: mIFrameCallback - Frame Received 1 !!
06-04 16:28:22.830 2868-5740/? I/UVC Camera Manager: mIFrameCallback - Frame Received - Streaming Not Stopped!!
06-04 16:28:22.830 2868-5740/? I/UVC Camera Manager: mIFrameCallback - Copying pixels!!
06-04 16:28:22.832 2868-5740/? I/UVC Camera Manager: mIFrameCallback - Copying Bitmap!!
06-04 16:28:22.834 2868-5740/? I/UVC Camera Manager: mIFrameCallback - Sending Bitmap!!
06-04 16:28:22.835 2868-5740/? I/Multi Camera Fragment: frameReceived: Initialising Countdown
06-04 16:28:22.835 2868-5740/? I/Multi Camera Fragment: frameReceived: Flip Bitmap
06-04 16:28:22.836 686-771/? D/PowerManagerService: [api] acquire WakeLock flags=0x2000000a tag=WindowManager uid=1000 pid=686 ws=WorkSource{10228} pkg=android
06-04 16:28:22.854 2868-2868/? D/ViewRootImpl@d2de87a[EnrolmentActivity]: mHardwareRenderer.destroy()#1
06-04 16:28:22.855 686-2531/? V/WindowManager: Relayout Window{ff74fc3d0 u0 com.facetech.facetech_facial_sdk_example/com.facetech.facetech_android_sdk.EnrolmentActivity}: viewVisibility=8 req=1024x768 WM.LayoutParams{(0,0)(fillxfill) sim=#120 ty=1 fl=#81810180 pfl=0x20000 wanim=0x103046f vsysui=0x600 needsMenuKey=2 naviIconColor=0}
06-04 16:28:22.863 2868-2868/? D/ViewRootImpl@d2de87a[EnrolmentActivity]: Relayout returned: oldFrame=[0,0][1024,768] newFrame=[0,0][1024,768] result=0x5 surface={isValid=false 0} surfaceGenerationChanged=true
06-04 16:28:22.864 2868-2868/? D/ViewRootImpl@d2de87a[EnrolmentActivity]: MSG_WINDOW_FOCUS_CHANGED 0
06-04 16:28:22.876 296-296/? D/SurfaceFlinger:   Display[0] configurations (* current):
                                                   * 0: 768x1024, xdpi=131.804993, ydpi=132.028000, refresh=16666667
                                                 numHwLayers=6, flags=00000000
                                                   type   |  handle  | hint | flag | tr | blnd |   format    |     source crop (l,t,r,b)      |          frame         | name 
                                               -----------+----------+------+------+----+------+-------------+--------------------------------+------------------------+------
                                                      HWC | aa546d20 | 0002 | 0000 | 00 | 0100 | RGBA_8888   |    0.0,   66.0,  768.0,  217.0 |    0,    0,  768,  151 | com.facetech.facetech_facial_sdk_example/com.facetech.facetech_facial_sdk_example.MainActivity
                                                     GLES | 00000000 | 0000 | 0001 | 04 | 0105 | ? ffffffff  |    0.0,    0.0,   -1.0,   -1.0 |    0,    0,  768, 1024 | com.facetech.facetech_facial_sdk_example/com.facetech.facetech_android_sdk.EnrolmentActivity_dim
                                                     GLES | aa5472c0 | 0000 | 0000 | 00 | 0100 | RGBA_8888   |    0.0,    0.0,  768.0,  873.0 |    0,  151,  768, 1024 | com.facetech.facetech_facial_sdk_example/com.facetech.facetech_and
06-04 16:28:22.876 296-296/? D/SurfaceFlinger: roid_sdk.EnrolmentActivity
                                                     GLES | aa5465a0 | 0000 | 0000 | 00 | 0105 | RGBA_8888   |    0.0,    0.0,   24.0, 1024.0 |  744,    0,  768, 1024 | StatusBar
[... 466 lines cut ...]
06-04 16:28:23.646 5750-5750/? A/DEBUG: pid: 2868, tid: 5740, name: Thread-23  >>> com.facetech.facetech_facial_sdk_example <<<
06-04 16:28:23.646 5750-5750/? A/DEBUG: signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
06-04 16:28:23.654 5750-5750/? A/DEBUG: Abort message: 'art/runtime/java_vm_ext.cc:470] JNI DETECTED ERROR IN APPLICATION: obj == null'
06-04 16:28:23.655 5750-5750/? A/DEBUG:     r0 00000000  r1 0000166c  r2 00000006  r3 00000008
06-04 16:28:23.655 5750-5750/? A/DEBUG:     r4 7ff81978  r5 00000006  r6 7ff81920  r7 0000010c
06-04 16:28:23.655 5750-5750/? A/DEBUG:     r8 00000000  r9 0000000a  sl 000007c9  fp 8634e000
06-04 16:28:23.655 5750-5750/? A/DEBUG:     ip 0000000b  sp 7ff815f0  lr a8badfe7  pc a8bb0868  cpsr 200d0010
06-04 16:28:23.683 5750-5750/? A/DEBUG: backtrace:
06-04 16:28:23.683 5750-5750/? A/DEBUG:     #00 pc 0004a868  /system/lib/libc.so (tgkill+12)
06-04 16:28:23.683 5750-5750/? A/DEBUG:     #01 pc 00047fe3  /system/lib/libc.so (pthread_kill+34)
06-04 16:28:23.683 5750-5750/? A/DEBUG:     #02 pc 0001dbad  /system/lib/libc.so (raise+10)
06-04 16:28:23.683 5750-5750/? A/DEBUG:     #03 pc 00019321  /system/lib/libc.so (__libc_android_abort+34)
06-04 16:28:23.683 5750-5750/? A/DEBUG:     #04 pc 00017388  /system/lib/libc.so (abort+4)
06-04 16:28:23.683 5750-5750/? A/DEBUG:     #05 pc 0031b5d5  /system/lib/libart.so (_ZN3art7Runtime5AbortEPKc+328)
06-04 16:28:23.683 5750-5750/? A/DEBUG:     #06 pc 000b5385  /system/lib/libart.so (_ZN3art10LogMessageD2Ev+1132)
06-04 16:28:23.683 5750-5750/? A/DEBUG:     #07 pc 0023a5cd  /system/lib/libart.so (_ZN3art9JavaVMExt8JniAbortEPKcS2_+1664)
06-04 16:28:23.683 5750-5750/? A/DEBUG:     #08 pc 0023a833  /system/lib/libart.so (_ZN3art9JavaVMExt9JniAbortFEPKcS2_z+66)
06-04 16:28:23.683 5750-5750/? A/DEBUG:     #09 pc 0026adcd  /system/lib/libart.so (_ZN3art3JNI15CallVoidMethodVEP7_JNIEnvP8_jobjectP10_jmethodIDSt9__va_list+488)
06-04 16:28:23.683 5750-5750/? A/DEBUG:     #10 pc 0000c674  /data/app/com.facetech.facetech_facial_sdk_example-1/lib/arm/libUVCCamera.so (_ZN7_JNIEnv14CallVoidMethodEP8_jobjectP10_jmethodIDz+36)
06-04 16:28:23.683 5750-5750/? A/DEBUG:     #11 pc 0000dcf8  /data/app/com.facetech.facetech_facial_sdk_example-1/lib/arm/libUVCCamera.so (_ZN10UVCPreview19do_capture_callbackEP7_JNIEnvP9uvc_frame+156)
06-04 16:28:23.683 5750-5750/? A/DEBUG:     #12 pc 0000de38  /data/app/com.facetech.facetech_facial_sdk_example-1/lib/arm/libUVCCamera.so (_ZN10UVCPreview20do_capture_idle_loopEP7_JNIEnv+56)
06-04 16:28:23.684 5750-5750/? A/DEBUG:     #13 pc 0000df9c  /data/app/com.facetech.facetech_facial_sdk_example-1/lib/arm/libUVCCamera.so (_ZN10UVCPreview10do_captureEP7_JNIEnv+104)
06-04 16:28:23.684 5750-5750/? A/DEBUG:     #14 pc 0000dfdc  /data/app/com.facetech.facetech_facial_sdk_example-1/lib/arm/libUVCCamera.so (_ZN10UVCPreview19capture_thread_funcEPv+52)
06-04 16:28:23.684 5750-5750/? A/DEBUG:     #15 pc 00047ab3  /system/lib/libc.so (_ZL15__pthread_startPv+22)
06-04 16:28:23.684 5750-5750/? A/DEBUG:     #16 pc 00019f55  /system/lib/libc.so (__start_thread+6)

xwilus, 2018-06-05:
I have moved the actual stopPreview of UVC Camera after the framecallback has finished. That seems to be working now. Now I had it hang after number 319 but I am testing again. It hanged on the moment when I called stopreview in IFrameCallback.

xwilus, 2018-06-05:
Scrap that. I am still getting an issue when testing on the actual client device

xwilus, 2018-06-07:
more Logging:

06-07 15:03:22.373 296-296/? D/SurfaceFlinger:   Display[0] configurations (* current):
                                                   * 0: 768x1024, xdpi=131.804993, ydpi=132.028000, refresh=16666667
                                                 numHwLayers=4, flags=00000000
                                                   type   |  handle  | hint | flag | tr | blnd |   format    |     source crop (l,t,r,b)      |          frame         | name 
                                               -----------+----------+------+------+----+------+-------------+--------------------------------+------------------------+------
                                                      HWC | a6d178a0 | 0002 | 0000 | 00 | 0100 | RGBA_8888   |    0.0,    0.0,  768.0, 1024.0 |    0,    0,  768, 1024 | com.facetech.facetech_facial_sdk_example/com.facetech.facetech_facial_sdk_example.MainActivity
                                                      HWC | aa822b60 | 0002 | 0000 | 00 | 0105 | RGBA_8888   |    0.0,    0.0,   24.0, 1024.0 |  744,    0,  768, 1024 | StatusBar
                                                      HWC | a6d17de0 | 0002 | 0000 | 04 | 0105 | RGBA_8888   |    0.0,    0.0,  232.0,   44.0 |   64,  396,  108,  628 | Toast
                                                FB TARGET | aa822560 | 0000 | 0000 | 00 | 0105 | RGBA_8888   |    0.0,    0.0,  768.0, 1024.0 |    0,    0,  768, 1024 | HWC_FRAMEBUFFER_TARGET
06-07 15:03:22.450 21956-24368/? A/art: art/runtime/java_vm_ext.cc:470]   native: #05 pc 0000c674  /data/app/com.facetech.facetech_facial_sdk_example-2/lib/arm/libUVCCamera.so (_ZN7_JNIEnv14CallVoidMethodEP8_jobjectP10_jmethodIDz+36)
06-07 15:03:22.450 21956-24368/? A/art: art/runtime/java_vm_ext.cc:470]   native: #06 pc 0000dcf8  /data/app/com.facetech.facetech_facial_sdk_example-2/lib/arm/libUVCCamera.so (_ZN10UVCPreview19do_capture_callbackEP7_JNIEnvP9uvc_frame+156)
06-07 15:03:22.450 21956-24368/? A/art: art/runtime/java_vm_ext.cc:470]   native: #07 pc 0000de38  /data/app/com.facetech.facetech_facial_sdk_example-2/lib/arm/libUVCCamera.so (_ZN10UVCPreview20do_capture_idle_loopEP7_JNIEnv+56)
06-07 15:03:22.450 21956-24368/? A/art: art/runtime/java_vm_ext.cc:470]   native: #08 pc 0000df9c  /data/app/com.facetech.facetech_facial_sdk_example-2/lib/arm/libUVCCamera.so (_ZN10UVCPreview10do_captureEP7_JNIEnv+104)
06-07 15:03:22.450 21956-24368/? A/art: art/runtime/java_vm_ext.cc:470]   native: #09 pc 0000dfdc  /data/app/com.facetech.facetech_facial_sdk_example-2/lib/arm/libUVCCamera.so (_ZN10UVCPreview19capture_thread_funcEPv+52)
06-07 15:03:22.660 21956-24368/? A/art: art/runtime/runtime.cc:437]   native: #09 pc 0000c674  /data/app/com.facetech.facetech_facial_sdk_example-2/lib/arm/libUVCCamera.so (_ZN7_JNIEnv14CallVoidMethodEP8_jobjectP10_jmethodIDz+36)
06-07 15:03:22.660 21956-24368/? A/art: art/runtime/runtime.cc:437]   native: #10 pc 0000dcf8  /data/app/com.facetech.facetech_facial_sdk_example-2/lib/arm/libUVCCamera.so (_ZN10UVCPreview19do_capture_callbackEP7_JNIEnvP9uvc_frame+156)
06-07 15:03:22.660 21956-24368/? A/art: art/runtime/runtime.cc:437]   native: #11 pc 0000de38  /data/app/com.facetech.facetech_facial_sdk_example-2/lib/arm/libUVCCamera.so (_ZN10UVCPreview20do_capture_idle_loopEP7_JNIEnv+56)
06-07 15:03:22.660 21956-24368/? A/art: art/runtime/runtime.cc:437]   native: #12 pc 0000df9c  /data/app/com.facetech.facetech_facial_sdk_example-2/lib/arm/libUVCCamera.so (_ZN10UVCPreview10do_captureEP7_JNIEnv+104)
06-07 15:03:22.660 21956-24368/? A/art: art/runtime/runtime.cc:437]   native: #13 pc 0000dfdc  /data/app/com.facetech.facetech_facial_sdk_example-2/lib/arm/libUVCCamera.so (_ZN10UVCPreview19capture_thread_funcEPv+52)
06-07 15:03:22.661 21956-24368/? A/art: art/runtime/runtime.cc:437]   native: #11 pc 0000c674  /data/app/com.facetech.facetech_facial_sdk_example-2/lib/arm/libUVCCamera.so (_ZN7_JNIEnv14CallVoidMethodEP8_jobjectP10_jmethodIDz+36)
06-07 15:03:22.661 21956-24368/? A/art: art/runtime/runtime.cc:437]   native: #12 pc 0000dcf8  /data/app/com.facetech.facetech_facial_sdk_example-2/lib/arm/libUVCCamera.so (_ZN10UVCPreview19do_capture_callbackEP7_JNIEnvP9uvc_frame+156)
06-07 15:03:22.661 21956-24368/? A/art: art/runtime/runtime.cc:437]   native: #13 pc 0000de38  /data/app/com.facetech.facetech_facial_sdk_example-2/lib/arm/libUVCCamera.so (_ZN10UVCPreview20do_capture_idle_loopEP7_JNIEnv+56)
06-07 15:03:22.661 21956-24368/? A/art: art/runtime/runtime.cc:437]   native: #14 pc 0000df9c  /data/app/com.facetech.facetech_facial_sdk_example-2/lib/arm/libUVCCamera.so (_ZN10UVCPreview10do_captureEP7_JNIEnv+104)
06-07 15:03:22.661 21956-24368/? A/art: art/runtime/runtime.cc:437]   native: #15 pc 0000dfdc  /data/app/com.facetech.facetech_facial_sdk_example-2/lib/arm/libUVCCamera.so (_ZN10UVCPreview19capture_thread_funcEPv+52)
06-07 15:03:22.661 21956-24368/? A/art: art/runtime/runtime.cc:437]   native: #02 pc 0000e3c8  /data/app/com.facetech.facetech_facial_sdk_example-2/lib/arm/libUVCCamera.so (_ZN10UVCPreview11stopPreviewEv+56)
06-07 15:03:22.661 21956-24368/? A/art: art/runtime/runtime.cc:437]   native: #03 pc 000072d8  /data/app/com.facetech.facetech_facial_sdk_example-2/lib/arm/libUVCCamera.so (_ZN9UVCCamera11stopPreviewEv+16)
06-07 15:03:22.661 21956-24368/? A/art: art/runtime/runtime.cc:437]   native: #04 pc 0002cf75  /data/app/com.facetech.facetech_facial_sdk_example-2/oat/arm/base.odex (Java_com_serenegiant_usb_UVCCamera_nativeStopPreview__J+88)
[... cut ...]
                                                      HWC | a6d17f00 | 0002 | 0000 | 00 | 0105 | RGBA_8888   |    0.0,    0.0,  437.0, 1024.0 |    0,    0,  437, 1024 | InputMethod
                                                      HWC | aa822f80 | 0002 | 0000 | 00 | 0105 | RGBA_8888   |    0.0,    0.0,   24.0, 1024.0 |  744,    0,  768, 1024 | StatusBar
                                                FB TARGET | aa822560 | 0000 | 0000 | 00 | 0105 | RGBA_8888   |    0.0,    0.0,  768.0, 1024.0 |    0,    0,  768, 1024 | HWC_FRAMEBUFFER_TAR

xwilus, 2018-06-08:
@saki4510t Please help if you do not mind. I have implemented the fix https://github.com/saki4510t/UVCCamera/issues/259 but are still experiencing the same issue. I am not sure what else to look at.

xwilus, 2018-06-13:
@saki4510t 

Fuller Error log:

06-13 10:22:10.916 25521-28261/? A/libc: Fatal signal 6 (SIGABRT), code -6 in tid 28261 (Thread-23)
                                         
                                         [ 06-13 10:22:10.918   244:  244 W/         ]
                                         debuggerd: handling request: pid=25521 uid=10232 gid=10232 tid=28261
06-13 10:22:11.017 28266-28266/? A/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
06-13 10:22:11.017 28266-28266/? A/DEBUG: Build fingerprint: 'samsung/gt5note10ltexx/gt5note10lte:7.1.1/NMF26X/P555XXU1CQJ3:user/release-keys'
06-13 10:22:11.017 28266-28266/? A/DEBUG: Revision: '7'
06-13 10:22:11.017 28266-28266/? A/DEBUG: ABI: 'arm'
06-13 10:22:11.017 28266-28266/? A/DEBUG: pid: 25521, tid: 28261, name: Thread-23  >>> com.facetech.facetech_facial_sdk_example <<<
06-13 10:22:11.017 28266-28266/? A/DEBUG: signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
06-13 10:22:11.022 28266-28266/? A/DEBUG: Abort message: 'art/runtime/java_vm_ext.cc:470] JNI DETECTED ERROR IN APPLICATION: obj == null'
06-13 10:22:11.022 28266-28266/? A/DEBUG:     r0 00000000  r1 00006e65  r2 00000006  r3 00000008
06-13 10:22:11.022 28266-28266/? A/DEBUG:     r4 848ff978  r5 00000006  r6 848ff920  r7 0000010c
06-13 10:22:11.022 28266-28266/? A/DEBUG:     r8 00000000  r9 0000000a  sl 000007ca  fp 881ce000
06-13 10:22:11.022 28266-28266/? A/DEBUG:     ip 0000000b  sp 848ff5f0  lr ac2a6fe7  pc ac2a9868  cpsr 200d0010
06-13 10:22:11.050 28266-28266/? A/DEBUG: backtrace:
06-13 10:22:11.050 28266-28266/? A/DEBUG:     #00 pc 0004a868  /system/lib/libc.so (tgkill+12)
06-13 10:22:11.050 28266-28266/? A/DEBUG:     #01 pc 00047fe3  /system/lib/libc.so (pthread_kill+34)
06-13 10:22:11.051 28266-28266/? A/DEBUG:     #02 pc 0001dbad  /system/lib/libc.so (raise+10)
06-13 10:22:11.051 28266-28266/? A/DEBUG:     #03 pc 00019321  /system/lib/libc.so (__libc_android_abort+34)
06-13 10:22:11.051 28266-28266/? A/DEBUG:     #04 pc 00017388  /system/lib/libc.so (abort+4)
06-13 10:22:11.051 28266-28266/? A/DEBUG:     #05 pc 0031b5d5  /system/lib/libart.so (_ZN3art7Runtime5AbortEPKc+328)
06-13 10:22:11.051 28266-28266/? A/DEBUG:     #06 pc 000b5385  /system/lib/libart.so (_ZN3art10LogMessageD2Ev+1132)
06-13 10:22:11.051 28266-28266/? A/DEBUG:     #07 pc 0023a5cd  /system/lib/libart.so (_ZN3art9JavaVMExt8JniAbortEPKcS2_+1664)
06-13 10:22:11.051 28266-28266/? A/DEBUG:     #08 pc 0023a833  /system/lib/libart.so (_ZN3art9JavaVMExt9JniAbortFEPKcS2_z+66)
06-13 10:22:11.051 28266-28266/? A/DEBUG:     #09 pc 0026adcd  /system/lib/libart.so (_ZN3art3JNI15CallVoidMethodVEP7_JNIEnvP8_jobjectP10_jmethodIDSt9__va_list+488)
06-13 10:22:11.051 28266-28266/? A/DEBUG:     #10 pc 0000d040  /data/app/com.facetech.facetech_facial_sdk_example-1/lib/arm/libUVCCamera.so (_ZN7_JNIEnv14CallVoidMethodEP8_jobjectP10_jmethodIDz+52)
06-13 10:22:11.051 28266-28266/? A/DEBUG:     #11 pc 0000f100  /data/app/com.facetech.facetech_facial_sdk_example-1/lib/arm/libUVCCamera.so (_ZN10UVCPreview19do_capture_callbackEP7_JNIEnvP9uvc_frame+160)
06-13 10:22:11.051 28266-28266/? A/DEBUG:     #12 pc 0000f238  /data/app/com.facetech.facetech_facial_sdk_example-1/lib/arm/libUVCCamera.so (_ZN10UVCPreview20do_capture_idle_loopEP7_JNIEnv+56)
06-13 10:22:11.051 28266-28266/? A/DEBUG:     #13 pc 0000f398  /data/app/com.facetech.facetech_facial_sdk_example-1/lib/arm/libUVCCamera.so (_ZN10UVCPreview10do_captureEP7_JNIEnv+104)
06-13 10:22:11.051 28266-28266/? A/DEBUG:     #14 pc 0000f3e8  /data/app/com.facetech.facetech_facial_sdk_example-1/lib/arm/libUVCCamera.so (_ZN10UVCPreview19capture_thread_funcEPv+68)
06-13 10:22:11.052 28266-28266/? A/DEBUG:     #15 pc 00047ab3  /system/lib/libc.so (_ZL15__pthread_startPv+22)
06-13 10:22:11.052 28266-28266/? A/DEBUG:     #16 pc 00019f55  /system/lib/libc.so (__start_thread+6)

xwilus, 2018-09-07:
Thank you. Will try it

hjhcs121, 2019-09-03:
> Thank you. Will try it

Do you solve the issue?

jjjjbababy, 2020-05-05:
> Thank you. Will try it

Do you solve the issue?