saki4510t/UVCCamera #374: startPreview error

xiayouli0122, 2018-06-14:
All Log：
06-14 09:33:37.682 18835-18835/? I/art: Late-enabling -Xcheck:jni
06-14 09:33:37.707 18835-18845/? I/art: Debugger is no longer active
06-14 09:33:37.915 18835-18835/? I/InstantRun: starting instant run server: is main process
06-14 09:33:38.065 18835-18856/? D/OpenGLRenderer: Use EGL_SWAP_BEHAVIOR_PRESERVED: true
06-14 09:33:38.071 18835-18835/? D/Atlas: Validating map...
06-14 09:33:38.106 18835-18856/? I/OpenGLRenderer: Initialized EGL, version 1.4
06-14 09:33:38.107 18835-18856/? D/OpenGLRenderer: Enabling debug mode 0
06-14 09:33:39.067 18835-18855/com.serenegiant.usbcameratest D/MainActivity: onAttach
06-14 09:33:39.126 18835-18856/com.serenegiant.usbcameratest V/RenderScript: 0xab4b4000 Launching thread(s), CPUs 4
06-14 09:33:52.698 18835-18855/com.serenegiant.usbcameratest D/UsbDeviceConnectionJNI: close
06-14 09:33:52.698 18835-18855/com.serenegiant.usbcameratest I/USBMonitor: name=/dev/bus/usb/002/004,desc=64,busnum=2,devnum=4,rawDesc=[B@128b186c
06-14 09:33:52.698 18835-18855/com.serenegiant.usbcameratest D/MainActivity: onConnect
06-14 09:33:52.732 18835-18854/com.serenegiant.usbcameratest W/linker: libjpeg-turbo1500.so: unused DT entry: type 0x6ffffffe arg 0x37d0
    libjpeg-turbo1500.so: unused DT entry: type 0x6fffffff arg 0x1
06-14 09:33:52.738 18835-18854/com.serenegiant.usbcameratest W/linker: libusb100.so: unused DT entry: type 0x6ffffffe arg 0x2f48
    libusb100.so: unused DT entry: type 0x6fffffff arg 0x1
06-14 09:33:52.742 18835-18854/com.serenegiant.usbcameratest W/linker: libuvc.so: unused DT entry: type 0x6ffffffe arg 0x3c9c
    libuvc.so: unused DT entry: type 0x6fffffff arg 0x1
06-14 09:33:52.746 18835-18854/com.serenegiant.usbcameratest W/linker: libUVCCamera.so: unused DT entry: type 0x6ffffffe arg 0x5abc
    libUVCCamera.so: unused DT entry: type 0x6fffffff arg 0x1
06-14 09:33:52.759 18835-18854/com.serenegiant.usbcameratest D/UsbDeviceConnectionJNI: close
06-14 09:33:52.759 18835-18854/com.serenegiant.usbcameratest I/libusb/core: [18854*core.c:2085:libusb_init2]:libusb v1.0.19.10903
06-14 09:33:52.759 18835-18854/com.serenegiant.usbcameratest I/libusb/usbfs: [18854*android_usbfs.c:504:op_init2]:start up hotplug event handler
    [18854*android_usbfs.c:512:op_init2]:call android_scan_devices
06-14 09:33:52.786 18835-18854/com.serenegiant.usbcameratest D/Yuri: onConnect.startPreview
06-14 09:33:52.807 18835-18898/com.serenegiant.usbcameratest I/libUVCCamera: [18898*UVCPreview.cpp:492:prepare_preview]:frameSize=(640,480)@MJPEG
06-14 09:34:07.319 18835-18897/com.serenegiant.usbcameratest A/libc: Fatal signal 11 (SIGSEGV), code 2, fault addr 0xdfd90000 in tid 18897 (BaseActivity)
06-14 09:34:07.323 2789-2789/? I/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
    Build fingerprint: 'Android/p201/p201:5.1.1/LMY47V/20171122:userdebug/release-keys'
    Revision: '0'
    ABI: 'arm'
06-14 09:34:07.324 2789-2789/? I/DEBUG: pid: 18835, tid: 18897, name: BaseActivity  >>> com.serenegiant.usbcameratest <<<
    signal 11 (SIGSEGV), code 2 (SEGV_ACCERR), fault addr 0xdfd90000
06-14 09:34:07.341 2789-2789/? I/DEBUG:     r0 dfd8ffe0  r1 e001fe48  r2 00000024  r3 3fd901e8
06-14 09:34:07.343 2789-2789/? I/DEBUG:     r4 000002a4  r5 00000018  r6 e001fc08  r7 000002a4
06-14 09:34:07.344 2789-2789/? I/DEBUG:     r8 ab4e4bd0  r9 ab4da6c0  sl e001fc09  fp e0263ac0
06-14 09:34:07.345 2789-2789/? I/DEBUG:     ip f6f7e648  sp e0263a78  lr e029edd8  pc f6f39c7e  cpsr 200f0030
06-14 09:34:07.347 2789-2789/? I/DEBUG: backtrace:
06-14 09:34:07.350 2789-2789/? I/DEBUG:     #00 pc 00012c7e  /system/lib/libc.so (__memcpy_base+121)
06-14 09:34:07.351 2789-2789/? I/DEBUG:     #01 pc 00014dd4  /data/app/com.serenegiant.usbcameratest-2/lib/arm/libuvc.so
06-14 09:34:07.352 2789-2789/? I/DEBUG:     #02 pc 0000a434  /data/app/com.serenegiant.usbcameratest-2/lib/arm/libusb100.so (usbi_handle_transfer_completion+320)
06-14 09:34:07.354 2789-2789/? I/DEBUG:     #03 pc 000104e0  /data/app/com.serenegiant.usbcameratest-2/lib/arm/libusb100.so
        #04 pc 0000b314  /data/app/com.serenegiant.usbcameratest-2/lib/arm/libusb100.so
06-14 09:34:07.355 2789-2789/? I/DEBUG:     #05 pc 0000ac7c  /data/app/com.serenegiant.usbcameratest-2/lib/arm/libusb100.so (libusb_handle_events_timeout_completed+964)
        #06 pc 0000b4f4  /data/app/com.serenegiant.usbcameratest-2/lib/arm/libusb100.so (libusb_handle_events+52)
        #07 pc 00012cd0  /data/app/com.serenegiant.usbcameratest-2/lib/arm/libuvc.so (_uvc_handle_events+136)
        #08 pc 00016933  /system/lib/libc.so (__pthread_start(void*)+30)
        #09 pc 0001485f  /system/lib/libc.so (__start_thread+6)
06-14 09:34:07.727 2789-2789/? I/DEBUG: Tombstone written to: /data/tombstones/tombstone_03
06-14 09:34:07.727 3104-3124/system_process I/BootReceiver: Copying /data/tombstones/tombstone_03 to DropBox (SYSTEM_TOMBSTONE)
06-14 09:34:07.727 3104-18932/system_process W/ActivityManager:   Force finishing activity 1 com.serenegiant.usbcameratest/.MainActivity

CPU Info：
Processor       : AArch64 Processor rev 4 (aarch64)
processor       : 0
Features        : fp asimd crc32 wp half thumb fastmult vfp edsp neon vfpv3 tlsi vfpv4 idiva idivt
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : Amlogic

Android OS Version: 5.1.1

Need Help!!!

xiayouli0122, 2018-06-14:
Need Help!!!

lixby, 2018-10-11:
Have you solved this problem? I  need help too!

xiayouli0122, 2018-10-15:
SORRY,NO

monxarat, 2021-11-04:
Have you solved this problem?