saki4510t/UVCCamera #384: Gradle stuck downloading dependencies

Sean-Kone, 2018-07-16:
Hello,

I've integrated libuvccamera in my project and everything works fine.

I have libuvccamera as a separate module in my project and I followed instructions from [here](https://github.com/saki4510t/UVCCamera/wiki/When-you-import-libuvccamera-module-in-your-project-instead-of-building-these-samples-from-this-repository) .
(I use commonLibVersion 1.5.20).

Everything worked great until I tried to update play services version.
It works fine with version 12.0.0 and 'com.google.gms:google-services:3.0.0', however,
when I update to latest version: 'com.google.gms:google-services:4.0.1'
and version 15.0.1 for some of the play services libs the gradle sync gets stuck on downloading the dependencies.
keeps trying to download deps, show this:

Gradle:Download http://raw.github.com/saki4510t/libcommon/master/repository/com/google/android/gms/some_play_services_lib/

Gradle:List http://raw.github.com/saki4510t/libcommon/master/repository/com/google/android/gms/some_play_services_lib/

I tried many times and one time let it run for more than an hour.
if I disable the libuvccamera module (unload module) gradle syncs quickly with the updated libs.

Any one encounter this issue??

Thank you

closed 2018-07-25 by Sean-Kone

LordParsley, 2018-07-25:
@Sean-Kone did you manage to resolve this issue?

Sean-Kone, 2018-07-29:
One of the times I tried to sync, it eventually worked and downloaded all the dependencies but it took a really long time.
I have a flavor that doesn't use the uvc lib. When building for this flavor I removed the uvc lib maven rep from the projcet gradle and unloaded the uvc module and it synced up all the DEPS really quickly.