saki4510t/UVCCamera #385: WebRTC Support

gilangabdulaziz, 2018-07-17:
**Hi mate,**

I got this error when creating a capturer in WebRTC:
`Camera name /dev/bus/usb/001/006 does not match any known camera device.`

I'm using Camera1Enumerator from WebRTC
```
Camera1Enumerator enumerator = new Camera1Enumerator();
mUsbCapurer = enumerator.createCapturer(devicesName, null);
```

@saki4510t is it possible to stream video to WebRTC using UVCCamera library?

Regards,
L

hanyc0914, 2018-08-01:
Hello，
Have you solve this problem？I 'm doing Android WebRTC application and encounter the same question with you.

gilangabdulaziz, 2018-08-02:
@hanyc0914 Hi,
I solved it by creating custom enumerator.. It's really tricky 😿

UTSAnonymous, 2019-01-18:
Hi @gilangabdulaziz ,

I'm currently working on something similar as well. How did you create a custom enumerator?

MelvinFMQ, 2020-02-15:
@gilangabdulaziz I would like to know this aswell!

acrolink, 2020-03-10:
@gilangabdulaziz 

Would be great if you could share the solution.

pumatus, 2020-07-20:
> @ hanyc0914嗨，
> 我通过创建自定义枚举器解决了它。😿

Hello~ How did you create this enumeration? (About transferring from usb camera to webrtc)

acrolink, 2020-07-20:
Would be great if @gilangabdulaziz could share that information. What I have done in my situation is to consume the frame arrays provided by USBCamera library (not enumerating available cameras directly).

mayankawaliya14, 2025-03-07:
I am also having issue in sending tracks to webRtc or streaming platform so could you help me with it?