saki4510t/UVCCamera #391: Low FPS at larger sizes

maym86, 2018-07-27:
At 1080p the frame rate drops to 5fps.  This seems like a bug given the camera can run at 30fps on other platforms.

Any suggestions on where to look to fix this?

maym86, 2018-07-27:
Solved this:

5Hz was when I was using motion jpeg. I set the frame format to YUYV  and its working at 30HZ hope this helps other people.

EavenHan, 2018-08-03:
Dear Maym86:
   
   I have tried your solution. It's useful to increase the FPS. But the screen is flikering. 
Do you know the problem reason?

Thanks.

UbuntuLover, 2018-08-04:
@maym86 @EavenHan  I got this issue same. First of all, you need to confirm the specifics of your USB camera. Such as the output ability. Then, you need to make sure the bandwidth of your USB port. According to the USB2.0 standard,480mb/s is not enough when you using the yuv420sp color format.

GoldenStrawberry, 2018-08-09:
I changed the code in` UVCPreview. cpp` to convert MJPEG directly to RGBX. I'm calling the function `uvc_mjpeg2rgbx` in the` frame_mjpepe.c` library,But sometimes the conversion fails, and the code goes wrong below:
`
dinfo.err = jpeg_std_error(&jerr.super); 
jerr.super.error_exit = _error_exit; 
if (setjmp(jerr.jmp)) { 
      goto fail; 
}
`
The phenomenon of flashing appears, which seriously affects the quality of the picture. How to solve this problem? @maym86 @EavenHan @UbuntuLover

maym86, 2018-08-09:
I'm unable to help with that. I haven't experienced it. Closing this issue as the original problem is solved. Please create a new issue for flickering.

closed 2018-08-09 by maym86

zipswich, 2022-03-14:
@GoldenStrawberry Have you solved the problem? I am encountering it now.