saki4510t/UVCCamera #396: Have detected camera without displaying cameraview

HowICanBeWithYou, 2018-08-06:
I have two UVC cameras and use them as stereo camera. I have make it work successfully on Xiaomi ,MI MAX(a Chinese brand,Android 6.0.1). Recent days ,I want to change my device ,which is the Xiaomi, MIX 2S(same brand but different type ,Android 8.0.0).But this time ,I face the problem that the phone have detected the camera but can not display the cameraview .The two device use the same code and the same program. Beside the device ,the connection is different ,MI MAX use mico usb while the MIX 2S use type-c to connect.

HowICanBeWithYou, 2018-08-06:
[image]
This is what i get from the ndk stack,From the information ,it may be the question about opencv,but Istill have no idea about it.
Any solution or relative thoughts will be nice.Thanks !

closed 2018-08-08 by HowICanBeWithYou

kirilldev1010, 2019-04-04:
I am faced same issue. How did you solve it? @HowICanBeWithYou