saki4510t/UVCCamera #398: Modify the code to increase the frame rate to 30

GoldenStrawberry, 2018-08-17:
I changed the code inUVCPreview. cpp to convert MJPEG directly to RGBX. I'm calling the function uvc_mjpeg2rgbx in theframe_mjpepe.c library,But sometimes the conversion fails, and the code goes wrong below:
`dinfo.err = jpeg_std_error(&jerr.super); jerr.super.error_exit = _error_exit; if (setjmp(jerr.jmp)) { goto fail; }`
The phenomenon of flashing appears, which seriously affects the quality of the picture. How to solve this problem?

hnlbxb2004, 2018-10-29:
[https://github.com/saki4510t/UVCCamera/issues/417](url)

I don't know if you solved this problem? I have the same doubts as you.

zipswich, 2022-03-14:
Has anyone solved this issue?

danji, 2023-12-04:
Does this issue solved? Encounter the same issue, too.

danji, 2023-12-07:
Found that uvc_mjpeg2rgbx used wrong out_step while jpeg_read_scanlines. Modify code like below can solve the flashing issue.

```
jpeg_start_decompress(&dinfo);

	// local copy, assign data/out_step after out set
	uint8_t *data = out->data;
	const int out_step = out->step;
     ... 
    if (LIKELY(dinfo.output_height == out->height)) {
```

zipswich, 2023-12-07:
@danji Thank you for sharing the solution. I am not clear what you meant. The[ current code](https://github.com/saki4510t/UVCCamera/blob/c9399e63dfab4b6d260d8cbef92182abc6de0ee0/libuvccamera/src/main/jni/libuvc/src/frame-mjpeg.c#L286) is the following:
	uint8_t *data = out->data;
	const int out_step = out->step;

	if (LIKELY(dinfo.output_height == out->height)) {
		for (; dinfo.output_scanline < dinfo.output_height ;) {
			buffer[0] = data + (lines_read) * out_step;
			for (i = 1; i < MAX_READLINE; i++)
				buffer[i] = buffer[i-1] + out_step;
			num_scanlines = jpeg_read_scanlines(&dinfo, buffer, MAX_READLINE);
			lines_read += num_scanlines;
		}
		out->actual_bytes = in->width * in->height * 3;	// XXX
	}

Did you mean to move the first 2 lines to after the if block?