saki4510t/UVCCamera #399: UsbDevice detached after being opened

rgparask, 2018-08-20:
I have been trying to make use an external USB camera on a Motorola E4  (Android 7.1.1) and I get some weird behavior. On every test project the USB device is detected and attached. As soon as I try to open it, right after giving permission to use it, the device is detached. With the code as is I get a NullPointerException because the connection is null. If I check whether the connection is null to avoid throwing the exception, the device is detached and I get the following error:

```
E/UsbManager: exception in UsbManager.openDevice
              java.lang.IllegalArgumentException: device /dev/bus/usb/001/008 does not exist or is restricted
                  at android.os.Parcel.readException(Parcel.java:1688)
                  at android.os.Parcel.readException(Parcel.java:1637)
                  at android.hardware.usb.IUsbManager$Stub$Proxy.openDevice(IUsbManager.java:443)
                  at android.hardware.usb.UsbManager.openDevice(UsbManager.java:353)
                  at com.serenegiant.usb.USBMonitor$UsbControlBlock.<init>(USBMonitor.java:1011)
                  at com.serenegiant.usb.USBMonitor$UsbControlBlock.clone(USBMonitor.java:1034)
                  at com.serenegiant.usb.UVCCamera.open(UVCCamera.java:192)
                  at com.serenegiant.usbcameracommon.AbstractUVCCameraHandler$CameraThread.handleOpen(AbstractUVCCameraHandler.java:444)
                  at com.serenegiant.usbcameracommon.AbstractUVCCameraHandler.handleMessage(AbstractUVCCameraHandler.java:300)
                  at com.serenegiant.usbcameracommon.UVCCameraHandler.handleMessage(UVCCameraHandler.java:31)
                  at android.os.Handler.dispatchMessage(Handler.java:102)
                  at android.os.Looper.loop(Looper.java:154)
                  at com.serenegiant.usbcameracommon.AbstractUVCCameraHandler$CameraThread.run(AbstractUVCCameraHandler.java:747)

```

Each time I try to connect the camera the device name (/dev/bus/usb/001/xxx) increments by 1, indicating that the device disconnects and reconnects as a different device and also a "Device Detached" toast message is shown.

I have tried downgrading to an older version of NDK, as suggested in other issues, without any luck. I also had to use `NDK_TOOLCHAIN_VERSION := 4.9` in Application.mk to make the samples work on other devices such as a Google Pixel (running correctly on Android 8.0, 8.1 and 9.0) and a Nexus 5 running Android 7.1.1.

Lastly, other USB camera application on Google Play that use this repository work correctly on the E4.

WangZhaosong, 2018-11-05:
I met the same question as you. Did you solve this question yet?

rgparask, 2018-11-05:
> I met the same question as you. Did you solve this question yet?

Unfortunately not, I have been using other devices for development and prioritized other issues. If I find a solution I will post it and close the thread.

Ambika-Gowda, 2020-01-21:
I have the same error. Did you guys solve it ??

Ambika-Gowda, 2020-01-22:
@saki4510t  Could you please provide solution or atleast tell what could be tried to solve the above error.

rgparask, 2020-02-11:
This eventually worked for me:

1. Request permission for any device, as soon as it is connected:
```
@Override
public void onAttach(UsbDevice device) {
    requestPermission(device);
}
```
2. When permission is granted try opening the device and catch a potential exception. If the device opens successfully keep it on a list of available cameras:
```
@Override
public void onConnect(UsbDevice device, USBMonitor.UsbControlBlock ctrlBlock, boolean createNew) {
    synchronized (mCameras) {
        try {
            int devKey = ctrlBlock.getDeviceKey();
            UVCCamera uvcCamera = new UVCCamera();
            uvcCamera.open(ctrlBlock);
            uvcCamera.setPreviewTexture(new SurfaceTexture(0));
            mCameras.put(devKey, uvcCamera);
        } catch (Exception e) {
            Log.e(TAG, "onConnect: ", e);
        }
    }
}
```
3. Modify UsbMonitor.updateDeviceInfo() to prevent NullPointerException when the device is detached while opening and the connection returned is null:
```
public static UsbDeviceInfo updateDeviceInfo(final UsbManager manager, final UsbDevice device, final UsbDeviceInfo _info) {
	final UsbDeviceInfo info = _info != null ? _info : new UsbDeviceInfo();
	info.clear();
	if (device != null) {
		if (BuildCheck.isLollipop()) {
			info.manufacturer = device.getManufacturerName();
			info.product = device.getProductName();
			info.serial = device.getSerialNumber();
		}
		if (BuildCheck.isMarshmallow()) {
			info.usb_version = device.getVersion();
		}
		if ((manager != null) && manager.hasPermission(device)) {
			final UsbDeviceConnection connection = manager.openDevice(device);
			if (connection == null){ // <---
				return null;     // <---
			}                        // <---
			final byte[] desc = connection.getRawDescriptors();
			if (TextUtils.isEmpty(info.usb_version)) {
				info.usb_version = String.format("%x.%02x", ((int)desc[3] & 0xff), ((int)desc[2] & 0xff));
			}
			if (TextUtils.isEmpty(info.version)) {
				info.version = String.format("%x.%02x", ((int)desc[13] & 0xff), ((int)desc[12] & 0xff));
			}
			if (TextUtils.isEmpty(info.serial)) {
				info.serial = connection.getSerial();
			}
			final byte[] languages = new byte[256];
			int languageCount = 0;
			// controlTransfer(int requestType, int request, int value, int index, byte[] buffer, int length, int timeout)
			try {
				int result = connection.controlTransfer(
					USB_REQ_STANDARD_DEVICE_GET, // USB_DIR_IN | USB_TYPE_STANDARD | USB_RECIP_DEVICE
    				USB_REQ_GET_DESCRIPTOR,
    				(USB_DT_STRING << 8) | 0, 0, languages, 256, 0);
				if (result > 0) {
        			languageCount = (result - 2) / 2;
				}
				if (languageCount > 0) {
					if (TextUtils.isEmpty(info.manufacturer)) {
						info.manufacturer = getString(connection, desc[14], languageCount, languages);
					}
					if (TextUtils.isEmpty(info.product)) {
						info.product = getString(connection, desc[15], languageCount, languages);
					}
					if (TextUtils.isEmpty(info.serial)) {
						info.serial = getString(connection, desc[16], languageCount, languages);
					}
				}
			} finally {
				connection.close();
			}
		}
		if (TextUtils.isEmpty(info.manufacturer)) {
			info.manufacturer = USBVendorId.vendorName(device.getVendorId());
		}
		if (TextUtils.isEmpty(info.manufacturer)) {
			info.manufacturer = String.format("%04x", device.getVendorId());
		}
		if (TextUtils.isEmpty(info.product)) {
			info.product = String.format("%04x", device.getProductId());
		}
	}
	return info;
}
```
4. When the device is needed get it from the list and call startPreview()

The device fails to connect a couple of times, but is eventually opened and shows the preview

closed 2020-02-28 by rgparask

Ambika-Gowda, 2020-02-28:
Ty @rgparask  will try it