saki4510t/UVCCamera #400: java.lang.UnsatisfiedLinkError: dalvik.system.PathClassLoader when insert rtmp video player

mrsiro153, 2018-08-21:
I have tried to apply live streaming to this project by some rtmp libraries. But whenever I insert rtmp library, I can not open camera view with error: 
java.lang.UnsatisfiedLinkError: dalvik.system.PathClassLoader[DexPathList[[zip file "/data/app/com.serenegiant.usbcameratest8-2/base.apk", zip file "/data/app/com.serenegiant.usbcameratest8-2/split_lib_dependencies_apk.apk", zip file "/data/app/com.serenegiant.usbcameratest8-2/split_lib_slice_0_apk.apk", zip file "/data/app/com.serenegiant.usbcameratest8-2/split_lib_slice_1_apk.apk", zip file "/data/app/com.serenegiant.usbcameratest8-2/split_lib_slice_2_apk.apk", zip file "/data/app/com.serenegiant.usbcameratest8-2/split_lib_slice_3_apk.apk", zip file "/data/app/com.serenegiant.usbcameratest8-2/split_lib_slice_4_apk.apk", zip file "/data/app/com.serenegiant.usbcameratest8-2/split_lib_slice_5_apk.apk", zip file "/data/app/com.serenegiant.usbcameratest8-2/split_lib_slice_6_apk.apk", zip file "/data/app/com.serenegiant.usbcameratest8-2/split_lib_slice_7_apk.apk", zip file "/data/app/com.serenegiant.usbcameratest8-2/split_lib_slice_8_apk.apk", zip file "/data/app/com.serenegiant.usbcameratest8-2/split_lib_slice_9_apk.apk"],nativeLibraryDirectories=[/data/app/com.serenegiant.usbcameratest8-2/lib/arm64, /system/fake-libs64, /data/app/com.serenegiant.usbcameratest8-2/base.apk!/lib/arm64-v8a, /data/app/com.serenegiant.usbcameratest8-2/split_lib_dependencies_apk.apk!/lib/arm64-v8a, /data/app/com.serenegiant.usbcameratest8-2/split_lib_slice_0_apk.apk!/lib/arm64-v8a, /data/app/com.serenegiant.usbcameratest8-2/split_lib_slice_1_apk.apk!/lib/arm64-v8a, /data/app/com.serenegiant.usbcameratest8-2/split_lib_slice_2_apk.apk!/lib/arm64-v8a, /data/app/com.serenegiant.usbcameratest8-2/split_lib_slice_3_apk.apk!/lib/arm64-v8a, /data/app/com.serenegiant.usbcameratest8-2/split_lib_slice_4_apk.apk!/lib/arm64-v8a, /data/app/com.serenegiant.usbcameratest8-2/split_lib_slice_5_apk.apk!/lib/arm64-v8a, /data/app/com.serenegiant.usbcameratest8-2/split_lib_slice_6_apk.apk!/lib/arm64-v8a, /data/app/com.serenegiant.usbcameratest8-2/split_lib_slice_7_apk.apk!/lib/arm64-v8a, /data/app/com.serenegiant.usbcameratest8-2/split_lib_slice_8_apk.apk!/lib/arm64-v8a, /data/app/com.serenegiant.usbcameratest8-2/split_lib_slice_9_apk.apk!/lib/arm64-v8a, /system/lib64, /vendor/lib64]]] couldn't find "libjpeg-turbo1500.so"
        at java.lang.Runtime.loadLibrary0(Runtime.java:972)
        at java.lang.System.loadLibrary(System.java:1567)
        at com.serenegiant.usb.UVCCamera.<clinit>(UVCCamera.java:121)
        at com.serenegiant.usbcameracommon.AbstractUVCCameraHandler$CameraThread.handleOpen(AbstractUVCCameraHandler.java:474)
        at com.serenegiant.usbcameracommon.AbstractUVCCameraHandler.handleMessage(AbstractUVCCameraHandler.java:328)
        at com.serenegiant.usbcameracommon.UVCCameraHandler.handleMessage(UVCCameraHandler.java:31)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:154)
        at com.serenegiant.usbcameracommon.AbstractUVCCameraHandler$CameraThread.run(AbstractUVCCameraHandler.java:780)
----
This error occur all project ( usbCamera 0 to 8), and the library for rtmp player I use:
compile 'net.butterflytv.utils:rtmp-client:3.0.1'
or compile 'com.google.android.exoplayer:extension-rtmp:2.8.0' and many others
I have just inserted these libraries --> this error happen 
Someone please help me with this problem ?

janoonaj, 2019-01-23:
Same here, did you find any solution?

SagarKhurana, 2019-07-31:
I am also trying to integrate TensorFlow with UVCCamera module, I am getting the same error. 

1. Tried to add the release AAR files of libuvccamera and usbCameraCommonto the TensorFlow module... still facing the same issue. 

2. Tried to add the TensorFlow dependency in the UVCCamera module, still facing the same issue.

I have seen the owner of the Repo has already created the project for OenCVWithUVCCamera. So it is all the way possible to integrate TensorFlow as well, but if possible guide me if I am missing anything. 
Or if anyone can refer me any document to integrate 2 NDK modules in 1 project.

oras, 2019-10-15:
This gave me the solution (Setting the architecture dependencies exclusively)

https://corbt.com/posts/2015/09/18/mixing-32-and-64bit-dependencies-in-android.html

Ivan421399, 2020-12-08:
[project_root]/libuvccamera/src/main/jni/Application.mk
APP_ABI := armeabi-v7a x86 arm64-v8a