saki4510t/UVCCamera #413: You need to use a Theme.AppCompat theme (or descendant) with this activity.

751643992, 2018-10-09:
java.lang.RuntimeException: Unable to start activity ComponentInfo{com.serenegiant.usbcameratest/com.serenegiant.usbcameratest.MainActivity}: java.lang.IllegalStateException: You need to use a Theme.AppCompat theme (or descendant) with this activity.
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2741)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2802)
        at android.app.ActivityThread.-wrap12(ActivityThread.java)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1549)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:163)
        at android.app.ActivityThread.main(ActivityThread.java:6377)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:904)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:794)
     Caused by: java.lang.IllegalStateException: You need to use a Theme.AppCompat theme (or descendant) with this activity.
        at android.support.v7.app.AppCompatDelegateImplV9.createSubDecor(AppCompatDelegateImplV9.java:354)
        at android.support.v7.app.AppCompatDelegateImplV9.ensureSubDecor(AppCompatDelegateImplV9.java:323)
        at android.support.v7.app.AppCompatDelegateImplV9.setContentView(AppCompatDelegateImplV9.java:284)
        at android.support.v7.app.AppCompatActivity.setContentView(AppCompatActivity.java:139)
        at com.serenegiant.usbcameratest.MainActivity.onCreate(MainActivity.java:61)
        at android.app.Activity.performCreate(Activity.java:6865)
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1119)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2694)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2802) 
        at android.app.ActivityThread.-wrap12(ActivityThread.java) 
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1549) 
        at android.os.Handler.dispatchMessage(Handler.java:102) 
        at android.os.Looper.loop(Looper.java:163) 
        at android.app.ActivityThread.main(ActivityThread.java:6377) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:904) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:794)

BernardJiang, 2018-10-11:
For example, change the line in UVCCamera/usbCameraTest0/src/main/AndroidManifest.xml:
from
        android:theme="@style/AppTheme" >
to
        android:theme="@style/Theme.AppCompat" >
It solves this problem.

closed 2018-10-11 by 751643992

surajsahani, 2020-08-11:
how do i avoid black screen of the app? By default my activity is opening in black!!

malikmuneeb981, 2022-02-13:
android:theme="@style/Theme.Design.Light"
CHANGE TO THIS IT WILL SOLVE THE BLACK SCREEN ISSUE

giwallet, 2022-07-05:
> @style/Theme.AppCompat

but its not working  and also give this error