saki4510t/UVCCamera #414: java.lang.UnsupportedOperationException: open failed:result=-1

751643992, 2018-10-09:
Process: com.serenegiant.usbcameratest, PID: 28464
    java.lang.UnsupportedOperationException: open failed:result=-1
        at com.serenegiant.usb.UVCCamera.open(UVCCamera.java:204)
        at com.serenegiant.usbcameratest.MainActivity$2$1.run(MainActivity.java:142)
        at android.os.Handler.handleCallback(Handler.java:754)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:163)
        at android.os.HandlerThread.run(HandlerThread.java:61)

mengnick, 2018-10-10:
10-10 13:50:24.978 12802 12902 V USBMonitor: UsbEndpoint[mAddress=134,mAttributes=5,mMaxPacketSize=100,mInterval=4]]]]
10-10 13:50:24.980 12802 12902 I USBMonitor: UsbControlBlock:constructor
10-10 13:50:24.988 12802 12902 D UsbDeviceConnectionJNI: close
10-10 13:50:24.988 12802 12902 I USBMonitor: name=/dev/bus/usb/001/002,desc=90,busnum=1,devnum=2,rawDesc=[B@f0fb0b8
10-10 13:50:24.989 12802 12902 V MainActivity: onConnect:
10-10 13:50:24.989 12802 12902 V MainActivity: startPreview:
10-10 13:50:24.989 12802 12908 V CameraThread: handleOpen:
10-10 13:50:24.989 12802 12908 V CameraThread: handleClose:
10-10 13:50:24.989 12802 12908 V CameraThread: handleStopRecording:mMuxer=null
10-10 13:50:24.989 12802 12908 V UVCCameraTextureView: setVideoEncoder:
10-10 13:50:24.989 12802 12922 V UVCCameraTextureView: RenderThread#setEncoder:encoder=null
10-10 13:50:24.989 12802 12802 V UVCCameraTextureView: getPreviewTexture:
10-10 13:50:24.989 12802 12922 I UVCCameraTextureView: RenderThread#updatePreviewSurface:
10-10 13:50:24.989 12802 12922 V UVCCameraTextureView: updatePreviewSurface:tex_id=1
10-10 13:50:24.991 12802 12802 V MainActivity: startImageProcessor:
10-10 13:50:25.009 12802 12983 D EGLBase14: EGLContext created, client version 3
10-10 13:50:25.013 12802 12908 D UsbDeviceConnectionJNI: close
10-10 13:50:25.013 12802 12908 I libusb/core: [12908*core.c:2085:libusb_init2]:libusb v1.0.19.10903
10-10 13:50:25.013 12802 12908 I libusb/usbfs: [12908*android_usbfs.c:504:op_init2]:start up hotplug event handler
10-10 13:50:25.014 12802 12908 I libusb/usbfs: [12908*android_usbfs.c:512:op_init2]:call android_scan_devices
10-10 13:50:25.014 12802 12908 E UVCCamera: [12908*UVCCamera.cpp:172:connect]:could not open camera:err=-1
10-10 13:50:25.014 12802 12908 W AbsUVCCameraHandler: supportedSize:null
10-10 13:50:25.014 12802 12908 V CameraThread: handleStartPreview:

751643992, 2018-10-11:
libuvccamera
Application.mk
NDK_TOOLCHAIN_VERSION := 4.9

closed 2018-10-11 by 751643992

xk990610, 2018-10-30:
I have same issue,if resloved,please contact me,thanks