saki4510t/UVCCamera #417: Why do you need to convert first and then draw?

hnlbxb2004, 2018-10-23:
https://github.com/saki4510t/UVCCamera/blob/c9399e63dfab4b6d260d8cbef92182abc6de0ee0/libuvccamera/src/main/jni/UVCCamera/UVCPreview.cpp#L533

Why do you need to convert first（   MJPEG => yuyv） and then draw_preview_one?

hnlbxb2004, 2018-10-23:
I tested this conversion on some phones and it was very time consuming, especially at 1080P, which made the perceived frame rate very low. I tested not converting, drawing directly, similar to the YUV method below, but I found that sometimes frames are skipped. It is currently uncertain what is the reason.

yeminnhtut, 2019-04-04:
I had the same thought as yours. So, I removed the line and tested. Didn't work out. I got "conversion failed" error logs.

zyowo, 2020-06-18:
someing new？

along1991, 2021-04-14:
How did you solve it?