saki4510t/UVCCamera #419: uvc_stream_start_bandwidth()  fails

skeletoncrewrobotics, 2018-10-26:
after wrestling with other issues (ndk version, abi list, camera can't do 640x480, styles/themes) I have the apps starting. But no pictures. mostly trying usbCameraTest0 for now.
for some reason, MPEG encoding fails, and it falls back to YUV. 
That seems a bit odd. 
But it gets to startPreview() and then fails doing a transfer. 
any ideas?

10-26 13:25:15.917 22269-22282/com.serenegiant.usbcameratest0 I/MainActivity: supportedSize:{"formats":[{"index":1,"type":4,"default":1,"size":["320x240","160x120"]}]}
...
10-26 13:25:15.927 22269-22282/com.serenegiant.usbcameratest0 D/ USBC UVCCamera: really setting preview size (3 PARAMS) 320 x 240   AND FORMAT 0
...
10-26 13:25:15.927 22269-22282/com.serenegiant.usbcameratest0 I/libUVCCamera: [22282*UVCPreview.cpp:171:setPreviewSize]:UVCPreview::setPreviewSize( width 320, height 240, minfps 1 maxfps 30, mode 0, bandwidth 1.000000
...
10-26 13:25:15.967 22269-22282/com.serenegiant.usbcameratest0 D/ USBC UVCCamera: startPreview()
10-26 13:25:15.967 22269-22282/com.serenegiant.usbcameratest0 I/USBC UVCCamera: [22282*UVCCamera.cpp:281:startPreview]:UVCCamera::startPreview()
10-26 13:25:15.967 22269-22282/com.serenegiant.usbcameratest0 D/USBC UVCCamera: [22282*UVCCamera.cpp:283:startPreview]:begin
10-26 13:25:15.967 22269-22295/com.serenegiant.usbcameratest0 I/libUVCCamera: [22295*UVCPreview.cpp:464:preview_thread_func]:UVCPreview::preview_thread_func()
    [22295*UVCPreview.cpp:484:prepare_preview]:UVCPreview::prepare_preview()  requestMode is 0
10-26 13:25:15.967 22269-22295/com.serenegiant.usbcameratest0 I/libuvc/stream: [22295*stream.c:547:uvc_get_stream_ctrl_format_size_fps]:uvc_get_stream_ctrl_format_size_fps( ... format 3, width 320, height 240, minfps 1, maxfps 30)
    [22295*stream.c:559:uvc_get_stream_ctrl_format_size_fps]:checking a format
    [22295*stream.c:387:_prepare_stream_ctrl]:_prepare_stream_ctrl()
    [22295*stream.c:171:uvc_query_stream_ctrl]:uvc_query_stream_ctrl()
    [22295*stream.c:217:uvc_query_stream_ctrl]:doing the transfer
10-26 13:25:15.967 22269-22295/com.serenegiant.usbcameratest0 I/libUVCCamera: [22295*io.c:1437:libusb_submit_transfer]:libusb_submit_transfer()
    [22295*io.c:1450:libusb_submit_transfer]:timeout 0
    [22295*io.c:1458:libusb_submit_transfer]:submit backend transfer
    [22295*io.c:1477:libusb_submit_transfer]:libusb_submit_transfer()  returning r 0
10-26 13:25:15.967 22269-22295/com.serenegiant.usbcameratest0 I/libuvc/stream: [22295*stream.c:171:uvc_query_stream_ctrl]:uvc_query_stream_ctrl()
    [22295*stream.c:217:uvc_query_stream_ctrl]:doing the transfer
10-26 13:25:15.967 22269-22295/com.serenegiant.usbcameratest0 I/libUVCCamera: [22295*io.c:1437:libusb_submit_transfer]:libusb_submit_transfer()
    [22295*io.c:1450:libusb_submit_transfer]:timeout 0
    [22295*io.c:1458:libusb_submit_transfer]:submit backend transfer
    [22295*io.c:1477:libusb_submit_transfer]:libusb_submit_transfer()  returning r 0
10-26 13:25:15.967 22269-22295/com.serenegiant.usbcameratest0 I/libuvc/stream: [22295*stream.c:171:uvc_query_stream_ctrl]:uvc_query_stream_ctrl()
    [22295*stream.c:217:uvc_query_stream_ctrl]:doing the transfer
10-26 13:25:15.977 22269-22295/com.serenegiant.usbcameratest0 I/libUVCCamera: [22295*io.c:1437:libusb_submit_transfer]:libusb_submit_transfer()
    [22295*io.c:1450:libusb_submit_transfer]:timeout 0
    [22295*io.c:1458:libusb_submit_transfer]:submit backend transfer
    [22295*io.c:1477:libusb_submit_transfer]:libusb_submit_transfer()  returning r 0
10-26 13:25:15.977 22269-22295/com.serenegiant.usbcameratest0 I/libuvc/stream: [22295*stream.c:387:_prepare_stream_ctrl]:_prepare_stream_ctrl()
    [22295*stream.c:171:uvc_query_stream_ctrl]:uvc_query_stream_ctrl()
    [22295*stream.c:217:uvc_query_stream_ctrl]:doing the transfer
10-26 13:25:15.977 22269-22295/com.serenegiant.usbcameratest0 I/libUVCCamera: [22295*io.c:1437:libusb_submit_transfer]:libusb_submit_transfer()
    [22295*io.c:1450:libusb_submit_transfer]:timeout 0
    [22295*io.c:1458:libusb_submit_transfer]:submit backend transfer
    [22295*io.c:1477:libusb_submit_transfer]:libusb_submit_transfer()  returning r 0
10-26 13:25:15.977 22269-22295/com.serenegiant.usbcameratest0 I/libuvc/stream: [22295*stream.c:171:uvc_query_stream_ctrl]:uvc_query_stream_ctrl()
    [22295*stream.c:217:uvc_query_stream_ctrl]:doing the transfer
10-26 13:25:15.977 22269-22295/com.serenegiant.usbcameratest0 I/libUVCCamera: [22295*io.c:1437:libusb_submit_transfer]:libusb_submit_transfer()
    [22295*io.c:1450:libusb_submit_transfer]:timeout 0
    [22295*io.c:1458:libusb_submit_transfer]:submit backend transfer
    [22295*io.c:1477:libusb_submit_transfer]:libusb_submit_transfer()  returning r 0
10-26 13:25:15.977 22269-22295/com.serenegiant.usbcameratest0 I/libuvc/stream: [22295*stream.c:171:uvc_query_stream_ctrl]:uvc_query_stream_ctrl()
10-26 13:25:15.997 22269-22295/com.serenegiant.usbcameratest0 I/libuvc/stream: [22295*stream.c:217:uvc_query_stream_ctrl]:doing the transfer
10-26 13:25:15.997 22269-22295/com.serenegiant.usbcameratest0 I/libUVCCamera: [22295*io.c:1437:libusb_submit_transfer]:libusb_submit_transfer()
    [22295*io.c:1450:libusb_submit_transfer]:timeout 0
    [22295*io.c:1458:libusb_submit_transfer]:submit backend transfer
    [22295*io.c:1477:libusb_submit_transfer]:libusb_submit_transfer()  returning r 0
10-26 13:25:15.997 22269-22295/com.serenegiant.usbcameratest0 I/libuvc/stream: [22295*stream.c:573:uvc_get_stream_ctrl_format_size_fps]:FOUND a matching frame descriptor. NOW WE PROBE
    [22295*stream.c:591:uvc_probe_stream_ctrl]:uvc_probe_stream_ctrl()
    [22295*stream.c:171:uvc_query_stream_ctrl]:uvc_query_stream_ctrl()
    [22295*stream.c:217:uvc_query_stream_ctrl]:doing the transfer
10-26 13:25:15.997 22269-22295/com.serenegiant.usbcameratest0 I/libUVCCamera: [22295*io.c:1437:libusb_submit_transfer]:libusb_submit_transfer()
    [22295*io.c:1450:libusb_submit_transfer]:timeout 0
    [22295*io.c:1458:libusb_submit_transfer]:submit backend transfer
    [22295*io.c:1477:libusb_submit_transfer]:libusb_submit_transfer()  returning r 0
10-26 13:25:15.997 22269-22295/com.serenegiant.usbcameratest0 I/libuvc/stream: [22295*stream.c:171:uvc_query_stream_ctrl]:uvc_query_stream_ctrl()
    [22295*stream.c:217:uvc_query_stream_ctrl]:doing the transfer
10-26 13:25:15.997 22269-22295/com.serenegiant.usbcameratest0 I/libUVCCamera: [22295*io.c:1437:libusb_submit_transfer]:libusb_submit_transfer()
    [22295*io.c:1450:libusb_submit_transfer]:timeout 0
    [22295*io.c:1458:libusb_submit_transfer]:submit backend transfer
[... 23 lines cut ...]
    [22295*io.c:1450:libusb_submit_transfer]:timeout 0
    [22295*io.c:1458:libusb_submit_transfer]:submit backend transfer
    [22295*io.c:1462:libusb_submit_transfer]: list_del  arm_timerfd
    [22295*io.c:1477:libusb_submit_transfer]:libusb_submit_transfer()  returning r -1

(above is the first time that the backend transfer failed)

10-26 13:25:16.027 22269-22295/com.serenegiant.usbcameratest0 I/libuvc/stream: [22295*stream.c:1620:uvc_stream_start_bandwidth]:libusb_submit_transfer failed
10-26 13:25:16.027 22269-22295/com.serenegiant.usbcameratest0 E/libuvc/stream: [22295*stream.c:1635:uvc_stream_start_bandwidth]:fail
10-26 13:25:16.027 22269-22295/com.serenegiant.usbcameratest0 W/libUVCCamera: [22295*diag.c:88:uvc_perror]:failed start_streaming:I/O error (-1)
10-26 13:25:16.027 22269-22295/com.serenegiant.usbcameratest0 I/libUVCCamera: [22295*UVCPreview.cpp:476:preview_thread_func]:UVCPreview::preview_thread_func() DONE
10-26 13:26:16.507 22269-22269/com.serenegiant.usbcameratest0 V/MainActivity: onStop:
10-26 13:27:41.347 22269-22269/com.serenegiant.usbcameratest0 V/MainActivity: onStart:
10-26 13:27:41.397 22269-22269/com.serenegiant.usbcameratest0 I/Timeline: Timeline: Activity_idle id: android.os.BinderProxy@4188e738 time:88421929
10-26 13:27:42.377 22269-22283/com.serenegiant.usbcameratest0 V/MainActivity: onAttach:
10-26 13:27:44.297 22269-22269/com.serenegiant.usbcameratest0 V/MainActivity: surfaceDestroyed:
10-26 13:27:44.297 22269-22269/com.serenegiant.usbcameratest0 D/ USBC UVCCamera: stopPreview()
10-26 13:27:44.297 22269-22269/com.serenegiant.usbcameratest0 D/USBC UVCCamera: [22269*UVCCamera.cpp:272:setFrameCallback]:begin
    [22269*UVCCamera.cpp:277:setFrameCallback]:end (0)
    [22269*UVCCamera.cpp:293:stopPreview]:begin
10-26 13:27:44.297 22269-22269/com.serenegiant.usbcameratest0 W/libUVCCamera: [22269*UVCPreview.cpp:366:stopPreview]:UVCPreview::terminate capture thread: pthread_join failed
10-26 13:27:44.297 22269-22269/com.serenegiant.usbcameratest0 D/USBC UVCCamera: [22269*UVCCamera.cpp:297:stopPreview]:end (0)
10-26 13:27:44.607 22269-22269/com.serenegiant.usbcameratest0 V/MainActivity: onStop:

10-26 13:25:15.967 22269-22282/com.serenegiant.usbcameratest0 D/ USBC UVCCamera: startPreview()

wolegeyun, 2019-03-27:
I have same problem，have answer？

GlassUp, 2019-03-28:
I have the same problem as well. Any news?

skeletoncrewrobotics, 2019-03-29:
I never got it to work.

MelvinFMQ, 2019-04-05:
I used different cameras, lower bandwidth and 2 cameras worked. Anyone knows how to reduce bandwidth requirements?