saki4510t/UVCCamera #426: Preview no image

manfeng, 2018-11-15:
11-15 16:30:33.123 20745-20779/com.serenegiant.usbcameratest2 I/USBMonitor: name=/dev/bus/usb/001/003,desc=68,busnum=1,devnum=3,rawDesc=[B@a78a9aa
11-15 16:30:33.138 20745-20775/com.serenegiant.usbcameratest2 D/UsbDeviceConnectionJNI: close
11-15 16:30:33.139 20745-20775/com.serenegiant.usbcameratest2 I/libusb/core: [20775*core.c:2085:libusb_init2]:libusb v1.0.19.10903
11-15 16:30:33.139 20745-20775/com.serenegiant.usbcameratest2 I/libusb/usbfs: [20775*android_usbfs.c:504:op_init2]:start up hotplug event handler
    [20775*android_usbfs.c:512:op_init2]:call android_scan_devices
11-15 16:30:33.140 20745-20775/com.serenegiant.usbcameratest2 I/MainActivity: supportedSize:{"formats":[{"index":1,"type":4,"default":1,"size":["640x480","640x480"]}]}
11-15 16:30:33.993 20745-2420/com.serenegiant.usbcameratest2 I/libUVCCamera: [2420*UVCPreview.cpp:492:prepare_preview]:frameSize=(640,480)@YUYV
11-15 16:30:34.004 20745-2388/com.serenegiant.usbcameratest2 E/libuvc/stream: [2388*stream.c:878:_uvc_process_payload_iso]:bad packet:status=0xcc
11-15 16:30:34.006 20745-2388/com.serenegiant.usbcameratest2 I/chatty: uid=10457(com.serenegiant.usbcameratest2) BaseActivity identical 2 lines
11-15 16:30:34.008 20745-2388/com.serenegiant.usbcameratest2 E/libuvc/stream: [2388*stream.c:878:_uvc_process_payload_iso]:bad packet:status=0xcc
11-15 16:30:34.009 20745-2388/com.serenegiant.usbcameratest2 E/libuvc/stream: [2388*stream.c:878:_uvc_process_payload_iso]:bad packet:status=0xcc
11-15 16:30:34.010 20745-2388/com.serenegiant.usbcameratest2 E/libuvc/stream: [2388*stream.c:878:_uvc_process_payload_iso]:bad packet:status=0xcc
11-15 16:30:34.012 20745-2388/com.serenegiant.usbcameratest2 E/libuvc/stream: [2388*stream.c:878:_uvc_process_payload_iso]:bad packet:status=0xcc
11-15 16:30:34.013 20745-2388/com.serenegiant.usbcameratest2 E/libuvc/stream: [2388*stream.c:878:_uvc_process_payload_iso]:bad packet:status=0xcc
11-15 16:30:34.014 20745-2388/com.serenegiant.usbcameratest2 E/libuvc/stream: [2388*stream.c:878:_uvc_process_payload_iso]:bad packet:status=0xcc
11-15 16:30:34.016 20745-2388/com.serenegiant.usbcameratest2 I/chatty: uid=10457(com.serenegiant.usbcameratest2) BaseActivity identical 1 line
11-15 16:30:34.017 20745-2388/com.serenegiant.usbcameratest2 E/libuvc/stream: [2388*stream.c:878:_uvc_process_payload_iso]:bad packet:status=0xcc
11-15 16:30:34.167 20745-2388/com.serenegiant.usbcameratest2 E/libuvc/stream: [2388*stream.c:878:_uvc_process_payload_iso]:bad packet:status=0xcc

rgm-22, 2020-09-13:
Hi manfeng!

Did you solver your issue? I think that I have a similiar problem :(

manfeng, 2020-09-17:
> Hi manfeng!
> 
> Did you solver your issue? I think that I have a similiar problem :(

I solved this problem. I commented a line of code in the _uvc_process_payload_iso function in the stream. file to preview the image. I forgot the specific line of code here. I have a packaged .SO library file. If necessary, leave an email and I will send it to you.

rgm-22, 2020-09-19:
Thank you very much for your answer manfeng!

I see. Every help is welcome! :)

My email:
100303777@alumnos.uc3m.es

manfeng, 2020-09-21:
> Thank you very much for your answer manfeng!
> 
> I see. Every help is welcome! :)
> 
> My email:
> [100303777@alumnos.uc3m.es](mailto:100303777@alumnos.uc3m.es)
> 
> My log is:
> 
> V/MainActivity: onDialogResult:canceled=false
> D/OpenGLRenderer: endAllActiveAnimators on 0x7b10b27600 (RippleDrawable) with handle 0x7b669c7d80
> D/UsbDeviceConnectionJNI: close
> I/USBMonitor: name=/dev/bus/usb/001/003,desc=77,busnum=1,devnum=3,rawDesc=[B@ab85579
> V/MainActivity: onConnect:
> V/UVCCameraTextureView: getPreviewTexture:
> V/CameraThread: handleOpen:
> handleClose:
> handleStopRecording:mMuxer=null
> V/UVCCameraTextureView: setVideoEncoder:
> I/UVCCameraTextureView: RenderThread#updatePreviewSurface:
> V/UVCCameraTextureView: updatePreviewSurface:tex_id=1
> RenderThread#setEncoder:encoder=null
> D/UsbDeviceConnectionJNI: close
> I/libusb/core: [3445_core.c:2085:libusb_init2]:libusb v1.0.19.10903 I/libusb/usbfs: [3445_android_usbfs.c:504:op_init2]:start up hotplug event handler
> [3445*android_usbfs.c:512:op_init2]:call android_scan_devices
> I/AbsUVCCameraHandler: supportedSize:{"formats":[{"index":1,"type":4,"default":1,"size":["160x120"]},{"index":2,"type":4,"default":1,"size":["160x120","160x122"]},{"index":3,"type":4,"default":1,"size":["160x120"]},{"index":4,"type":4,"default":1,"size":["160x120"]},{"index":5,"type":4,"default":1,"size":["160x120"]}]}
> V/CameraThread: handleStartPreview:
> 
> And nothing happens :(

.so libs has been sent to your email, hope it helps you.