saki4510t/UVCCamera #427: Can not open camera

ivan-ushakov, 2018-11-15:
I'm trying to use library on TV with Android 4.4 and with master branch I have this error

    D/UsbDeviceConnectionJNI( 1227): close
    I/USBMonitor( 1227): name=/dev/bus/usb/003/003,desc=58,busnum=3,devnum=3,rawDesc=[B@41f7bff0
    V/CameraThread( 1227): handleOpen:
    V/CameraThread( 1227): handleClose:
    V/UVCCameraTextureView( 1227): setVideoEncoder:
    V/UVCCameraTextureView( 1227): RenderThread#setEncoder:encoder=null
    D/UVCCamera( 1227): [1241*UVCCamera.cpp:69:UVCCamera]:begin
    D/UVCCamera( 1227): [1241*UVCCamera.cpp:71:UVCCamera]:end
    D/UsbDeviceConnectionJNI( 1227): close
    D/UVCCamera( 1227): [1241*UVCCamera.cpp:138:connect]:begin
    D/libusb/core( 1227): [1241*core.c:2040:libusb_init2]:begin
    I/libusb/core( 1227): [1241*core.c:2085:libusb_init2]:libusb v1.0.19.10903
    D/libusb/core( 1227): [1241*core.c:2108:libusb_init2]:call usbi_backend->init2
    I/libusb/usbfs( 1227): [1241*android_usbfs.c:504:op_init2]:start up hotplug event handler
    I/libusb/usbfs( 1227): [1241*android_usbfs.c:512:op_init2]:call android_scan_devices
    D/libusb/core( 1227): [1241*core.c:2138:libusb_init2]:end (0)
    D/libusb/core( 1227): [1241*core.c:1252:libusb_get_device_with_fd]:begin
    D/libusb/core( 1227): [1241*core.c:1262:libusb_get_device_with_fd]:end
    D/libusb  ( 1227): [libusb_open] open (bus/addr)=(3.3)
    D/libusb  ( 1227): [op_open] getcap not available
    D/libusb  ( 1227): [usbi_add_pollfd] add fd 66 events 4
    D/libusb  ( 1227): [libusb_get_config_descriptor] index 0
    E/libusb  ( 1227): [parse_interface] invalid extra intf desc len (0)
    E/libusb  ( 1227): [raw_desc_to_config] parse_configuration failed with error -1
    D/libusb/core( 1227): [1241*core.c:1995:libusb_set_auto_detach_kernel_driver]:begin
    D/libusb/core( 1227): [1241*core.c:1997:libusb_set_auto_detach_kernel_driver]:enable=0
    D/libusb/core( 1227): [1241*core.c:2004:libusb_set_auto_detach_kernel_driver]:end (0)
    D/libusb  ( 1227): [libusb_close]
    D/libusb  ( 1227): [usbi_remove_pollfd] remove fd 66
    E/UVCCamera( 1227): [1241*UVCCamera.cpp:172:connect]:could not open camera:err=-1
    D/UVCCamera( 1227): [1241*UVCCamera.cpp:187:connect]:end (-1)

I see that problem happens while parsing interface but I don't know how to debug this.

FROGbots-4634, 2018-11-23:
I'm having the same error with a Logitech C920 and Android 7.1.2

FROGbots-4634, 2018-11-23:
Update: I was able to fix it by uncommenting a line in Android.mk as described here: https://github.com/saki4510t/UVCCamera/issues/181