saki4510t/UVCCamera #436: Not working properly on Android 9.0

juechenshiai, 2019-01-15:
1. Some usb camera‘s device class is not 239，maybe 0, 14 and so on.
2. UsbManager.getDeviceList() can not get the UsbDevice of some usb camera.
3. Some usb camera can not be connected after permission granted.
These three occur in Android 9.0，lower than it is OK

juechenshiai, 2019-01-30:
> I found I can not obtain the same class information in Android 9.0 and 8.0 when I used the same usb camera. The device class is 239 in Android 8.0, but 0 in Android 9.0.
> So I added code "if (device.getDeviceClass()==0&&device.getDeviceSubclass()==0) return true;" in DeviceFilter matches(UsbDevice), and it work normally!

It may not work again if u use another usb camera. I tested at least three results of the device subclass.

martin6960, 2019-02-13:
> It may not work again if u use another usb camera. I tested at least three results of the device subclass.

I am so sorry, it is not a good solution and I have no better solution, if you find, please tell me, thanks!