saki4510t/UVCCamera #440: Zoom functionality does not work

janoonaj, 2019-01-31:
Hello; I am trying to implement zooming and UVCCamera.setZoom(int) does nothing.

getZoom() always returns 0

I have seen this thread ([#367](https://github.com/saki4510t/UVCCamera/issues/367)) but the solution provided can't be implemented. It is using classes and interfaces that are not on master branch.

Did somebody figure out how to fiz the zoom?

Thanks

mingtong, 2019-03-06:
1, Use NDK-14b to compile
2, Check if Zoom Control(UVC feature) is enabled("dumpControls" function in UVCCamera.java)
3, Check if you can get the ZoomMin/ZoomMax value.
4, Try to set zoom value like setZoom(getZoom()+1);

athakur-systango, 2021-09-23:
Any update on the same ? Anyone got its solution