saki4510t/UVCCamera #441: Dose sample code work with Android Studio3.3 and Gradle4.10.1?

TommySato, 2019-02-12:
I compiled sample code with Android studio 3.2 and Gradle 3.3 (previous env.), it works fine.
But updated to Android studio 3.3 and Gradle 4.10.1(current env.), not work.
When open the camera device, application will down.
If I link libuvccamera.aar which is compiled with previous env. , no problem.
Dose anyone give me the advice? I'm sorry that I'm Android beginner.

wikihr, 2019-02-14:
Hello,  Did you try this? #181

TommySato, 2019-02-15:
Yes, I tried 

"int len = usbi_parse_descriptor(buffer, "bb", &header, 0);"
App will crush when press OK on camera selecting dialog.
It's same as before with NDK r19.
E/UVCCamera: [20785*UVCCamera.cpp:180:connect]:could not open camera:err=-1
It's no problem with NDK r17c.
 
"LOCAL_CFLAGS += -O1 -fstrict-aliasing -fprefetch-loop-arrays"
error has been changed.
E/libuvc/stream: [8206*stream.c:1605:uvc_stream_start_bandwidth]:fail
It's NDK r19 and r17c both, very strange.

TommySato, 2019-02-20:
Finally, I found the solution.
in descriptor.c
-----------------------------------------------------------------------------------
#pragma clang optimize off
static int parse_interface(libusb_context *ctx,
	struct libusb_interface *usb_interface, unsigned char *buffer, int size,
	int host_endian)
{
...........................
}
#pragma clang optimize on
-----------------------------------------------------------------------------------
It works with my environment. 
Android studio 3.3
Gradle 4.10.1 or  5.2
Android plugin version 3.3.1
NDK r17 r18 r19

closed 2019-02-20 by TommySato