saki4510t/UVCCamera #442: crash on android 9.0

lvi166, 2019-02-16:
when i Unplug the usb camera and this error will occur.
phone:  HUAWEI Mata20 pro
OS:  Android 9.0

2019-02-16 13:34:45.354 28617-28617/? A/DEBUG: pid: 28239, tid: 28601, name: USBMonitor  >>> com.serenegiant.usbcameratest <<<
2019-02-16 13:34:45.355 28617-28617/? A/DEBUG:     #00 pc 00011c34  /data/app/com.serenegiant.usbcameratest-oC158m_o_EgCmU7x2N086A==/lib/arm/libusb100.so
2019-02-16 13:34:45.355 28617-28617/? A/DEBUG:     #01 pc 0000d200  /data/app/com.serenegiant.usbcameratest-oC158m_o_EgCmU7x2N086A==/lib/arm/libusb100.so
2019-02-16 13:34:45.355 28617-28617/? A/DEBUG:     #02 pc 0000cb60  /data/app/com.serenegiant.usbcameratest-oC158m_o_EgCmU7x2N086A==/lib/arm/libusb100.so (libusb_handle_events_timeout_completed+968)
2019-02-16 13:34:45.355 28617-28617/? A/DEBUG:     #03 pc 0000d3cc  /data/app/com.serenegiant.usbcameratest-oC158m_o_EgCmU7x2N086A==/lib/arm/libusb100.so (libusb_handle_events+52)
2019-02-16 13:34:45.355 28617-28617/? A/DEBUG:     #04 pc 00012c20  /data/app/com.serenegiant.usbcameratest-oC158m_o_EgCmU7x2N086A==/lib/arm/libuvc.so (_uvc_handle_events+136)

Can anyone tell me how to solve this problem? thanks

mingtong, 2019-02-20:
You'd better re-compile the 4 libs using NDK-r14b.
And paste the accurate error information.

lvi166, 2019-02-21:
> You'd better re-compile the 4 libs using NDK-r14b.
> And paste the accurate error information.

`2019-02-21 09:24:35.320 20269-20366/? A/libc: Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0 in tid 20366 (BaseActivity), pid 20269 (.usbcameratest0)
2019-02-21 09:24:35.394 20372-20372/? A/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
2019-02-21 09:24:35.394 20372-20372/? A/DEBUG: Build fingerprint: 'HUAWEI/LYA-AL00/HWLYA:9/HUAWEILYA-AL00/195C00R1:user/release-keys'
2019-02-21 09:24:35.394 20372-20372/? A/DEBUG: Revision: '0'
2019-02-21 09:24:35.394 20372-20372/? A/DEBUG: ABI: 'arm'
2019-02-21 09:24:35.394 20372-20372/? A/DEBUG: pid: 20269, tid: 20366, name: BaseActivity  >>> com.serenegiant.usbcameratest0 <<<
2019-02-21 09:24:35.394 20372-20372/? A/DEBUG: signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0
2019-02-21 09:24:35.395 20372-20372/? A/DEBUG: Cause: null pointer dereference
2019-02-21 09:24:35.395 20372-20372/? A/DEBUG:     r0  00000000  r1  ecf9efe8  r2  0000ffff  r3  c587eb00
2019-02-21 09:24:35.395 20372-20372/? A/DEBUG:     r4  c587eb00  r5  c90969cc  r6  c9096800  r7  00000004
2019-02-21 09:24:35.395 20372-20372/? A/DEBUG:     r8  e80eff80  r9  00000000  r10 c9096820  r11 c577f858
2019-02-21 09:24:35.395 20372-20372/? A/DEBUG:     ip  ec5cefe4  sp  c577f6a0  lr  ec5cdd83  pc  c91b2ab8
2019-02-21 09:24:35.396 20372-20372/? A/DEBUG: backtrace:
2019-02-21 09:24:35.396 20372-20372/? A/DEBUG:     #00 pc 00011ab8  /data/app/com.serenegiant.usbcameratest0-8VPrewEQvandQeDyVH6p6g==/lib/arm/libusb100.so
2019-02-21 09:24:35.396 20372-20372/? A/DEBUG:     #01 pc 0000d200  /data/app/com.serenegiant.usbcameratest0-8VPrewEQvandQeDyVH6p6g==/lib/arm/libusb100.so
2019-02-21 09:24:35.396 20372-20372/? A/DEBUG:     #02 pc 0000cb60  /data/app/com.serenegiant.usbcameratest0-8VPrewEQvandQeDyVH6p6g==/lib/arm/libusb100.so (libusb_handle_events_timeout_completed+968)
2019-02-21 09:24:35.396 20372-20372/? A/DEBUG:     #03 pc 0000d3cc  /data/app/com.serenegiant.usbcameratest0-8VPrewEQvandQeDyVH6p6g==/lib/arm/libusb100.so (libusb_handle_events+52)
2019-02-21 09:24:35.396 20372-20372/? A/DEBUG:     #04 pc 00012c20  /data/app/com.serenegiant.usbcameratest0-8VPrewEQvandQeDyVH6p6g==/lib/arm/libuvc.so (_uvc_handle_events+136)
2019-02-21 09:24:35.396 20372-20372/? A/DEBUG:     #05 pc 00065413  /system/lib/libc.so (__pthread_start(void*)+22)
2019-02-21 09:24:35.396 20372-20372/? A/DEBUG:     #06 pc 0001e819  /system/lib/libc.so (__start_thread+32)`

-------------------------------------------------------------------------
I located the error,
`#02 pc 0000cb60  /data/app/com.serenegiant.usbcameratest0-8VPrewEQvandQeDyVH6p6g==/lib/arm/libusb100.so (libusb_handle_events_timeout_completed+968)`

In 
UVCCamera\libuvccamera\src\main\jni\libusb\libusb\io.c line 1215
`static int calculate_timeout(struct usbi_transfer *transfer) {

	int r;
	struct timespec current_time;
	unsigned int timeout =
		USBI_TRANSFER_TO_LIBUSB_TRANSFER(transfer)->timeout;
	if (!timeout)
		return LIBUSB_SUCCESS;
	r = usbi_backend->clock_gettime(USBI_CLOCK_MONOTONIC, &current_time);
	if (UNLIKELY(r < 0)) {
		usbi_err(ITRANSFER_CTX(transfer),
			"failed to read monotonic clock, errno=%d", errno);
		return r;
	}
  	current_time.tv_sec += timeout / 1000;
  	current_time.tv_nsec += (timeout % 1000) * 1000000;//----> The error is here
	while (current_time.tv_nsec >= 1000000000) {
		current_time.tv_nsec -= 1000000000;
		current_time.tv_sec++;
	}
	TIMESPEC_TO_TIMEVAL(&transfer->timeout, &current_time);
	return 1;
}`

lvi166, 2019-02-27:
> You'd better re-compile the 4 libs using NDK-r14b.
> And paste the accurate error information.
```
********** Crash dump: **********
Build fingerprint: 'HUAWEI/LYA-AL00/HWLYA:9/HUAWEILYA-AL00/195C00R1:user/release
-keys'
pid: 2660, tid: 2807, name: BaseActivity  >>> com.serenegiant.usbcameratest <<<
signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0
Stack frame 02-27 14:56:17.854  2822  2822 F DEBUG   :     #00 pc 00011c34  /dat
a/app/com.serenegiant.usbcameratest-lDYAayQNCO1MnD5AkwLPnA==/lib/arm/libusb100.s
o: Routine op_reset_device at E:\sdk\UVCCamera\libuvccamera\src\main/jni/libusb/
android/jni/../../libusb/os/android_usbfs.c:1866 (discriminator 1)
Stack frame 02-27 14:56:17.854  2822  2822 F DEBUG   :     #01 pc 0000d200  /dat
a/app/com.serenegiant.usbcameratest-lDYAayQNCO1MnD5AkwLPnA==/lib/arm/libusb100.s
o: Routine list_del at E:\sdk\UVCCamera\libuvccamera\src\main/jni/libusb/android
/jni/../../libusb/libusbi.h:129
Stack frame 02-27 14:56:17.854  2822  2822 F DEBUG   :     #02 pc 0000cb60  /dat
a/app/com.serenegiant.usbcameratest-lDYAayQNCO1MnD5AkwLPnA==/lib/arm/libusb100.s
o (libusb_handle_events_timeout_completed+968): Routine libusb_hotplug_register_
callback at E:\sdk\UVCCamera\libuvccamera\src\main/jni/libusb/android/jni/../../
libusb/hotplug.c:249
Stack frame 02-27 14:56:17.854  2822  2822 F DEBUG   :     #03 pc 0000d3cc  /dat
a/app/com.serenegiant.usbcameratest-lDYAayQNCO1MnD5AkwLPnA==/lib/arm/libusb100.s
o (libusb_handle_events+52): Routine calculate_timeout at E:\sdk\UVCCamera\libuv
ccamera\src\main/jni/libusb/android/jni/../../libusb/io.c:1218
Stack frame 02-27 14:56:17.854  2822  2822 F DEBUG   :     #04 pc 00012c20  /dat
a/app/com.serenegiant.usbcameratest-lDYAayQNCO1MnD5AkwLPnA==/lib/arm/libuvc.so (
_uvc_handle_events+136): Routine uvc_mjpeg2bgr at E:\sdk\UVCCamera\libuvccamera\
src\main/jni/libuvc/android/jni/../../src/frame-mjpeg.c:298
Stack frame 02-27 14:56:17.854  2822  2822 F DEBUG   :     #05 pc 00065413  /sys
tem/lib/libc.so (__pthread_start(void*)+22)
Stack frame 02-27 14:56:17.854  2822  2822 F DEBUG   :     #06 pc 0001e819  /sys
tem/lib/libc.so (__start_thread+32)
```

xuwei9, 2019-02-27:
> ```
> ********** Crash dump: **********
> Build fingerprint: 'HUAWEI/LYA-AL00/HWLYA:9/HUAWEILYA-AL00/195C00R1:user/release
> -keys'
> pid: 2660, tid: 2807, name: BaseActivity  >>> [...]
is a new problem?

anykey-skylocker, 2019-02-28:
I have exactly the same crash on device detach using Android 9.0 on Samsung Galaxy S9 and One Plus 6

```
I/crash_dump64: obtaining output fd from tombstoned, type: kDebuggerdTombstone
I//system/bin/tombstoned: received crash request for pid 3378
I/crash_dump64: performing dump of process 3295 (target tid = 3378)

*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
Build fingerprint: 'samsung/star2lteser/star2lte:9/PPR1.180610.011/G965FPUU2CSA2:user/release-keys'
Revision: '26'
ABI: 'arm64'
pid: 3295, tid: 3378, name: Thread-13  >>> com.example <<<
signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
Abort message: 'FORTIFY: pthread_mutex_lock called on a destroyed mutex (0x6fadd4aab4)'
    x0  0000000000000000  x1  0000000000000d32  x2  0000000000000006  x3  0000000000000008
    x4  0000000000008080  x5  0000000000008080  x6  0000000000008080  x7  0000000000000038
    x8  0000000000000083  x9  000000704767b890  x10 fffffff87ffffbdf  x11 0000000000000001
    x12 0000000000000030  x13 000000005c6ab0db  x14 002f4236c518a708  x15 0000487955d7bee6
    x16 00000070476b22b0  x17 00000070475f1958  x18 0000000000000000  x19 0000000000000cdf
    x20 0000000000000d32  x21 0000000000000083  x22 0000000000000004  x23 0000000000000003
    x24 0000006fbe926f40  x25 000000000000005c  x26 0000000000000002  x27 0000006fadd51100
    x28 0000006faa1bcdf0  x29 0000006fa7fff090
    sp  0000006fa7fff050  lr  00000070475e4da0  pc  00000070475e4dcc
backtrace:
    #00 pc 0000000000021dcc  /system/lib64/libc.so (abort+124)
    #01 pc 0000000000086508  /system/lib64/libc.so (__fortify_fatal(char const*, ...)+120)
    #02 pc 0000000000085bb4  /system/lib64/libc.so (HandleUsingDestroyedMutex(pthread_mutex_t*, char const*)+52)
    #03 pc 0000000000085a60  /system/lib64/libc.so (pthread_mutex_lock+256)
    #04 pc 000000000000fa68  /data/app/com.example-nig2tYAMAzRejsDuxIP7KA==/lib/arm64/libusb100.so
    #05 pc 000000000000f970  /data/app/com.example-nig2tYAMAzRejsDuxIP7KA==/lib/arm64/libusb100.so
    #06 pc 000000000000dd68  /data/app/com.example-nig2tYAMAzRejsDuxIP7KA==/lib/arm64/libusb100.so
    #07 pc 000000000000abe4  /data/app/com.example-nig2tYAMAzRejsDuxIP7KA==/lib/arm64/libusb100.so
    #08 pc 000000000000a818  /data/app/com.example-nig2tYAMAzRejsDuxIP7KA==/lib/arm64/libusb100.so (libusb_handle_events_timeout_completed+244)
    #09 pc 000000000000ad54  /data/app/com.example-nig2tYAMAzRejsDuxIP7KA==/lib/arm64/libusb100.so (libusb_handle_events+44)
    #10 pc 0000000000011854  /data/app/com.example-nig2tYAMAzRejsDuxIP7KA==/lib/arm64/libuvc.so (_uvc_handle_events+136)
    #11 pc 0000000000084df8  /system/lib64/libc.so (__pthread_start(void*)+208)
    #12 pc 0000000000023ac4  /system/lib64/libc.so (__start_thread+68)
```

Is there any other information that could be helpful in resolving this issue?

lvi166, 2019-03-01:
> I have exactly the same crash on device detach using Android 9.0 on Samsung Galaxy S9 and One Plus 6
>
> ```
> I/crash_dump64: obtaining output fd from tombstoned, type: kDebuggerdTombstone [...]

you can use the ndk-stack  to locatate the  the accurate error information.

anykey-skylocker, 2019-03-01:
`ndk-stack` gives me not much more information than I posted above. And I can't figure out how could I get complete information about lines 4-7 in stack trace. I tried building with NDK r14b and 19.0.5232133, tried `APP_OPTIM := debug` and `LOCAL_CFLAGS := -g` but had no luck.

lbb1993, 2019-03-06:
I hava the same problem,is there anybody solved the problem？

mengnick, 2019-03-06:
I hava the same problem with huawei hornor 6X and 7X

03-06 15:36:58.061 24676 24676 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
03-06 15:36:58.061 24676 24676 F DEBUG   : Build fingerprint: 'HONOR/BLN-AL20/HWBLN-H:8.0.0/HONORBLN-AL20/541(C00):user/release-keys'
03-06 15:36:58.061 24676 24676 F DEBUG   : Revision: '0'
03-06 15:36:58.062 24676 24676 F DEBUG   : ABI: 'arm'
03-06 15:36:58.062 24676 24676 F DEBUG   : pid: 24593, tid: 24668, name: CameraThread  >>> com.hk.qhyl <<<
03-06 15:36:58.062 24676 24676 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0xbef00000
03-06 15:36:58.062 24676 24676 F DEBUG   :     r0 beeffff0  r1 c071e032  r2 000001b2  r3 0000000a
03-06 15:36:58.062 24676 24676 F DEBUG   :     r4 000002bc  r5 c071df68  r6 c0f1ae00  r7 000002bc
03-06 15:36:58.062 24676 24676 F DEBUG   :     r8 c17fce68  r9 c071df69  sl 00000000  fp c0a54828
03-06 15:36:58.062 24676 24676 F DEBUG   :     ip 40000000  sp c0a547d0  lr 00000000  pc e9636f8c  cpsr 200b0010
03-06 15:36:58.064 24676 24676 F DEBUG   : 
03-06 15:36:58.064 24676 24676 F DEBUG   : backtrace:
03-06 15:36:58.064 24676 24676 F DEBUG   :     #00 pc 00018f8c  /system/lib/libc.so (memcpy+112)
03-06 15:36:58.064 24676 24676 F DEBUG   :     #01 pc 00012e1c  /data/app/com.hk.qhyl-iELzi8S6rxaOQClMqwsPCg==/lib/arm/libuvc.so
03-06 15:36:58.064 24676 24676 F DEBUG   :     #02 pc 00008ec4  /data/app/com.hk.qhyl-iELzi8S6rxaOQClMqwsPCg==/lib/arm/libusb100.so (usbi_handle_transfer_completion+296)
03-06 15:36:58.064 24676 24676 F DEBUG   :     #03 pc 0000e608  /data/app/com.hk.qhyl-iELzi8S6rxaOQClMqwsPCg==/lib/arm/libusb100.so
03-06 15:36:58.064 24676 24676 F DEBUG   :     #04 pc 0000ca7c  /data/app/com.hk.qhyl-iELzi8S6rxaOQClMqwsPCg==/lib/arm/libusb100.so
03-06 15:36:58.064 24676 24676 F DEBUG   :     #05 pc 000097dc  /data/app/com.hk.qhyl-iELzi8S6rxaOQClMqwsPCg==/lib/arm/libusb100.so
03-06 15:36:58.064 24676 24676 F DEBUG   :     #06 pc 00009418  /data/app/com.hk.qhyl-iELzi8S6rxaOQClMqwsPCg==/lib/arm/libusb100.so (libusb_handle_events_timeout_completed+244)
03-06 15:36:58.064 24676 24676 F DEBUG   :     #07 pc 0000994c  /data/app/com.hk.qhyl-iELzi8S6rxaOQClMqwsPCg==/lib/arm/libusb100.so (libusb_handle_events+36)
03-06 15:36:58.064 24676 24676 F DEBUG   :     #08 pc 00010e7c  /data/app/com.hk.qhyl-iELzi8S6rxaOQClMqwsPCg==/lib/arm/libuvc.so (_uvc_handle_events+132)
03-06 15:36:58.064 24676 24676 F DEBUG   :     #09 pc 00048c9f  /system/lib/libc.so (_ZL15__pthread_startPv+22)
03-06 15:36:58.064 24676 24676 F DEBUG   :     #10 pc 0001b09f  /system/lib/libc.so (__start_thread+32)
03-06 15:37:11.958 24679 24739 F libc    : Fatal signal 11 (SIGSEGV), code 1, fault addr 0xbc600000 in tid 24739 (CameraThread)

mengnick, 2019-03-06:
I hava the same problem with Huawei Honor 6X and 7X ，It is hard to  find the root cause.

juechenshiai, 2019-03-25:
I have the same problem when camera device is detached with Huawei mate 10

AndreiStanciu89, 2019-05-02:
I randomly stumbled upon this thread a few days ago, while investigating a similar crashlog, and figured I might as well post my findings

I believe these crashes are related to a change done by google in Bionic, for Android 9:
https://android.googlesource.com/platform/bionic/+/9e989f12d1186231d97dac6d038db7955acebdf3%5E%21

To be precise, this is the commit that introduced the HandleUsingDestroyedMutex method, which is present in all of the stacktraces that have been posted here:

+static int __attribute__((noinline)) HandleUsingDestroyedMutex(pthread_mutex_t* mutex,
+                                                               const char* function_name) {
+    if (bionic_get_application_target_sdk_version() >= __ANDROID_API_P__) {
+        __fortify_fatal("%s called on a destroyed mutex (%p)", function_name, mutex);
+    }
+    return EBUSY;
+}

Also, this statement was added inside the pthread_mutex_lock (also present in all the stacktraces):
+    if (__predict_false(IsMutexDestroyed(old_state))) {
+        return HandleUsingDestroyedMutex(mutex_interface, __FUNCTION__);
+    }

So by the looks of it, something inside the libusb100.so library is calling pthread_mutex_lock on an already destroyed mutex, which results in a hard crash only on Android 9.

mingtong, 2019-05-05:
> I randomly stumbled upon this thread a few days ago, while investigating a similar crashlog, and figured I might as well post my findings
>
> I believe these crashes are related to a change done by [...]

Boy, I don't think libusb100 will result this, and what you pasted seemed like a low level error that is not expected to happen.
My Sumsang N950 with Android 9.0 works well.

AndreiStanciu89, 2019-05-10:
I'm just pointing out what the stacktrace is saying :)

    #00 pc 0000000000021dcc  /system/lib64/libc.so (abort+124)
    #01 pc 0000000000086508  /system/lib64/libc.so (__fortify_fatal(char const*, ...)+120)
    #02 pc 0000000000085bb4  /system/lib64/libc.so (HandleUsingDestroyedMutex(pthread_mutex_t*, char const*)+52)
    #03 pc 0000000000085a60  /system/lib64/libc.so (pthread_mutex_lock+256)
    #04 pc 000000000000fa68  /data/app/com.example-nig2tYAMAzRejsDuxIP7KA==/lib/arm64/libusb100.so

The exception is triggered in libusb100.so (line #04) and then you can follow the trace through the native methods that I explained above, which google introduced in android 9.

As a side note, some of the devs maintaining some custom ROMs (DotOS, ArrowOS, LineageOS, and a few others) have reverted the changes google did in the commit I linked, precisely because they spotted the potential issues with it. So I wouldn't expect this crash to occur on them.

Huawei-Dev, 2019-05-22:
Hi !
I had the same issue as you all here, but was sensors related ... so basically if you use older blobs on Android Pie, you souldn't build the service related to your blobs ( in my case was : android.hardware.sensors@1.0-service ) ... you need to keep only implementation interface ( in my case was : android.hardware.sensors@1.0-impl ) ... and on manifest.xml you need to change the hwbinder transport ( <transport>hwbinder</transport> ) with passthrough ( <transport arch="32+64">passthrough</transport> ) !   

Logically, now if your issue is camera related, you should remove android.hardware.camera.provider@2.4-service from device.mk and change on manifest.xml ( <transport>hwbinder</transport> ) with  (<transport arch="32+64">passthrough</transport>) for (android.hardware.camera.provider) !

Rostyk, 2021-06-22:
@Huawei-Dev  @AndreiStanciu89 Were you guys able to resolve this pthread_mutex_lock issue at a certain point?

radheyjat, 2021-08-31:
D/InputTransport(16535): Input channel constructed: 'b4f35ac', fd=140
I/ViewRootImpl@ce67b10[ProductView](16535): setView = com.android.internal.policy.DecorView@22f6038 TM=true
V/ViewRootImpl@61afd99[ProductView](16535): updateAppliedLetterboxDirection, direction=0, Caller=android.view.ViewRootImpl.handleDispatchLetterboxDirectionChanged:12053
F/libc    (16535): Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 24732 (CameraThread), pid 16535 (enesis.fieldapp)
*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
Build fingerprint: 'samsung/a71nsxx/a71:11/RP1A.200720.012/A715FXXU6BUH1:user/release-keys'
Revision: '6'
ABI: 'arm64'
Timestamp: 2021-08-31 21:12:02+0530
pid: 16535, tid: 24732, name: CameraThread  >>> **package name** <<<
uid: 10537
signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
Abort message: 'FORTIFY: pthread_mutex_lock called on a destroyed mutex (0x71a24f5fe4)'
    x0  0000000000000000  x1  000000000000609c  x2  0000000000000006  x3  00000070381eb830
    x4  0000000000008080  x5  0000000000008080  x6  0000000000008080  x7  0000000000000038
    x8  00000000000000f0  x9  0a8c7ce8e43c1cab  x10 0000000000000000  x11 ffffffc0fffffbdf
    x12 0000000000000001  x13 00000000612e4dc9  x14 002f684aefd4e4aa  x15 000099278f69afbd
    x16 0000007356046c80  x17 0000007356026600  x18 000000700f57e000  x19 0000000000004097
    x20 000000000000609c  x21 00000000ffffffff  x22 0000000000000000  x23 0000007112498b68
    x24 000000715246d910  x25 0000000000000004  x26 0000000000000002  x27 000000712245e760
    x28 0000000000000003  x29 00000070381eb8b0
    lr  0000007355fd6038  sp  00000070381eb810  pc  0000007355fd6064  pst 0000000000000000
backtrace:
      #00 pc 000000000004f064  /apex/com.android.runtime/lib64/bionic/libc.so (abort+164) (BuildId: c5ebfbffe15ece8c13ef9afa88b800fb)
      #01 pc 00000000000b7c20  /apex/com.android.runtime/lib64/bionic/libc.so (__fortify_fatal(char const*, ...)+124) (BuildId: c5ebfbffe15ece8c13ef9afa88b800fb)
      #02 pc 00000000000b70bc  /apex/com.android.runtime/lib64/bionic/libc.so (HandleUsingDestroyedMutex(pthread_mutex_t*, char const*)+52) (BuildId: c5ebfbffe15ece8c13ef9afa88b800fb)
      #03 pc 00000000000b6f70  /apex/com.android.runtime/lib64/bionic/libc.so (pthread_mutex_lock+276) (BuildId: c5ebfbffe15ece8c13ef9afa88b800fb)
      #04 pc 0000000000010208  /data/app/~~YBWfL9ZhjqaNvg-uxgf1aA==/com.genesis.fieldapp-i4qAQ37_g7VON5yCb5GpfA==/base.apk!libusb100.so (offset 0x42d5000) (BuildId: d3d320d24e295f7658d8c85a9c614c0626b535cd)
      #05 pc 000000000000c28c  /data/app/~~YBWfL9ZhjqaNvg-uxgf1aA==/com.genesis.fieldapp-i4qAQ37_g7VON5yCb5GpfA==/base.apk!libusb100.so (offset 0x42d5000) (BuildId: d3d320d24e295f7658d8c85a9c614c0626b535cd)
      #06 pc 000000000000bcf4  /data/app/~~YBWfL9ZhjqaNvg-uxgf1aA==/com.genesis.fieldapp-i4qAQ37_g7VON5yCb5GpfA==/base.apk!libusb100.so (offset 0x42d5000) (libusb_handle_events_timeout_completed+748) (BuildId: d3d320d24e295f7658d8c85a9c614c0626b535cd)
      #07 pc 000000000000c414  /data/app/~~YBWfL9ZhjqaNvg-uxgf1aA==/com.genesis.fieldapp-i4qAQ37_g7VON5yCb5GpfA==/base.apk!libusb100.so (offset 0x42d5000) (libusb_handle_events+44) (BuildId: d3d320d24e295f7658d8c85a9c614c0626b535cd)
      #08 pc 0000000000011e24  /data/app/~~YBWfL9ZhjqaNvg-uxgf1aA==/com.genesis.fieldapp-i4qAQ37_g7VON5yCb5GpfA==/base.apk!libuvc.so (offset 0x7bcc000) (_uvc_handle_events+132) (BuildId: bbd55feb65914680ce546556f4819427a6cc0cfd)
      #09 pc 00000000000b6374  /apex/com.android.runtime/lib64/bionic/libc.so (__pthread_start(void*)+64) (BuildId: c5ebfbffe15ece8c13ef9afa88b800fb)
      #10 pc 0000000000050fa4  /apex/com.android.runtime/lib64/bionic/libc.so (__start_thread+64) (BuildId: c5ebfbffe15ece8c13ef9afa88b800fb)

panping1314, 2021-10-13:
Has the problem been solved

YuSiwenXC, 2021-11-19:
> I found this issue happend when I disconnect device which is previewing。 I found that in libusb_free_transfer ， comment out //usbi_mutex_destroy(&itransfer->lock); will solve this issue

Could u please share the detail solution? thx

johnnybib, 2022-06-17:
In case anyone else is looking for a solution to this problem, I think I've found one.
In the file `io.c`, 
in the function `libusb_free_transfer`, 
above `usbi_mutex_destroy(&itransfer->lock);` on line 1367 add this code:

`usbi_mutex_lock(&itransfer->lock);`

This will lock the mutex before destroying it, so that the code in `android_usbfs.c` in the function `handle_iso_completion` doesn't try to lock the destroyed mutex (that's where my code was erroring out)

singledevil0, 2022-08-22:
> 
你的意思是注销掉这个锁就可以啦，多了一个 destroy

singledevil0, 2022-08-23:
> 

添加一个锁对称

BunnyBuddy, 2023-07-12:
I have the same crash specific to Huawei and Honor devices in Flutter,

[HONOR HWLLD-H (Honor 9 Lite)]
[Android 9 (SDK 28)]

PlayStore gives this crash log,

Thread: FORTIFY: pthread_mutex_lock called on a destroyed mutex (0x<sanitized>)
*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
pid: 0, tid: 24203 >>> com.appName <<<
backtrace:
  #00  pc 0x0000000000022e90  /system/lib64/libc.so (abort+116)
  #01  pc 0x0000000000047694  /system/lib64/libc++.so (abort_message+248)
  #02  pc 0x00000000000477b0  /system/lib64/libc++.so (demangling_terminate_handler()+44)
  #03  pc 0x0000000000059f3c  /system/lib64/libc++.so (std::__terminate(void (*)())+12)
  #04  pc 0x0000000000059fc4  /system/lib64/libc++.so (std::terminate()+72)
  #05  pc 0x00000000000b2380  /system/lib64/libc++.so (std::__1::thread::~thread()+20)
  #06  pc 0x0000000000086750  /system/lib64/libc.so (__cxa_finalize+196)
  #07  pc 0x000000000001c798  /system/lib64/libc.so (exit+24)
  #08  pc 0x000000000134bea8  /data/app/com.appName-PvphphBM4skYyarheRhc6Q==/split_config.arm64_v8a.apk

redelivreplus, 2026-03-26:
"Thank you for the help! I am not a professional developer; I used ChatGPT to build an app using the RootEncoder library, and it worked great. However, I now want to use an external camera via OTG cable, but I'm having trouble configuring the project with the older native libraries. For this reason, I am looking for a reference project that is already compiling correctly in the latest versions of Android Studio. I truly appreciate your time and assistance!"