saki4510t/UVCCamera #444: could not open camera:err=-99

ahmadazarnia, 2019-02-21:
Hi, I have connected two webcams to my Odroid device. at the first run when I selected the camera, it perfectly worked. also when I turned the camera off and toggled it o another camera , again it worked without any problem. But when I was going to get back the first camera, I was faced with this error:

`[4464*android_usbfs.c:1776:release_interface]:release interface failed, error -1 errno 22
[4464*UVCCamera.cpp:172:connect]:could not open camera:err=-99`

Thanks in advance.

Heipi, 2019-03-01:
I meet she same problem , how solve it ?

HoekWax, 2019-04-16:
Hello

Before opening up the camera again, you have to close it.
You should find in the samples a function called releaseCamera()
It basically contains this

            mUVCCamera?.close()
            mUVCCamera?.destroy()
            mUVCCamera = null
            mPreviewSurface?.release()
            mPreviewSurface = null

Good luck.

heaveninfosoft, 2020-05-19:
> Hi, I have connected two webcams to my Odroid device. at the first run when I selected the camera, it perfectly worked. also when I turned the camera off and toggled it to another camera , again it worked without any problem. But when I was going to get back the first camera, I was faced with this error:
> 
> `[4464*android_usbfs.c:1776:release_interface]:release interface failed, error -1 errno 22 [4464*UVCCamera.cpp:172:connect]:could not open camera:err=-99`
> 
> Thanks in advance.

in my second when attach this error show please help in my second cam not working

allencheung2018, 2021-03-20:
@ahmadazarnia Have you solved it? I have the same issue.

Micoder-dev, 2023-10-09:
I solved this by calling this updated 
    private fun stopService() {
        try {
            stopService(Intent(this, StreamService::class.java))
            mService = null
            unbindService(this)
        } catch (e: Exception) {
            e.printStackTrace()
        }
    }
    
    call this fun before `startService(getServiceIntent())` in onCreate(), hope this will helps you

luan-tran-michel, 2024-07-08:
> [4464*android_usbfs.c:1776:release_interface]:release interface failed, error -1 errno 22 [4464*UVCCamera.cpp:172:connect]:could not open camera:err=-99

Hi I have the same problem and the app crashs. In usbCameraTest project MainActivity onConnect Function I commented :

synchronized (mSync) {
   mUVCCamera = camera;
}

Now it's working but still have error : 

[4464*android_usbfs.c:1776:release_interface]:release interface failed, error -1 errno 22 
[4464*UVCCamera.cpp:172:connect]:could not open camera:err=-99