saki4510t/UVCCamera #446: crash with android 9

lbb1993, 2019-03-06:
When I using second screen show the uvc camera preview，if I remove the second screen  app will crash。I don't know how to avoid crash.who can help me ？Thank you very much !
Phone: HUAWEI Mata10 
OS: Android 9.0
Crash log：
2019-03-06 09:47:46.831 30681-30681/? A/DEBUG: pid: 30405, tid: 30660, name: CameraThread  >>> com.seengene.xmanapp <<<
2019-03-06 09:47:46.832 30681-30681/? A/DEBUG:     #04 pc 00010c34  /data/app/com.seengene.xmanapp-F3ctHGmz0Elk3-4dbj04hg==/lib/arm/libusb100.so
2019-03-06 09:47:46.832 30681-30681/? A/DEBUG:     #05 pc 0000a484  /data/app/com.seengene.xmanapp-F3ctHGmz0Elk3-4dbj04hg==/lib/arm/libusb100.so
2019-03-06 09:47:46.832 30681-30681/? A/DEBUG:     #06 pc 0000afd0  /data/app/com.seengene.xmanapp-F3ctHGmz0Elk3-4dbj04hg==/lib/arm/libusb100.so (libusb_handle_events_timeout_completed+292)
2019-03-06 09:47:46.833 30681-30681/? A/DEBUG:     #07 pc 0000b15c  /data/app/com.seengene.xmanapp-F3ctHGmz0Elk3-4dbj04hg==/lib/arm/libusb100.so (libusb_handle_events+48)
2019-03-06 09:47:46.833 30681-30681/? A/DEBUG:     #08 pc 00012094  /data/app/com.seengene.xmanapp-F3ctHGmz0Elk3-4dbj04hg==/lib/arm/libuvc.so (_uvc_handle_events+64)
2019-03-06 09:47:47.120 30405-30405/? E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.seengene.xmanapp, PID: 30405
    java.lang.IllegalArgumentException: view must not be null
        at android.view.WindowManagerGlobal.removeView(WindowManagerGlobal.java:406)
        at android.view.WindowManagerImpl.removeView(WindowManagerImpl.java:187)
        at com.seengene.xmansdk.preso.PresentationService$1.run(PresentationService.java:118)
        at android.os.Handler.handleCallback(Handler.java:891)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:207)
        at android.app.ActivityThread.main(ActivityThread.java:7470)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:524)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:958)

lbb1993, 2019-03-06:
Sorry,above log has some jrrelevant info.
Crash Log:
2019-03-06 10:22:47.826 822-1361/com.seengene.xmanapp A/libc: FORTIFY: pthread_mutex_lock called on a destroyed mutex (0xc6b11e20)
2019-03-06 10:22:47.830 822-1361/com.seengene.xmanapp A/libc: Fatal signal 6 (SIGABRT), code -6 (SI_TKILL) in tid 1361 (CameraThread), pid 822 (eengene.xmanapp)
2019-03-06 10:22:47.833 1995-2577/? I/ash_trans: com.seengene.xmanapp { running duration=14700 Uptime=14701 } transition to: end, reason:not_running
2019-03-06 10:22:47.833 1995-2577/? I/hibernation: remove app record pkg: com.seengene.xmanapp
2019-03-06 10:22:47.931 1158-1228/? V/WindowManager: postWindowRemoveCleanupLocked: Window{c29e2c8 u0 com.seengene.xmanapp}
2019-03-06 10:22:47.932 1158-1228/? V/WindowManager: Removing Window{c29e2c8 u0 com.seengene.xmanapp} from WindowToken{26e0db0 android.os.BinderProxy@ca39cf3}
2019-03-06 10:22:47.939 1380-1380/? A/DEBUG: pid: 822, tid: 1361, name: CameraThread  >>> com.seengene.xmanapp <<<
2019-03-06 10:22:47.940 1380-1380/? A/DEBUG:     #04 pc 00010c34  /data/app/com.seengene.xmanapp-r5-bTqOBp9FDXhHZJSUo-w==/lib/arm/libusb100.so
2019-03-06 10:22:47.940 1380-1380/? A/DEBUG:     #05 pc 0000a484  /data/app/com.seengene.xmanapp-r5-bTqOBp9FDXhHZJSUo-w==/lib/arm/libusb100.so
2019-03-06 10:22:47.940 1380-1380/? A/DEBUG:     #06 pc 0000afd0  /data/app/com.seengene.xmanapp-r5-bTqOBp9FDXhHZJSUo-w==/lib/arm/libusb100.so (libusb_handle_events_timeout_completed+292)
2019-03-06 10:22:47.940 1380-1380/? A/DEBUG:     #07 pc 0000b15c  /data/app/com.seengene.xmanapp-r5-bTqOBp9FDXhHZJSUo-w==/lib/arm/libusb100.so (libusb_handle_events+48)
2019-03-06 10:22:47.940 1380-1380/? A/DEBUG:     #08 pc 00012094  /data/app/com.seengene.xmanapp-r5-bTqOBp9FDXhHZJSUo-w==/lib/arm/libuvc.so (_uvc_handle_events+64)

lbb1993, 2019-03-06:
After long time struggle,I find the moudle "usbCameraTest" has the same problem.Pull out the usb when previewing will  trigger the crash.HuaWei mate10 with android 8.0 don't have this crash,after updating  to 9.0 the crash appear.If anyone has the same problem,please support this comment,thank you !

icemanyandy, 2019-03-10:
I tried ,not crash ,but  no Image for the Camera. does Any one make it right?

juechenshiai, 2019-03-25:
I have the same problem when camera device is detached

goldboy123, 2019-04-29:
i have meet the same problem!

dejianlei, 2019-05-31:
i have meet the same problem!

quocnhat7, 2019-06-18:
Hi all, that is a crash in native library. You wait for lib owner fix it.
Or if you can build native source, you can copy and build the lib yourself. I asked for help from my colleague , he put some line of codes more and the crash is gone.
Crash is not because of android 9 only. Some android phones OK, some are not.

lbb1993, 2019-06-18:
Can you post the some line of codes?thank you sir~

quocnhat7, 2019-06-19:
> Can you post the some line of codes?thank you sir~

pls check your email :)

lbb1993, 2019-06-19:
> > Can you post the some line of codes?thank you sir~
> 
> pls check your email :)

I have checked my gmail,but I still need your help.Please check your gmail.Thank you !

mengnick, 2019-06-19:
@quocnhat7  Would you please send your changes to my email ? I have the same problem.

oury1, 2019-06-19:
Hi can you post the correction here i have the same problem
thanks

oy-paddy, 2019-06-20:
Android9.0 requires camera permissions to work properly! Or change the targetSdkVersion to 26, not to 29！ @oury1 @mengnick @lbb1993 @quocnhat7 @goldboy123 @icemanyandy @juechenshiai @dejianlei

quocnhat7, 2019-06-20:
> Android9.0 requires camera permissions to work properly! Or change the targetSdkVersion to 26, not to 29！ @oury1 @mengnick @lbb1993 @quocnhat7 @goldboy123 @icemanyandy @juechenshiai @dejianlei

Pls check the question, understand and verify the real reason of the issue. Why you say "android9 require camera permission?" ? android 6/7/8 required.
And 1 thing you don't know is: only some android9 got crash, some are not.
And 1 have 2 apps which target 22 and 28, both got crash if run in 1 phone android9.

lbb1993, 2019-06-20:
> Android9.0 requires camera permissions to work properly! Or change the targetSdkVersion to 26, not to 29！ @oury1 @mengnick @lbb1993 @quocnhat7 @goldboy123 @icemanyandy @juechenshiai @dejianlei
I already have the camera permission.It doesn't work for me.Actually it is not the reason that results in crash.

quocnhat7, 2019-06-20:
@mengnick @oury1 pls see this android module.
I know this lib from its child project: [https://github.com/jiangdongguo/AndroidUSBCamera](url)

You can copy native libs only in folder "jniLibs" to use in your code.
[https://drive.google.com/file/d/1DmRhEcj1xnzcubRJKayxtcIhsUVrdOLg/view?usp=sharing](url)

When my colleague come back, I will ask him share his code update.

guyikun, 2019-06-21:
i have met the same problem,
Can you post the some line of codes?
Thanks.

quocnhat7, 2019-06-21:
See yellow highlightee in my picture: [https://prnt.sc/o4yv93](url)
static int handle_iso_completion(struct libusb_device_handle *handle,	// XXX added saki
		struct usbi_transfer *itransfer,
		struct usbfs_urb *urb) {
	struct libusb_transfer *transfer = USBI_TRANSFER_TO_LIBUSB_TRANSFER(itransfer);
	struct android_transfer_priv *tpriv = usbi_transfer_get_os_priv(itransfer);
	int num_urbs = tpriv->num_urbs;
	int urb_idx = 0;
	int i;
	enum libusb_transfer_status status = LIBUSB_TRANSFER_COMPLETED;

	usbi_mutex_lock(&itransfer->lock);
	for (i = 0; i < num_urbs; i++) {
		if( tpriv->iso_urbs == 0 )
			break;
		if (urb == tpriv->iso_urbs[i]) {
			urb_idx = i + 1;
			break;
		}
	}

guyikun, 2019-06-24:
> See yellow highlightee in my picture: [https://prnt.sc/o4yv93](url)
> .....
> And copy the code below...
> .....
> static int handle_iso_completion(struct libusb_device_handle *handle,	// XXX added saki
> struct usbi_transfer *itransfer,
> struct usbfs_urb *urb) {
> struct libusb_transfer *transfer = USBI_TRANSFER_TO_LIBUSB_TRANSFER(itransfer);
> struct android_transfer_priv *tpriv = usbi_transfer_get_os_priv(itransfer);
> int num_urbs = tpriv->num_urbs;
> int urb_idx = 0;
> int i;
> enum libusb_transfer_status status = LIBUSB_TRANSFER_COMPLETED;
> usbi_mutex_lock(&itransfer->lock);
> for (i = 0; i < num_urbs; i++) {
> if( tpriv->iso_urbs == 0 )
> break;
> if (urb == tpriv->iso_urbs[i]) {
> urb_idx = i + 1;
> break;
> }
> }
> ...

Thanks a lot !

guyikun, 2019-06-24:
Thanks a lot.
  

 

 ------------------ 原始邮件 ------------------

guyikun, 2019-06-24:
stream.c::
static void _uvc_delete_transfer(struct libusb_transfer *transfer) {
	ENTER();
......

	pthread_mutex_lock(&strmh->cb_mutex);	// XXX crash while calling uvc_stop_streaming
	{
		// Mark transfer as deleted.

		for (i = 0; i < LIBUVC_NUM_TRANSFER_BUFS; i++) {
			if (strmh->transfers[i] == transfer) {
				libusb_cancel_transfer(strmh->transfers[i]);	// XXX 20141112追加
				UVC_DEBUG("Freeing transfer %d (%p)", i, transfer);
				free(transfer->buffer);

				//+++  remove for it will cause crash in android 9, through it will cause memoryleak in ap +++
				**//libusb_free_transfer(transfer);**
				//-- ---

				strmh->transfers[i] = NULL;
				break;
			}
		}
......
}

quocnhat7, 2019-06-24:
My app don't use RECORDING, STREAMING or CAPTURE_PICTURE features so my colleague only fix at the place he found. So please report to lib owner if you crash when using them.

heaveninfosoft, 2020-05-31:
> My app don't use RECORDING, STREAMING or CAPTURE_PICTURE features so my colleague only fix at the place he found. So please report to lib owner if you crash when using them.

hey bro i need help i will pay you please comment here

oury1, 2020-05-31:
What u need?

heaveninfosoft, 2020-05-31:
hello msg me on my skype i really need your help
i need code which can support daul  HD,FHD Usb Camera support app with RECORDING, STREAMING or CAPTURE_PICTURE
please suggest me , i am struggling from 15 day

oury1, 2020-05-31:
Mail me on private please

heaveninfosoft, 2020-05-31:
@oury1 I cant find your mail so unble to send you private mail
My mail is harshad18ghori@gmail.com   please mail me or my skype is harshad.ghori   please msg me

oury1, 2020-05-31:
I have send you an email

quocnhat7, 2020-06-03:
> hello msg me on my skype i really need your help
> i need code which can support daul HD,FHD Usb Camera support app with RECORDING, STREAMING or CAPTURE_PICTURE
> please suggest me , i am struggling from 15 day

wow, you got many issue with the lib?

heaveninfosoft, 2020-06-06:
> > hello msg me on my skype i really need your help
> > i need code which can support daul HD,FHD Usb Camera support app with RECORDING, STREAMING or CAPTURE_PICTURE
> > please suggest me , i am struggling from 15 day
> 
> wow, you got many issue with the lib?

can you give me any Dual demo of HD and FHD .

heaveninfosoft, 2020-06-06:
> > > hello msg me on my skype i really need your help
> > > i need code which can support daul HD,FHD Usb Camera support app with RECORDING, STREAMING or CAPTURE_PICTURE
> > > please suggest me , i am struggling from 15 day
> > 
> > 
> > wow, you got many issue with the lib?
> 
> can you give me any Dual demo of HD and FHD .

i need code which can support daul HD,FHD Usb Camera support app with RECORDING, STREAMING or CAPTURE_PICTURE

quocnhat7, 2020-06-11:
> > can you give me any Dual demo of HD and FHD .
> 
> i need code which can support daul HD,FHD Usb Camera support app with RECORDING, STREAMING or CAPTURE_PICTURE

Sorry, I dont understand what is "dual demo of HD and FHD", pls write down the requirement in details.
You mean: app open 2 camera at the same time, one opens HD, another opens FHD?
And one capture picture, another do streaming?

heaveninfosoft, 2020-06-12:
> Sorry, I dont understand what is "dual demo of HD and FHD", pls write down the requirement in details.
> You mean: app open 2 camera at the same time, one opens HD, another opens FHD?
> And one capture [...]

yes  , but need to two usb camera can support HD and FHD .app open 2 camera at the same time, one opens HD, another opens FHD.   recording,streaming and capturing need for both camera

quocnhat7, 2020-06-12:
> yes , but need to two usb camera can support HD and FHD .app open 2 camera at the same time, one opens HD, another opens FHD. recording,streaming and capturing need for both camera
- How can you connect 2 cameras into 1 phone? Phone only has one port to connect @@
- Which type of apps need to use 2 cameras at once like that? 
- Perhaps you are using another Android device such as set-top-box or ???

heaveninfosoft, 2020-06-12:
> * How can you connect 2 cameras into 1 phone? Phone only has one port to connect @@
> * Which type of apps need to use 2 cameras at once like that?
> * Perhaps you are using another Android device such [...]

hey please see your mail or skype i already msg there we can discuuss there please??

mayi203, 2020-06-28:
> https://prnt.sc/o4yv93

Thanks a lot !

heaveninfosoft, 2020-06-28:
> > https://prnt.sc/o4yv93
> 
> Thanks a lot !

can you do Dual USB Camera need to two usb camera can support HD and FHD

one opens HD, another opens FHD. recording,streaming and capturing need for both camera

kamaljk86, 2023-01-30:
> See yellow highlightee in my picture: [https://prnt.sc/o4yv93](url) ..... And copy the code below... ..... static int handle_iso_completion(struct libusb_device_handle *handle, // XXX added saki [...]

Hi @quocnhat7 Yellow highligtee picture is not loading. Could you please send the updated code to fix this crash issue. Thanks for understanding.

Clark-caipeiyuan, 2023-01-30:
收到了，谢谢！

NanayaHaruki, 2023-06-29:
> 

stream.c::_uvc_delete_transfer
`pthread_mutex_lock(&strmh->cb_mutex);	// XXX crash while calling uvc_stop_streaming`
how to solve this problem?