saki4510t/UVCCamera #450: How can I rename test4?

wikihr, 2019-03-12:
Hi, I using test4 for my own project and want to rename about application's name.
I know I can change label at the strings.xml but I also want to change service permission how can I change it?
[image]

lvi166, 2019-03-14:
The first you should rename package. if you dont know how to do it , please search in google .and do <clean project>  <rebuild project> in Android Studio。
enjoy it

wikihr, 2019-03-18:
@lvi166 I also already try that several times(with https://stackoverflow.com/questions/14610013/change-package-name-of-java-and-aidl-files and https://stackoverflow.com/questions/16804093/rename-package-in-android-studio) but when I did that, then I couldn't see permission dialog.
Logs are only show like below. I changed only package's name.
03-18 14:47:01.478 17077-17077/? E/Zygote: isWhitelistProcess - Process is Whitelisted
03-18 14:47:01.479 17077-17077/? E/libpersona: scanKnoxPersonas
    Couldn't open the File - /data/system/users/0/personalist.xml - No such file or directory
03-18 14:47:01.482 17077-17077/? W/SELinux: SELinux selinux_android_compute_policy_index : Policy Index[2],  Con:u:r:zygote:s0 RAM:SEPF_SM-G930K_8.0.0_0018, [-1 -1 -1 -1 0 1]
03-18 14:47:01.483 17077-17077/? I/SELinux: SELinux: seapp_context_lookup: seinfo=untrusted, level=s0:c512,c768, pkgname=com.testRn.servicetest 
03-18 14:47:01.486 17077-17077/? I/zygote: Late-enabling -Xcheck:jni
03-18 14:47:01.534 17077-17077/? D/TimaKeyStoreProvider: TimaKeyStore is not enabled: cannot add TimaSignature Service and generateKeyPair Service
03-18 14:47:01.535 17077-17077/? D/ActivityThread: Added TimaKeyStore provider
03-18 14:47:01.558 17077-17090/? E/zygote: Failed sending reply to debugger: Broken pipe
03-18 14:47:01.559 17077-17090/? I/zygote: Debugger is no longer active
03-18 14:47:01.591 17077-17077/? I/zygote: no shared libraies, dex_files: 1
03-18 14:47:01.738 17077-17077/? V/CameraFragment: Constructor:
03-18 14:47:01.739 17077-17077/? V/CameraFragment: onAttach:
    onCreate:
03-18 14:47:01.744 17077-17077/? V/CameraFragment: onCreateView:
03-18 14:47:01.802 17077-17077/? V/CameraFragment: onResume:
03-18 14:47:01.815 17077-17133/? D/OpenGLRenderer: HWUI GL Pipeline
03-18 14:47:01.832 17077-17077/? D/ViewRootImpl@c2e6e47[MainActivity]: setView = DecorView@748c674[MainActivity] TM=true MM=false
03-18 14:47:01.849 17077-17077/? D/ViewRootImpl@c2e6e47[MainActivity]: dispatchAttachedToWindow
03-18 14:47:01.867 17077-17077/? V/Surface: sf_framedrop debug : 0x4f4c, game : false, logging : 0
03-18 14:47:01.868 17077-17077/? D/ViewRootImpl@c2e6e47[MainActivity]: Relayout returned: old=[0,0][0,0] new=[0,0][1080,1920] result=0x7 surface={valid=true 3517509632} changed=true
03-18 14:47:01.872 17077-17133/? I/OpenGLRenderer: Initialized EGL, version 1.4
03-18 14:47:01.872 17077-17133/? D/OpenGLRenderer: Swap behavior 2
03-18 14:47:01.876 17077-17133/? D/libGLESv1: STS_GLApi : DTS, ODTC are not allowed for Package : com.testRn.servicetest
03-18 14:47:01.876 17077-17133/? D/mali_winsys: EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, egl_color_buffer_format *, EGLBoolean) returns 0x3000,  [1080x1920]-format:1
03-18 14:47:01.876 17077-17133/? D/OpenGLRenderer: eglCreateWindowSurface = 0xd095e028
03-18 14:47:01.881 17077-17077/? D/SurfaceView: BG show() Surface(name=Background for - SurfaceView - com.testRn.servicetest/com.testRn.servicetest.MainActivity@5b436a@0) android.view.SurfaceView{5b436a VFE...C.. ......ID 753,459-1053,759 #7f08002b app:id/camera_view_sub}
03-18 14:47:01.897 17077-17077/? V/Surface: sf_framedrop debug : 0x4f4c, game : false, logging : 0
03-18 14:47:01.897 17077-17077/? D/SurfaceView: surfaceCreated 0 android.view.SurfaceView{5b436a VFE...C.. ......ID 753,459-1053,759 #7f08002b app:id/camera_view_sub}
03-18 14:47:01.898 17077-17077/? D/SurfaceView: surfaceChanged (300,300) 0 android.view.SurfaceView{5b436a VFE...C.. ......ID 753,459-1053,759 #7f08002b app:id/camera_view_sub}
03-18 14:47:01.916 17077-17077/? D/ViewRootImpl@c2e6e47[MainActivity]: Relayout returned: old=[0,0][1080,1920] new=[0,0][1080,1920] result=0x3 surface={valid=true 3517509632} changed=false
03-18 14:47:01.920 17077-17077/? V/Surface: sf_framedrop debug : 0x4f4c, game : false, logging : 0
03-18 14:47:01.920 17077-17077/? V/UVCCameraTextureView: onSurfaceTextureAvailable:android.graphics.SurfaceTexture@7cea9d1
03-18 14:47:01.921 17077-17077/? V/UVCCameraTextureView: RenderThread#getHandler:
03-18 14:47:01.921 17077-17161/? D/UVCCameraTextureView: RenderThread started
03-18 14:47:01.921 17077-17161/? V/UVCCameraTextureView: RenderThread#init:
03-18 14:47:01.921 17077-17161/? V/UVCCameraTextureView: RenderThread#init:
03-18 14:47:01.923 17077-17161/? D/EGLBase14: EGLContext created, client version 3
03-18 14:47:01.924 17077-17161/? V/Surface: sf_framedrop debug : 0x4f4c, game : false, logging : 0
03-18 14:47:01.924 17077-17161/? D/mali_winsys: EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, egl_color_buffer_format *, EGLBoolean) returns 0x3000,  [1056x791]-format:1
03-18 14:47:01.998 17077-17077/? D/ViewRootImpl@c2e6e47[MainActivity]: MSG_RESIZED_REPORT: frame=Rect(0, 0 - 1080, 1920) ci=Rect(0, 72 - 0, 0) vi=Rect(0, 72 - 0, 0) or=1
    MSG_WINDOW_FOCUS_CHANGED 1
03-18 14:47:02.004 17077-17077/? V/InputMethodManager: Starting input: tba=android.view.inputmethod.EditorInfo@3622c36 nm : com.testRn.servicetest ic=null
03-18 14:47:02.004 17077-17077/? I/InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
03-18 14:47:02.020 17077-17077/? V/InputMethodManager: Starting input: tba=android.view.inputmethod.EditorInfo@2097737 nm : com.testRn.servicetest ic=null
03-18 14:47:11.530 17077-17077/com.testRn.servicetest V/CameraFragment: onPause:
[... cut ...]
03-18 14:47:21.496 17077-17317/com.testRn.servicetest V/CameraClient: Constructor:
03-18 14:47:21.500 17077-17452/com.testRn.servicetest V/CameraClientThread: run:
03-18 14:47:21.501 17077-17317/com.testRn.servicetest V/CameraClient: doBindService:
03-18 14:47:21.506 17077-17317/com.testRn.servicetest V/CameraClient: select:device=/dev/bus/usb/001/002
03-18 14:47:21.506 17077-17452/com.testRn.servicetest V/CameraClientThread: handleSelect:
03-18 14:47:21.508 17077-17317/com.testRn.servicetest V/CameraClient: resize(640,480)
    connect:

wikihr, 2019-03-19:
@lvi166 Oh, I was stupid. I had missed one on the doBindService() side. I also changed it and I can see permission dialog. I really thank you for your reply.

closed 2019-03-19 by wikihr

kolyadruz, 2020-11-23:
hi!

> @lvi166 Oh, I was stupid. I had missed one on the doBindService() side. I also changed it and I can see permission dialog. I really thank you for your reply.

Hi! What did you missed? Which line?

wikihr, 2020-11-23:
> hi!
> 
> > @lvi166 Oh, I was stupid. I had missed one on the doBindService() side. I also changed it and I can see permission dialog. I really thank you for your reply.
> 
> Hi! What did you missed? Which line?

Hello, I mean line 161 intent.setPackage("com.serenegiant.usbcameratest4");
in the UVCCamera/usbCameraTest4/src/main/java/com/serenegiant/serviceclient/CameraClient.java

I also have to change setPackage which I remaned.

kolyadruz, 2020-11-23:
> Hello, I mean line 161 intent.setPackage("com.serenegiant.usbcameratest4");
> in the UVCCamera/usbCameraTest4/src/main/java/com/serenegiant/serviceclient/CameraClient.java
> 
> I also have to change setPackage which I renamed.

Thanks a lot for your fast reply! I've changed it, but my USB permission dialog isn't appear. Difference of projects in AndroidX, the new one using it. 

gradle plugin 3.4.0
ndk 14r
commonlib 2.12.4

wikihr, 2020-11-23:
> Thanks a lot for your fast reply! I've changed it, but my USB permission dialog isn't appear. Difference of projects in AndroidX, the new one using it.
>
> gradle plugin 3.4.0
> ndk 14r
> commonlib 2.12.4

I found my miss part with Find in path with whole project. I'm not sure, but you may have missed something! Search with test4.