saki4510t/UVCCamera #455: based on usbCameraTest5,can IFrameCallback run on service

284921310, 2019-03-25:
in usbCameraTest5, I want to let the  IFrameCallback  runing when the mainactivity is background.

i try to not use the setPreviewDisplay function ,and make the related variables in the service, but IFrameCallback  is not working .
is there some methods to realizing this in a short time?
i konw usbCameraTest4 maybe ok,but i think usbCameraTest4 is too complex to realizing this.
thx a lot!

juechenshiai, 2019-03-26:
IFrameCallback need to be displayed，test4 base on aidl，but aidl supports basic types or implement Parcelable，It‘s difficult to achieve

284921310, 2019-03-28:
thx a lot,but try again, i move the related var to the service,the onframe can be run when the mainactivity is background!!!!!!!!