saki4510t/UVCCamera #456: NDK 19 error

Eido95, 2019-03-26:
After cloning the repository, according to [#358 comment](https://github.com/saki4510t/UVCCamera/issues/358#issuecomment-390202111) I replaced:
```
APP_ABI := armeabi armeabi-v7a x86 mips
``` 
with: 
```
APP_ABI := all
```
According to [#181 comment](https://github.com/saki4510t/UVCCamera/issues/181#issuecomment-392642736) I additionally tried to use ```NDK_TOOLCHAIN_VERSION := 4.9``` in each of the following NDK's.
### NDK 19 - didn't work
Errors:
- ```E/UVCCamera: [27355*UVCCamera.cpp:172:connect]:could not open camera:err=-1```
- No video (blank)
- Not possible to use ```NDK_TOOLCHAIN_VERSION := 4.9```:
  - ```/ndk-bundle/build/core/setup-toolchain.mk:52: *** Android NDK: Invalid NDK_TOOLCHAIN_VERSION value: 4.9. GCC is no longer supported. See https://android.googlesource.com/platform/ndk/+/master/docs/ClangMigration.md.    .  Stop.```

### NDK 18 - didn't work
Errors:
- ```E/UVCCamera: [27355*UVCCamera.cpp:172:connect]:could not open camera:err=-1```
- No video (blank)
- Not possible to use ```NDK_TOOLCHAIN_VERSION := 4.9```:
  - ```/ndk-bundle/build/core/setup-toolchain.mk:52: *** Android NDK: Invalid NDK_TOOLCHAIN_VERSION value: 4.9. GCC is no longer supported. See https://android.googlesource.com/platform/ndk/+/master/docs/ClangMigration.md.    .  Stop.```

### NDK 17 - work
Erros:
- ```E/libusb/usbfs: [31358*android_usbfs.c:1776:release_interface]:release interface failed, error -1 errno 22```
- ```E/UVCCamera: [31358*UVCCamera.cpp:172:connect]:could not open camera:err=-99```

When using `NDK_TOOLCHAIN_VERSION := 4.9`, the video **worked**, with the following warning:
```
Android NDK: WARNING: Deprecated NDK_TOOLCHAIN_VERSION 
value: 4.9. GCC is no longer supported and will be removed in the next release. 
See https://android.googlesource.com/platform/ndk/+/master/docs/ClangMigration.md.    
```

tutelarix, 2019-04-24:
@Eido95 Big thanks. It also helped me.

mkrupczak3, 2019-12-10:
👍

mkrupczak3, 2019-12-10:
Shame on the Goog for not supporting GCC

mkrupczak3, 2019-12-10:
For a related issue with a different app, I tried to migrate with [these instructions] (https://android.googlesource.com/platform/ndk/+/master/docs/ClangMigration.md), but encountered a build error

I downloaded v17c from [here](https://developer.android.com/ndk/downloads/older_releases) and everything worked fine I think