saki4510t/UVCCamera #460: Preview isn't displayed

andrey-blak, 2019-04-01:
I've used ndk r14b to build the project. The camera preview isn't displayed. I couldn't find anything suspicious in the logs. 

Project: usbCameraTest0
Device manufacturer: mid
Device model: T7N
Android: 6.0
Camera: SVEN IC-950 HD

Please find the logs attached:
[logs.txt](https://github.com/saki4510t/UVCCamera/files/3030279/logs.txt)

parrotbill, 2019-04-10:
I'm not sure but I think it's problem with frame rate.
When I search your Camera manual, it's frame rate is up to 33 at 1280 × 720 and library's default is 30.

Did you try change setPreviewSize?

andrey-blak, 2019-04-10:
I've tried changing camera.setPreviewSize in MainActivity call to 
```camera.setPreviewSize(640, 480, 1, 33, UVCCamera.FRAME_FORMAT_MJPEG, 1);```
and 
```camera.setPreviewSize(1280, 720, 1, 33, UVCCamera.FRAME_FORMAT_MJPEG, 1);```
It didn't help.

rgm-22, 2020-09-13:
Hi andrey-blak!

Did you solver your issue? I think that I have a similar problem :(

andrey-blak, 2021-01-13:
@rgm-22 No, unfortunately I didn't find any solution.

rgm-22, 2021-01-15:
> @rgm-22 No, unfortunately I didn't find any solution.

Thank you very much for your reply Andrey! 😀
I think that we have to change, compile and modify the library...