saki4510t/UVCCamera #472: Unplug usb camera while it is still in use and application end up with ANR.

chenxi1991, 2019-05-07:
Input dispatching timed out

ANR Input dispatching timed out (Waiting because the touched window's input channel is full. Outbound queue length: 1. Wait queue length: 32.)
1 java.lang.Object.wait!(Native method)
--
2 com.serenegiant.usbcameracommon.AbstractUVCCameraHandler.void stopPreview()(AbstractUVCCameraHandler.java:208)
3 com.serenegiant.usbcameracommon.AbstractUVCCameraHandler.void close()(AbstractUVCCameraHandler.java:175)

xiedong2002, 2019-05-19:
Unluckily , The issue occurs on me . It can easily be reproduced on Android 7.1 with Rockchip 3288. I am trying to debug the issue. It might be caused by the fact that a  lock is not fired since some of outstanding transfer requests are not handled properly ( cancelled or completed). Continue ......

mingtong, 2019-05-21:
You should handle the event when unplug camera and do release things.

closed 2021-05-11 by chenxi1991