saki4510t/UVCCamera #478: uvc_find_device 返回-4， uvc_get_device_with_fd 返回 0

ChineseBoyLY, 2019-05-15:
uvc_find_device(ctx, &dev, 0, 0, NULL); 返回-4 

而用uvc_get_device_with_fd(ctx, &dev, vid, pid, NULL, fd, busnum, devaddr); 就没有问题

请问这是为何？

mingtong, 2019-05-21:
请使用英文.
Please use English to describe your issue, then more people can help you.

jumpy88, 2020-12-30:
Hi @ChineseBoyLY, I'm trying to call uvc_get_device_with_fd but I can't find any documentation explaining how to use it. How should I get fd?
I think the need to use uvc_get_device_with_fd is related to root privileges. Look at this comment in source code https://github.com/saki4510t/UVCCamera/blob/c9399e63dfab4b6d260d8cbef92182abc6de0ee0/libuvccamera/src/main/jni/libuvc/src/device.c#L206