saki4510t/UVCCamera #481: 调用stopPreview(关闭Camera预览).so报错，程序崩溃 项目很着急，求救呀

ChenChao1991, 2019-05-29:
调用代码：
public void stopPreview(Context context) {
if (DEBUG) Log.e(TAG, "stopPreview开始");
removeMessages(MSG_PREVIEW_START);
if (isRecording()) {
stopRecording();
}
if (isPreviewing()) {
final CameraThread thread = mWeakThread.get();
if (thread == null) return;
synchronized (thread.mSync) {
sendEmptyMessage(MSG_PREVIEW_STOP);
if (DEBUG) Log.e(TAG, "MSG_PREVIEW_STOP");
if (!isCameraThread()) {
// wait for actually preview stopped to avoid releasing Surface/SurfaceTexture while preview is still running. therefore this method will take a time to execute
try {
if (DEBUG) Log.e(TAG, "wait(...开始...)");
thread.mSync.wait();
if (DEBUG) Log.e(TAG, "wait(...完成...)");
} catch (InterruptedException e) {
e.printStackTrace();
if (DEBUG) Log.e(TAG, "wait(...InterruptedException...)");
// 恢复中断状态 thread.interrupt();
thread.currentThread().interrupt();
if (DEBUG) Log.e(TAG, e.toString());
Toast.makeText(context, "===" + e.toString() + "===", Toast.LENGTH_LONG).show();
}
}
}
}
if (DEBUG) Log.e(TAG, "stopPreview:完成");
}
报错日志：

Thread-2374(2374) SIGABRT
1 #00 pc 0003c3c4 /system/lib/libc.so (tgkill+12) [armeabi-v7a]
2 #1 pc 00016dad /system/lib/libc.so (pthread_kill+52) [armeabi-v7a]
3 #2 pc 000179bf /system/lib/libc.so (raise+10) [armeabi-v7a]
4 #3 pc 00014165 /system/lib/libc.so (__libc_android_abort+36) [armeabi-v7a]
5 #4 pc 000124ec /system/lib/libc.so (abort+4) [armeabi-v7a]
6 #5 pc 00228837 /system/lib/libart.so (art::Runtime::Abort()+170) [armeabi-v7a]
7 #6 pc 000a6a35 /system/lib/libart.so (art::LogMessage::~LogMessage()+1360) [armeabi-v7a]
8 #7 pc 000b11d7 /system/lib/libart.so (art::JniAbort(char const*, char const*)+1118) [armeabi-v7a]
9 #8 pc 000b1715 /system/lib/libart.so (art::JniAbortF(char const*, char const*, ...)+68) [armeabi-v7a]
10 #9 pc 001acabb /system/lib/libart.so (art::JNI::CallVoidMethodV(JNIEnv*, _jobject*, _jmethodID*, std::_va_list)+598) [armeabi-v7a]
11 #10 pc 00011578 /data/app/com.techmation.hotelrobot-1/lib/arm/libUVCCamera.so (_JNIEnv::CallVoidMethod(_jobject*, _jmethodID*, ...)+52) [armeabi-v5te]
12 #11 pc 00011498 /data/app/com.techmation.hotelrobot-1/lib/arm/libUVCCamera.so (UVCPreview::do_capture_callback(_JNIEnv*, uvc_frame*)+252) [armeabi-v5te]
13 #12 pc 0001113c /data/app/com.techmation.hotelrobot-1/lib/arm/libUVCCamera.so (UVCPreview::do_capture(_JNIEnv*)+144) [armeabi-v5te]
14 #13 pc 00010b7c /data/app/com.techmation.hotelrobot-1/lib/arm/libUVCCamera.so (UVCPreview::capture_thread_func(void*)+64) [armeabi-v5te]
15 #14 pc 0001659b /system/lib/libc.so (__pthread_start(void*)+30) [armeabi-v7a]
16 #15 pc 000144c3 /system/lib/libc.so (__start_thread+6) [armeabi-v7a]
main(1)
1 android.os.MessageQueue.nativePollOnce(Native Method)
2 android.os.MessageQueue.next(MessageQueue.java:143)
3 android.os.Looper.loop(Looper.java:122)
4 android.app.ActivityThread.main(ActivityThread.java:5284)
5 java.lang.reflect.Method.invoke(Native Method)
6 java.lang.reflect.Method.invoke(Method.java:372)
7 com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:963)
8 com.android.internal.os.ZygoteInit.main(ZygoteInit.java:758)

Singleswolf, 2019-06-06:
同样的问题，有找到解决办法吗？

zzJune, 2019-10-31:
请问解决了吗

qwertxyz, 2020-01-20:
解决了吗？！

sysylcf, 2020-02-15:
这个问题有解决了的吗？

ChenChao1991, 2020-02-15:
没有。还是偶尔报错

------------------ 原始邮件 ------------------
发件人: sysylcf <notifications@github.com&gt;
发送时间: 2020年2月15日 08:32
收件人: saki4510t/UVCCamera <UVCCamera@noreply.github.com&gt;
抄送: ChenChao1991 <530183536@qq.com&gt;, Author <author@noreply.github.com&gt;
主题: 回复：[saki4510t/UVCCamera] 调用stopPreview(关闭Camera预览).so报错，程序崩溃 项目很着急，求救呀 (#481)

sysylcf, 2020-02-15:
> 没有。还是偶尔报错
> […](#)
> ------------------ 原始邮件 ------------------ 发件人: sysylcf <notifications@github.com&gt; 发送时间: 2020年2月15日 08:32 收件人: saki4510t/UVCCamera <UVCCamera@noreply.github.com&gt; 抄送: [...]

你们通过android studio  profiler 可以查看实时内存吗？

shengjunhu, 2020-09-24:
修改UVCPreview.cpp#stopPreview(), 加锁即可
 pthread_mutex_lock(&preview_mutex);
 pthread_cond_signal(&preview_sync);
 pthread_mutex_unlock(&preview_mutex);