saki4510t/UVCCamera #491: How to import 'libuvccamera' and 'usbCameraCommon' to new project

icetea0822, 2019-06-26:
I'have put this 
    implementation("com.serenegiant:common:${commonLibVersion}") {
        exclude module: 'support-v4'
    }
and 
     maven { url 'http://raw.github.com/saki4510t/libcommon/master/repository/' }
to my project ,but i can't find module 'libuvccamera' and 'usbCameraCommon' ,how can i put this two modules to my project?
Thanks!

LouisChain, 2019-07-05:
open settings.gradle, add the following 2 lines:

include ':libuvccamera'
include ':usbCameraCommon'

icetea0822, 2019-07-11:
Thanks a lot

closed 2019-07-11 by icetea0822

EjazAmir, 2020-11-05:
hey, i did the same but i got some errors. 
AAPT: error: duplicate value for resource 'attr/orientation' with config ''.
AAPT: error: resource previously defined here.

kindly prvode any solution, im hell tried all things but no luck.