saki4510t/UVCCamera #492: Unable to record video on Android 9.0

bharathnr21, 2019-06-27:
Caused by: android.media.MediaCodec$CodecException: Error 0x80001001
        at android.media.MediaCodec.native_configure(Native Method)
        at android.media.MediaCodec.configure(MediaCodec.java:1960)
        at android.media.MediaCodec.configure(MediaCodec.java:1889)
        at org.uvccamera.mediacodec.MPEG4Encoder.open(MPEG4Encoder.java:138)
        at org.uvccamera.usb.USBMonitor.startRecoder(USBMonitor.java:348) 
        at org.uvccamera.preview.LocalUVCPreviewActivity$6.onClick(LocalUVCPreviewActivity.java:274) 
        at android.view.View.performClick(View.java:7352) 
        at android.view.View.performClickInternal(View.java:7318) 
        at android.view.View.access$3200(View.java:846) 
        at android.view.View$PerformClick.run(View.java:27800) 
        at android.os.Handler.handleCallback(Handler.java:873) 
        at android.os.Handler.dispatchMessage(Handler.java:99) 
        at android.os.Looper.loop(Looper.java:214) 
        at android.app.ActivityThread.main(ActivityThread.java:7032) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:494) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:965)

swaggymiller, 2019-09-05:
hey,I also meet this issue,but i don't know how to solve this. Now you solve this problem ?

bharathnr21, 2019-09-05:
> hey,I also meet this issue,but i don't know how to solve this. Now you solve this problem ?

Remove the project and compile the new project with ndk r14b.