saki4510t/UVCCamera #494: how can i preview fast?

0x7962, 2019-06-30:


0x7962, 2019-06-30:
hi, saki.

Is very slow(low fps) when i preview my USB camera with resolution 4196x3104, but i use other app is ok. like this one [https://play.google.com/store/apps/details?id=com.shenyaocn.android.usbcamera](url) (this app also use your lib) or on the windows/Mac use same camera. 

My cam data is MJPEG, i tried to modify here :
```
diff --git a/libuvccamera/src/main/jni/UVCCamera/UVCPreview.cpp b/libuvccamera/src/main/jni/UVCCamera/UVCPreview.cpp
index 02c1a60..16f9e18 100644
--- a/libuvccamera/src/main/jni/UVCCamera/UVCPreview.cpp
+++ b/libuvccamera/src/main/jni/UVCCamera/UVCPreview.cpp
@@ -530,7 +530,7 @@ void UVCPreview::do_preview(uvc_stream_ctrl_t *ctrl) {
                                frame_mjpeg = waitPreviewFrame();
                                if (LIKELY(frame_mjpeg)) {
                                        frame = get_frame(frame_mjpeg->width * frame_mjpeg->height * 2);
-                                       result = uvc_mjpeg2yuyv(frame_mjpeg, frame);   // MJPEG => yuyv
+                                       result = uvc_mjpeg2rgbx(frame_mjpeg, frame);   // MJPEG => yuyv
                                        recycle_frame(frame_mjpeg);
                                        if (LIKELY(!result)) {
                                                frame = draw_preview_one(frame, &mPreviewWindow, uvc_any2rgbx, 4);
```
but just a little improvement and the preview sometimes flicker.

zipswich, 2022-01-27:
@0x7962 Have you solved the problem? I am facing a similar issue.

1119child, 2025-06-28:
https://blog.csdn.net/weixin_44640701/article/details/148977286?sharetype=blogdetail&sharerId=148977286&sharerefer=PC&sharesource=weixin_44640701&spm=1011.2480.3001.8118
you can see this blog, i solve this problem in my device

zipswich, 2025-06-29:
> https://blog.csdn.net/weixin_44640701/article/details/148977286?sharetype=blogdetail&sharerId=148977286&sharerefer=PC&sharesource=weixin_44640701&spm=1011.2480.3001.8118 you can see this blog, i solve this problem in my device

Thank you. I actually also took a similar approach. What frame rate did you achieve for 1080p?

1119child, 2025-06-30:
> Thank you. I actually also took a similar approach. What frame rate did you achieve for 1080p?

by RK3568, fps is between 28 - 30 FPS