saki4510t/UVCCamera #498: 华为 android 9.0 开启预览后拔掉摄像头出现崩溃

dwf131, 2019-08-03:
`2019-08-03 16:05:08.448 10090-10591/com.serenegiant.usbcameratest8 A/libc: Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0 in tid 10591 (CameraThread), pid 10090 (.usbcameratest8)`
2019-08-03 16:05:08.450 1241-1864/? I/usbhost: gone device /dev/bus/usb/002/021
2019-08-03 16:05:08.451 1241-1864/? D/UsbHostManager: Removed device at deviceAddress = /dev/bus/usb/002/021: MSK Camera
2019-08-03 16:05:08.451 1241-1864/? I/UsbAlsaManager: USB Audio Device Removed: UsbAlsaDevice: [card: 2, device: 0, name: USB-Audio - MSK Camera, hasOutput: false, hasInput: true]
2019-08-03 16:05:08.457 1241-1864/? I/UsbAlsaDevice: INPUT JACK connected: true
2019-08-03 16:05:08.459 1241-1864/? E/HwAudioServiceEx: identifyAudioDevice return device: -1
2019-08-03 16:05:08.459 1241-1864/? I/HwAudioServiceEx: updateTypeCNotify recognizedDevice: -1
2019-08-03 16:05:08.459 1241-1864/? W/ActivityManager: Sending non-protected broadcast huawei.intent.action.hideHiResIcon from system 1241:system/1000 pkg android
2019-08-03 16:05:08.459 1241-1864/? I/HwAudioServiceEx: updateTypeCNotify plug out device: -1
2019-08-03 16:05:08.460 1241-1864/? I/AudioService: setWiredDeviceConnectionState(0 nm: USB-Audio - MSK Camera addr:card=2;device=0;) type: -2147479552 caller: UsbAlsaDevice
2019-08-03 16:05:08.459 1241-1864/? W/ActivityManager: Sending non-protected broadcast huawei.intent.action.hideHiResIcon from system 1241:system/1000 pkg android
2019-08-03 16:05:08.460 1241-1864/? D/AudioService: checkSendBecomingNoisyIntent device:-2147479552 state:0 musicDevice:0
2019-08-03 16:05:08.461 1914-1914/? D/StatusBarIconController: setIconVisibility icon is null or visibility not change visibility=false
2019-08-03 16:05:08.463 1241-1787/? I/AudioService: handle msg:wired device connection
2019-08-03 16:05:08.463 1241-1787/? I/AudioService: onSetWiredDeviceConnectionState(dev:80001000 state:0 address:card=2;device=0; deviceName:USB-Audio - MSK Camera caller: UsbAlsaDevice);
2019-08-03 16:05:08.463 1241-1787/? I/AudioService: handleDeviceConnection(false dev:80001000 address:card=2;device=0; name:USB-Audio - MSK Camera)
2019-08-03 16:05:08.463 1241-1787/? I/AudioService: deviceKey:0x80001000:card=2;device=0;
2019-08-03 16:05:08.463 1241-1787/? I/AudioService: deviceSpec:[type:0x80001000 name:USB-Audio - MSK Camera address:card=2;device=0;] is(already)Connected:true
2019-08-03 16:05:08.463 669-13539/? I/APM_AudioPolicyManager: setDeviceConnectionStateInt() device: 0x80001000, state 0, address card=2;device=0; name USB-Audio - MSK Camera
2019-08-03 16:05:08.465 647-12833/? I/typec_headset_hal: get_typec_usbinfo:usb headset usbid:0x00000000 AP_OR_HIFI:AP ulink_count:0 dlink_count:0 name: devicemask:0x00000000
2019-08-03 16:05:08.465 647-12833/? I/typec_headset_hal: get_typec_usbinfo:dnlink_rate_table[0]:0
2019-08-03 16:05:08.465 647-12833/? I/typec_headset_hal: get_typec_usbinfo:dnlink_rate_table[1]:0
2019-08-03 16:05:08.465 647-12833/? I/typec_headset_hal: get_typec_usbinfo:dnlink_rate_table[2]:0
2019-08-03 16:05:08.465 647-12833/? I/typec_headset_hal: get_typec_usbinfo:dnlink_rate_table[3]:0
2019-08-03 16:05:08.465 647-12833/? I/typec_headset_hal: get_typec_usbinfo:dnlink_rate_table[4]:0
2019-08-03 16:05:08.465 647-12833/? I/typec_headset_hal: get_typec_usbinfo:dnlink_rate_table[5]:0
2019-08-03 16:05:08.465 647-12833/? I/typec_headset_hal: get_typec_usbinfo:dnlink_rate_table[6]:0
2019-08-03 16:05:08.465 647-12833/? I/audio_hw_primary: usb_ext_get_parameters: check change usb device (false)
2019-08-03 16:05:08.465 669-13539/? I/hash_map_utils: key: 'need_change_usb_device' value: ''
2019-08-03 16:05:08.466 647-12833/? I/audio_dp_hal: soc_dsp_get_usbid_cmd: usbid value = 0x00000000, is_vr_glass = 0
2019-08-03 16:05:08.466 647-12833/? I/audio_dp_hal: dp_usb_ext_get_parameters: dp hal change usb device ((null))
2019-08-03 16:05:08.466 669-889/? I/AudioFlinger: setParameters(): io 0, keyvalue card=2;device=0;disconnect=-2147479552, calling pid 669 calling uid 1041
2019-08-03 16:05:08.467 647-12833/? I/audio_hw_primary: soundtrigger_set_enforce voice_wake value 
2019-08-03 16:05:08.467 647-12833/? I/spk_dev_interface: spk_set_parameters: use new smartpakit version
2019-08-03 16:05:08.467 647-12833/? I/audio_hw_primary: update stream types 0
2019-08-03 16:05:08.467 26444-26444/? V/AudioManager: isWiredHeadsetOn...
2019-08-03 16:05:08.470 669-13539/? V/SoundTrigger: setCaptureState(0)
2019-08-03 16:05:08.470 669-13539/? V/SoundTriggerHwService: setCaptureState 0
2019-08-03 16:05:08.470 669-13539/? V/SoundTriggerHwService::Module: Module::setCaptureState_l 0
2019-08-03 16:05:08.471 10090-10090/com.serenegiant.usbcameratest8 D/UsbDeviceConnectionJNI: close
2019-08-03 16:05:08.471 10090-10090/com.serenegiant.usbcameratest8 V/MainActivity: onDisconnect:
2019-08-03 16:05:08.472 1241-1787/? I/AudioService: sendDeviceConnectionIntent(dev:0x80001000 state:0x0 address:card=2;device=0; name:USB-Audio - MSK Camera);
2019-08-03 16:05:08.473 1241-1787/? I/HwAudioService: updateAftPolicy SetModeDeathHandlers is empty, trun off aft
2019-08-03 16:05:08.473 10090-10208/com.serenegiant.usbcameratest8 V/AbsUVCCameraHandler: close:
2019-08-03 16:05:08.473 10090-10208/com.serenegiant.usbcameratest8 V/AbsUVCCameraHandler: stopPreview:
2019-08-03 16:05:08.475 10090-10249/com.serenegiant.usbcameratest8 V/CameraThread: handleStopRecording:mMuxer=null
2019-08-03 16:05:08.475 10090-10249/com.serenegiant.usbcameratest8 V/UVCCameraTextureView: setVideoEncoder:
2019-08-03 16:05:08.475 10090-10249/com.serenegiant.usbcameratest8 V/CameraThread: handleStopPreview:
2019-08-03 16:05:08.476 10090-10291/com.serenegiant.usbcameratest8 V/UVCCameraTextureView: RenderThread#setEncoder:encoder=null
2019-08-03 16:05:08.484 10090-10249/com.serenegiant.usbcameratest8 V/CameraThread: handleStopPreview:finished
2019-08-03 16:05:08.484 10090-10208/com.serenegiant.usbcameratest8 V/AbsUVCCameraHandler: stopPreview:finished
2019-08-03 16:05:08.484 10090-10208/com.serenegiant.usbcameratest8 V/AbsUVCCameraHandler: close:finished
2019-08-03 16:05:08.484 10090-10249/com.serenegiant.usbcameratest8 V/CameraThread: handleClose:
2019-08-03 16:05:08.484 10090-10249/com.serenegiant.usbcameratest8 V/CameraThread: handleStopRecording:mMuxer=null
2019-08-03 16:05:08.484 10090-10249/com.serenegiant.usbcameratest8 V/UVCCameraTextureView: setVideoEncoder:
2019-08-03 16:05:08.487 1241-28180/? V/BroadcastQueue: Finished with ordered broadcast BroadcastRecord{85fedd8 u-1 android.hardware.usb.action.USB_DEVICE_DETACHED}
2019-08-03 16:05:08.490 10090-10291/com.serenegiant.usbcameratest8 V/UVCCameraTextureView: RenderThread#setEncoder:encoder=null
2019-08-03 16:05:08.490 10090-10291/com.serenegiant.usbcameratest8 I/iGraphics: [0030080c] opt app: 0
2019-08-03 16:05:08.532 10617-10617/? I/crash_dump32: obtaining output fd from tombstoned, type: kDebuggerdTombstone
2019-08-03 16:05:08.534 831-831/? I//system/bin/tombstoned: received crash request for pid 10591
2019-08-03 16:05:08.536 10617-10617/? I/crash_dump32: performing dump of process 10090 (target tid = 10591)
2019-08-03 16:05:08.549 2092-2486/? I/hibernation: view add pid:10090 uid:10285
2019-08-03 16:05:08.549 2092-2486/? I/hibernation: app:com.serenegiant.usbcameratest8  has toast view ,pid :10090
2019-08-03 16:05:08.552 10090-10090/com.serenegiant.usbcameratest8 I/ViewRootImpl: jank_removeInvalidNode jank list is null
2019-08-03 16:05:08.557 10617-10617/? A/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
2019-08-03 16:05:08.557 10617-10617/? A/DEBUG: Build fingerprint: 'HUAWEI/HMA-AL00/HWHMA:9/HUAWEIHMA-AL00/200C00R2:user/release-keys'
2019-08-03 16:05:08.557 10617-10617/? A/DEBUG: Revision: '0'
2019-08-03 16:05:08.557 10617-10617/? A/DEBUG: ABI: 'arm'
2019-08-03 16:05:08.557 10617-10617/? A/DEBUG: pid: 10090, tid: 10591, name: CameraThread  >>> com.serenegiant.usbcameratest8 <<<
2019-08-03 16:05:08.557 10617-10617/? A/DEBUG: signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0
2019-08-03 16:05:08.557 10617-10617/? A/DEBUG: Cause: null pointer dereference
2019-08-03 16:05:08.557 10617-10617/? A/DEBUG:     r0  00000000  r1  e3e1efe8  r2  0000ffff  r3  c2b4a9c0
2019-08-03 16:05:08.557 10617-10617/? A/DEBUG:     r4  c2b4a9c0  r5  bf42a800  r6  bf42a820  r7  00000004
2019-08-03 16:05:08.557 10617-10617/? A/DEBUG:     r8  c1a144a0  r9  00000000  r10 bf42a9cc  r11 bdd06858
[... cut ...]
2019-08-03 16:05:08.561 10090-10265/com.serenegiant.usbcameratest8 D/OpenGLRenderer: disableOutlineDraw is true
2019-08-03 16:05:08.574 1241-1252/? V/WindowManager: addWindow: Window{ee1dba2 u0 Toast}
2019-08-03 16:05:08.576 10090-10265/com.serenegiant.usbcameratest8 D/OpenGLRenderer:   HWUI Binary is  enabled
2019-08-03 16:05:08.576 10090-10265/com.serenegiant.usbcameratest8 D/OpenGLRenderer: disableOutlineDraw is true
2019-08-03 16:05:08.587 2092-2486/? I/hibernation: 1241 binder call 1914
2019-08-03 16:05:08.600 671-3474/? E/BufferQueueProducer: [] mConsumerName == NULL!!!!!!

hnlbxb2004, 2019-08-27:
同样遇到这个问题。

Dizner, 2019-08-30:
@hnlbxb2004 @dwf131   看这个 https://blog.csdn.net/vspiders/article/details/90201818  需要修改libusb100.so的代码，然后重新编译

hnlbxb2004, 2019-08-31:
@Dizner  谢谢，最后就是按照你这个修改的。已经解决了。我见已经有人提了PR 了。

hnlbxb2004, 2019-08-31:
我主要就是没搞明白为什么9.0 的系统有这个问题，那一段我没太看懂。为什么会空。

jjjjbababy, 2020-05-05:
> @Dizner 谢谢，最后就是按照你这个修改的。已经解决了。我见已经有人提了PR 了。

有修改好的so吗？

Dizner, 2020-05-18:
@jjjjbababy  抱歉，不太懂c代码，我也只是处理了问题，并没有查看其发生的原因，无法帮到你

Dizner, 2020-05-18:
@hnlbxb2004  抱歉，不太懂c代码，我也只是处理了问题，并没有查看其发生的原因，无法帮到你

Dizner, 2020-05-18:
@jjjjbababy  使用原工程，改一下这段代码，然后ndk编译就行了，时间久远，我这边也没有编译好的so文件了，抱歉