saki4510t/UVCCamera #501: How can i connect camera without CameraDialog ?

Sattayapong, 2019-08-13:
I try module usbCameraTest0 to usbCameraTest8 every module has using CameraDialog for Preview Camera.

I want to auto-connect camera after activity onCreate, but I can't find a way to do it.

Please give me an advice concerning what to do.

thankyou.

gxshao, 2019-08-26:
You can build you own `UVCCamera.aar` by `./gradlew build`. after building, you will find libuvccamera.aar under `UVCCamara/libuvccamera/build/outputs/aar/`

Sattayapong, 2019-08-26:
thank you for your advice, I did it but I forgot to close.

closed 2019-08-26 by Sattayapong

EjazAmir, 2020-11-02:
> thank you for your advice, I did it but I forgot to close.

hey, can you share the piece of code how you achieve this.

EjazAmir, 2020-11-02:
> You can build your own `UVCCamera.aar` by `./gradlew build`. after building, you will find libuvccamera.aar under `UVCCamara/libuvccamera/build/outputs/aar/`
> 
> And include it into your project.

hey, can you share the piece of code how you achieve this.