saki4510t/UVCCamera #505: 高端手机花屏问题，低端手机没有问题；不清楚是不是跟Android9.0有关，谁遇到过？

DeboZhao, 2019-08-20:


libanghui, 2019-08-23:
兄弟  解决了吗 我这边vivo y7s也是这个问题

gxshao, 2019-08-26:
贴出来看看啊，用的什么镜头啊。。

DeboZhao, 2019-08-27:
> 兄弟 解决了吗 我这边vivo y7s也是这个问题

只有一点儿线索，测试了几个手机，发现getUsbVersion = 2.1的手机没问题，getUsbVersion=3.0的手机会花屏，目前还不知道怎么解决？老兄如果有解决方案，辛苦同步一下。。

DeboZhao, 2019-08-27:
> 贴出来看看啊，用的什么镜头啊。。

用的goPro，中间使用了一个AVer Media的转接口，好像和设备没有关系，跟手机的USB协议版本又关系，测试了几个手机，发现getUsbVersion = 2.1的手机没问题，getUsbVersion=3.0的手机会花屏，目前还不知道怎么解决？ 要替换libUSB库吗？？工程中的libUSB有修改过源码，对libUSB不熟，不清楚替换的代价有多大？

gxshao, 2019-08-29:
> > 贴出来看看啊，用的什么镜头啊。。
> 
> 用的goPro，中间使用了一个AVer Media的转接口，好像和设备没有关系，跟手机的USB协议版本又关系，测试了几个手机，发现getUsbVersion = 2.1的手机没问题，getUsbVersion=3.0的手机会花屏，目前还不知道怎么解决？ 要替换libUSB库吗？？工程中的libUSB有修改过源码，对libUSB不熟，不清楚替换的代价有多大？

没遇到过，不过要看看2.1和3.0的 usbfileDescriptor 有啥区别。在libusb里

libanghui, 2019-10-22:
楼组 问题解决了吗

libanghui, 2019-11-07:
兄弟 解决了吗  告诉下方法

GQii, 2019-12-12:
楼主，8.1版本，usbversion=2.1，也花屏。