saki4510t/UVCCamera #508: NDK build failed

SkywingWang, 2019-08-28:
> Configure project :libuvccamera
executing ndkBuild
executing ndkBuild clean

> Task :libuvccamera:ndkClean FAILED
make: Entering directory `/Users/SkywingWang/NNGWork/GitLab/cockpit_meter/libuvccamera/src/main'
obj/local/armeabi-v7a/objs/UVCCamera/Parameters.o.d:1: *** multiple target patterns.  Stop.
make: Leaving directory `/Users/SkywingWang/NNGWork/GitLab/cockpit_meter/libuvccamera/src/main'

FAILURE: Build failed with an exception.

Mac OS  
NDK 14b
cimpileSdkVersion 28
targetSdkVersion 28
please help

gxshao, 2019-08-29:
检查*.mk文件

alphaDroid89, 2021-04-30:
Hi @SkywingWang is there is any solution for this. When I tried to run my application I get the same above-mentioned issue. if you have any workaround please share with me. That will help me a lot.

@gxshao , what should i need to check in MK file? can you please elaborate? That will help me to resolve the blocker.

luxunchina, 2022-07-29:
NDK: r14b

1、Delete all files in this folder: src/main/obj

2、Run build->clean project

3、Make project