saki4510t/UVCCamera #512: stopPreview概率性奔溃

yzq666, 2019-09-10:
JNI DETECTED ERROR IN APPLICATION: jmethodID was NULL in call to CallVoidMethodV
art/runtime/runtime.cc:415]   at com.serenegiant.usb.UVCCamera.nativeStopPreview(Native method)
    art/runtime/runtime.cc:415]   at com.serenegiant.usb.UVCCamera.stopPreview(UVCCamera.java:402)
    art/runtime/runtime.cc:415]   - locked <0x01c36057> (a com.serenegiant.usb.UVCCamera)
    art/runtime/runtime.cc:415]   at com.lgh.uvccamera.UVCCameraProxy.stopPreview(UVCCameraProxy.java:421)
    art/runtime/runtime.cc:415]   at com.lgh.uvccamera.UVCCameraProxy$2.onSurfaceTextureDestroyed(UVCCameraProxy.java:216)
    art/runtime/runtime.cc:415]   at android.view.TextureView.destroySurface(TextureView.java:242)

zzJune, 2019-10-31:
我也是，怎么办

sysylcf, 2020-02-15:
> 我也是，怎么办

这个问题有解决掉吗？

lliecho, 2020-03-18:
> > 我也是，怎么办
> 
> 这个问题有解决掉吗？

请问有解决掉吗

jjjjbababy, 2020-04-30:
JNI DETECTED ERROR IN APPLICATION: obj == null
 java_vm_ext.cc:542]     in call to CallVoidMethodV
java_vm_ext.cc:542] "Thread-16" prio=5 tid=14 Runnable
 java_vm_ext.cc:542]   | group="main" sCount=0 dsCount=0 flags=0 obj=0x12f80140 self=0xe4288600
java_vm_ext.cc:542]   | sysTid=29872 nice=0 cgrp=default sched=0/0 handle=0xc1a83970
 java_vm_ext.cc:542]   | state=R schedstat=( 50160416 1144274 97 ) utm=3 stm=1 core=5 HZ=100
java_vm_ext.cc:542]   | stack=0xc1988000-0xc198a000 stackSize=1010KB
 java_vm_ext.cc:542]   | held mutexes= "mutator lock"(shared held)
java_vm_ext.cc:542]   native: #00 pc 002e7713  /system/lib/libart.so (art::DumpNativeStack(std::__1::basic_ostream<char, std::__1::char_traits<char>>&, int, BacktraceMap*, char const*, art::ArtMethod*, void*, bool)+134)
java_vm_ext.cc:542]   native: #01 pc 0038d135  /system/lib/libart.so (art::Thread::DumpStack(std::__1::basic_ostream<char, std::__1::char_traits<char>>&, bool, BacktraceMap*, bool) const+212)
java_vm_ext.cc:542]   native: #02 pc 00389713  /system/lib/libart.so (art::Thread::Dump(std::__1::basic_ostream<char, std::__1::char_traits<char>>&, bool, BacktraceMap*, bool) const+34)
java_vm_ext.cc:542]   native: #03 pc 0023a433  /system/lib/libart.so (art::JavaVMExt::JniAbort(char const*, char const*)+722)
java_vm_ext.cc:542]   native: #04 pc 000114f0  /data/app/cn.com.sinoup.mfr-mWXPgIIvsx7RMTO4sqrp7g==/lib/arm/libUVCCamera.so (_JNIEnv::CallVoidMethod(_jobject*, _jmethodID*, ...)+52)
java_vm_ext.cc:542]   native: #05 pc 0001140c  /data/app/cn.com.sinoup.mfr-mWXPgIIvsx7RMTO4sqrp7g==/lib/arm/libUVCCamera.so (UVCPreview::do_capture_callback(_JNIEnv*, uvc_frame*)+244)
 java_vm_ext.cc:542]   native: #06 pc 000110bc  /data/app/cn.com.sinoup.mfr-mWXPgIIvsx7RMTO4sqrp7g==/lib/arm/libUVCCamera.so (UVCPreview::do_capture(_JNIEnv*)+144)
04-30 09:26:17.047 28779 29872 F .com.sinoup.mf: java_vm_ext.cc:542]   native: #07 pc 00010b04  /data/app/cn.com.sinoup.mfr-mWXPgIIvsx7RMTO4sqrp7g==/lib/arm/libUVCCamera.so (UVCPreview::capture_thread_func(void*)+64)
 java_vm_ext.cc:542]   native: #08 pc 00065413  /system/lib/libc.so (__pthread_start(void*)+22)
 java_vm_ext.cc:542]   native: #09 pc 0001e819  /system/lib/libc.so (__start_thread+32)
java_vm_ext.cc:542]   (no managed stack frames)
java_vm_ext.cc:542] 
runtime.cc:565] Runtime aborting...
runtime.cc:565] Dumping all threads without appropriate locks held: thread list lock mutator lock

我是这个报错

jjjjbababy, 2020-04-30:
JNI DETECTED ERROR IN APPLICATION: obj == null
 java_vm_ext.cc:542]     in call to CallVoidMethodV
java_vm_ext.cc:542] "Thread-16" prio=5 tid=14 Runnable
 java_vm_ext.cc:542]   | group="main" sCount=0 dsCount=0 flags=0 obj=0x12f80140 self=0xe4288600
java_vm_ext.cc:542]   | sysTid=29872 nice=0 cgrp=default sched=0/0 handle=0xc1a83970
 java_vm_ext.cc:542]   | state=R schedstat=( 50160416 1144274 97 ) utm=3 stm=1 core=5 HZ=100
java_vm_ext.cc:542]   | stack=0xc1988000-0xc198a000 stackSize=1010KB
 java_vm_ext.cc:542]   | held mutexes= "mutator lock"(shared held)
java_vm_ext.cc:542]   native: #00 pc 002e7713  /system/lib/libart.so (art::DumpNativeStack(std::__1::basic_ostream<char, std::__1::char_traits<char>>&, int, BacktraceMap*, char const*, art::ArtMethod*, void*, bool)+134)
java_vm_ext.cc:542]   native: #01 pc 0038d135  /system/lib/libart.so (art::Thread::DumpStack(std::__1::basic_ostream<char, std::__1::char_traits<char>>&, bool, BacktraceMap*, bool) const+212)
java_vm_ext.cc:542]   native: #02 pc 00389713  /system/lib/libart.so (art::Thread::Dump(std::__1::basic_ostream<char, std::__1::char_traits<char>>&, bool, BacktraceMap*, bool) const+34)
java_vm_ext.cc:542]   native: #03 pc 0023a433  /system/lib/libart.so (art::JavaVMExt::JniAbort(char const*, char const*)+722)
java_vm_ext.cc:542]   native: #04 pc 000114f0  /data/app/cn.com.sinoup.mfr-mWXPgIIvsx7RMTO4sqrp7g==/lib/arm/libUVCCamera.so (_JNIEnv::CallVoidMethod(_jobject*, _jmethodID*, ...)+52)

我是这个报错

shengjunhu, 2020-07-21:
> JNI DETECTED ERROR IN APPLICATION: obj == null
> java_vm_ext.cc:542] in call to CallVoidMethodV
> java_vm_ext.cc:542] "Thread-16" prio=5 tid=14 Runnable
> java_vm_ext.cc:542] | group="main" sCount=0 dsCount=0 flags=0 obj=0x12f80140 self=0xe4288600
> java_vm_ext.cc:542] | sysTid=29872 nice=0 cgrp=default sched=0/0 handle=0xc1a83970
> java_vm_ext.cc:542] | state=R schedstat=( 50160416 1144274 97 ) utm=3 stm=1 core=5 HZ=100
> java_vm_ext.cc:542] | stack=0xc1988000-0xc198a000 stackSize=1010KB
> java_vm_ext.cc:542] | held mutexes= "mutator lock"(shared held)
> java_vm_ext.cc:542] native: #00 pc 002e7713 /system/lib/libart.so (art::DumpNativeStack(std::__1::basic_ostream<char, std::__1::char_traits>&, int, BacktraceMap*, char const*, art::ArtMethod*, void*, bool)+134)
> java_vm_ext.cc:542] native: #1 pc 0038d135 /system/lib/libart.so (art::Thread::DumpStack(std::__1::basic_ostream<char, std::__1::char_traits>&, bool, BacktraceMap*, bool) const+212)
> java_vm_ext.cc:542] native: #2 pc 00389713 /system/lib/libart.so (art::Thread::Dump(std::__1::basic_ostream<char, std::__1::char_traits>&, bool, BacktraceMap*, bool) const+34)
> java_vm_ext.cc:542] native: #3 pc 0023a433 /system/lib/libart.so (art::JavaVMExt::JniAbort(char const*, char const*)+722)
> java_vm_ext.cc:542] native: #4 pc 000114f0 /data/app/cn.com.sinoup.mfr-mWXPgIIvsx7RMTO4sqrp7g==/lib/arm/libUVCCamera.so (_JNIEnv::CallVoidMethod(_jobject*, _jmethodID*, ...)+52)
> java_vm_ext.cc:542] native: #5 pc 0001140c /data/app/cn.com.sinoup.mfr-mWXPgIIvsx7RMTO4sqrp7g==/lib/arm/libUVCCamera.so (UVCPreview::do_capture_callback(_JNIEnv*, uvc_frame*)+244)
> java_vm_ext.cc:542] native: #6 pc 000110bc /data/app/cn.com.sinoup.mfr-mWXPgIIvsx7RMTO4sqrp7g==/lib/arm/libUVCCamera.so (UVCPreview::do_capture(_JNIEnv*)+144)
> 04-30 09:26:17.047 28779 29872 F .com.sinoup.mf: java_vm_ext.cc:542] native: #7 pc 00010b04 /data/app/cn.com.sinoup.mfr-mWXPgIIvsx7RMTO4sqrp7g==/lib/arm/libUVCCamera.so (UVCPreview::capture_thread_func(void*)+64)
> java_vm_ext.cc:542] native: #8 pc 00065413 /system/lib/libc.so (__pthread_start(void*)+22)
> java_vm_ext.cc:542] native: #9 pc 0001e819 /system/lib/libc.so (__start_thread+32)
> java_vm_ext.cc:542] (no managed stack frames)
> java_vm_ext.cc:542]
> runtime.cc:565] Runtime aborting...
> runtime.cc:565] Dumping all threads without appropriate locks held: thread list lock mutator lock
> 
> 我是这个报错

这是stopview时把setFrameCallback()置空，导致jni层还在使用报空指针，在UVCPreview.cpp中对回调加空判断即可
/**
* call IFrameCallback#onFrame if needs
 */
void UVCPreview::do_capture_callback(JNIEnv *env, uvc_frame_t *frame) {
	ENTER();

	if (LIKELY(frame)) {
		uvc_frame_t *callback_frame = frame;
		if (mFrameCallbackObj) {
			if (mFrameCallbackFunc) {
				callback_frame = get_frame(callbackPixelBytes);
				if (LIKELY(callback_frame)) {
					int b = mFrameCallbackFunc(frame, callback_frame);
					recycle_frame(frame);
					if (UNLIKELY(b)) {
						LOGW("failed to convert for callback frame");
						goto SKIP;
					}
				} else {
					LOGW("failed to allocate for callback frame");
					callback_frame = frame;
					goto SKIP;
				}
			}
			jobject buf = env->NewDirectByteBuffer(callback_frame->data, callbackPixelBytes);
			//Add by shengjunhu for nullpointer
			if(iframecallback_fields.onFrame) {
			    env->CallVoidMethod(mFrameCallbackObj, iframecallback_fields.onFrame, buf);
			}
			env->ExceptionClear();
			env->DeleteLocalRef(buf);
		}
 SKIP:
		recycle_frame(callback_frame);
	}
	EXIT();
}

febflyer, 2020-12-28:
同问，找到解决办法没？

febflyer, 2020-12-31:
试了，崩溃问题确实有效，但是还有另一个导致崩溃的问题：内存一直增长，目前还没找到有效的办法

LucasMoskun, 2021-04-27:
Could it e a race condition issue -- trying to synchronize the threads at java level? https://stackoverflow.com/questions/10692980/android-synchronization-of-native-thread-with-main-thread

leach-chen, 2021-07-03:
这里也有篇帖子讲了
https://blog.csdn.net/hanye2020/article/details/111741630

jiaozebo, 2022-01-11:
try this repo:
https://github.com/tsinglink/uvccamera
test with this apk:
https://github.com/tsinglink/uvccamera/releases/download/1.0/simpleuvccamera-release.apk

yl84321, 2024-01-18:
> native层是异步模式，线程比较多，stopPreview的时候会等待capture线程结束。可能capture线程根本就没有启动或者在其他地方被阻塞了，导致加入崩溃无法或者返回。 我改成了同步模式了，可以试一下尝试 这个repo： https: [//github.com/tsinglink/uvccamera](https://github.com/tsinglink/uvccamera) 使用这个apk进行测试： https://github.com/tsinglink/uvccamera/releases/download/1.0/simpleuvccamera-release.apk

您好，能看下您是如何修改的吗，这个连接访问不了

Clark-caipeiyuan, 2024-01-18:
收到了，谢谢！

jenny850113, 2024-10-28:
> native 层是异步模式,线程比较多,stopPreview的时候会等待capture线程结束.可能capture线程根本就没有启动或者被阻在其它地方了,导致join崩溃或者无法返回. 我改成同步模式了,可以试试 try this repo: https://github.com/tsinglink/uvccamera test with this apk: https://github.com/tsinglink/uvccamera/releases/download/1.0/simpleuvccamera-release.apk

您好, 請問可否看下如何修改的,link失效了,謝謝!