saki4510t/UVCCamera #517: Why is my native heap increasing?

alless1, 2019-09-29:
As I kept turning on and off the camera, I dumpsys meminfo showed that Native Heap was rising and could not be released. Not even calling GC. I'm sure I've completely freed up resources at the Java layer.
Please help me, thank you.
@saki4510t

yzq666, 2019-09-30:
I also encountered your problem. Thank you for your test. If you have any solution, please contact me.

yzq666, 2019-09-30:
yuanzuqiang@cvte.com

alless1, 2019-09-30:
It's because I performed a mCamera.destroy() in the onFrame method; the native heap cannot be freed. It's amazing.

closed 2019-09-30 by alless1

alless1, 2019-09-30:
> I also encountered your problem. Thank you for your test. If you have any solution, please contact me.
> [846384016@qq.com](mailto:846384016@qq.com)
Do not release resources in callback threads