saki4510t/UVCCamera #524: mUVCCameraView 组件大小更改后surface不更新

JiaHub, 2019-11-05:
我在小屏预览视频，但是全屏后，视频就停止在某一帧。有没有同样遇到这问题的前辈。江湖救急！
全屏实现是通过设置其他组件不可见实现的。

bin318, 2019-11-06:
com.serenegiant:common:2.12.4 我下载不下来，你可以吗？

JiaHub, 2019-11-07:
你需要在项目级的build.gradle 文件的allprojects 标签下添加代码仓库；
allprojects {
    repositories {
        maven { url 'https://maven.google.com' }
        maven { url 'http://raw.github.com/saki4510t/libcommon/master/repository/' }
        maven { url "https://jitpack.io" }
        jcenter()
    }
}
还有希望你可以提供详细的报错！
注意从git仓库添加依赖，需要添加对应的代码仓库。

bin318, 2019-11-07:
maven加了的
http://raw.github.com/saki4510t/libcommon/master/repository/
就是单纯的这个仓库访问不了了
你可以把你下载的aar包发我吗？
万分感谢！我的qq 281525288

chenxiaofei910714, 2021-12-24:
> 我在小屏预览视频，但全屏后，视频就停止在某一瞬间。有没有遇到这问题的前辈。湖救急！ 全屏实现是通过设置其他组件无法看到的。

你的全屏是通过什么方法设置的