saki4510t/UVCCamera #528: 不停的关闭和开启摄像头，运行一段时间后程序就会崩溃

hugoleong, 2019-11-11:
间隔3秒钟，不停的开启和关闭摄像头，大概运行40分钟左右就会出现崩溃，大概是内存不足引起的，但这是uvccamera自身的错误，请问有没有遇到过这种情况的朋友
调用代码：

打开摄像头的接口
 public boolean startCam()
    {
        mUSBMonitor =new USBMonitor(mContext,mOnDeviceConnectListener);
        mUSBMonitor.register();
        filter = DeviceFilter.getDeviceFilters(mContext, device_filter_id);
        if (filter.size() > 0){
            item = mUSBMonitor.getDeviceList(filter.get(0));
            if (item.size() > 0){
                for(int i=0; i< item.size(); i++)
                {
                    for (int c=0; c<8; c++)
                    {
                        if (item.get(i).getVendorId() == arrayCamPro[c].vid && item.get(i).getProductId() == arrayCamPro[c].pid)
                        {
                            mUSBMonitor.requestPermission(item.get(i));
                            if (!mUSBMonitor.hasPermission(item.get(i))){
                                OpenUSBCamThread.start();
                                break;
                            }
                        }
                    }
                }
            }else {
                //callback.openFail();
                return false;
            }
        }else {
            //callback.openFail();
            return false;
        }
        return true;
    }

关闭摄像头的代码：
public void closeCam()
    {
        if (mUVCCamera != null) {
            mUVCCamera.stopPreview();
            mUVCCamera.destroy();
            mUVCCamera = null;
        }
        if (mUSBMonitor != null) {
            mUSBMonitor.unregister();
            mUSBMonitor.destroy();
            mUSBMonitor = null;
        }
    }

崩溃日志如下：

I/libUVCCamera: [27796UVCPreview.cpp:271:callbackPixelFormatChanged]:PIXEL_FORMAT_RGBX:
I/libUVCCamera: [20800UVCPreview.cpp:492:prepare_preview]:frameSize=(1200,800)@Mjpeg
I/libUVCCamera: [20802UVCPreview.cpp:271:callbackPixelFormatChanged]:PIXEL_FORMAT_RGBX:
W/libUVCCamera: [20801UVCPreview.cpp:119:get_frame]:allocate new frame
W/libUVCCamera: [20800UVCPreview.cpp:119:get_frame]:allocate new frame
W/libUVCCamera: [20801UVCPreview.cpp:119:get_frame]:allocate new frame
W/libUVCCamera: [20800UVCPreview.cpp:119:get_frame]:allocate new frame
E/[Gralloc-ERROR]: int gralloc_backend_register(private_handle_t):82 mmap( share_fd:485 ) failed with Out of memory
W/GraphicBufferMapper: registerBuffer(0xb77faee8) failed -12 (Out of memory)
E/GraphicBuffer: unflatten: registerBuffer failed: Out of memory (-12)
E/[Gralloc-ERROR]: int gralloc_lock(const gralloc_module_t*, buffer_handle_t, int, int, int, int, int, void**):193 Locking in
W/GraphicBufferMapper: lockAsync(...) failed -22 (Invalid argument)
W/Surface: failed locking buffer (handle = 0x0)
W/libUVCCamera: [20801UVCPreview.cpp:119:get_frame]:allocate new frame
E/libUVCCamera: [20800UVCPreview.cpp:647:draw_preview_one]:failed converting
E/libUVCCamera: [20800UVCPreview.cpp:647:draw_preview_one]:failed converting
[... 314 lines cut ...]
I/libUVCCamera: [27796UVCPreview.cpp:271:callbackPixelFormatChanged]:PIXEL_FORMAT_RGBX:
I/libUVCCamera: [20904UVCPreview.cpp:492:prepare_preview]:frameSize=(1200,800)@Mjpeg
I/libUVCCamera: [20906UVCPreview.cpp:271:callbackPixelFormatChanged]:PIXEL_FORMAT_RGBX:
A/libc: Fatal signal 11 (SIGSEGV), code 1, fault addr 0x0 in tid 20902 (pool-1-thread-1)
I/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
I/DEBUG: Build fingerprint: 'Android/rk3288/rk3288:5.1.1/LMY49F/hwit:userdebug/test-keys'
I/DEBUG: Revision: '0'
I/DEBUG: ABI: 'arm'
I/DEBUG: pid: 27753, tid: 20902, name: pool-1-thread-1 >>> com.example.test<<<
I/DEBUG: signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0
I/DEBUG: r0 00000000 r1 a0af2854 r2 00000210 r3 00000000
I/DEBUG: r4 0000029c r5 00000001 r6 a0af2809 r7 00000290
I/DEBUG: r8 b7b68990 r9 b77aff60 sl a0af2808 fp 9fda2ab8
I/DEBUG: ip b6ded648 sp 9fda2a70 lr a0d67fcc pc b6da89f0 cpsr 200f0010
I/DEBUG: backtrace:
I/DEBUG: #00 pc 000129f0 /system/lib/libc.so (__memcpy_base+104)
I/DEBUG: #1 pc 00014fc8 /data/app/com.example.test-1/lib/arm/libuvc.so
I/DEBUG: #2 pc 0000c314 /data/app/com.example.test-1/lib/arm/libusb100.so (usbi_handle_transfer_completion+320)
I/DEBUG: #3 pc 000126a8 /data/app/com.example.test-1/lib/arm/libusb100.so
I/DEBUG: #4 pc 0000d200 /data/app/com.example.test-1/lib/arm/libusb100.so
I/DEBUG: #5 pc 0000cb60 /data/app/com.example.test-1/lib/arm/libusb100.so (libusb_handle_events_timeout_completed+968)
I/DEBUG: #6 pc 0000d3cc /data/app/com.example.test-1/lib/arm/libusb100.so (libusb_handle_events+52)
I/DEBUG: #7 pc 00012c20 /data/app/com.example.test-1/lib/arm/libuvc.so (_uvc_handle_events+136)
I/DEBUG: #8 pc 0001659b /system/lib/libc.so (__pthread_start(void)+30)
I/DEBUG: #9 pc 000144c3 /system/lib/libc.so (__start_thread+6)

hugoleong, 2019-11-14:
增加了点崩溃日志
11-14 10:35:52.934 2810-10730/com.example.paycamlivedetect A/libc: Fatal signal 11 (SIGSEGV), code 1, fault addr 0x0 in tid 10730 (pool-1-thread-1)
11-14 10:35:52.985 184-184/? I/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
11-14 10:35:52.985 184-184/? I/DEBUG: Build fingerprint: 'Android/rk3288/rk3288:5.1.1/LMY49F/hwit:userdebug/test-keys'
11-14 10:35:52.985 184-184/? I/DEBUG: Revision: '0'
11-14 10:35:52.985 184-184/? I/DEBUG: ABI: 'arm'
11-14 10:35:52.986 184-184/? I/DEBUG: pid: 2810, tid: 10730, name: pool-1-thread-1  >>> com.example.paycamlivedetect <<<
11-14 10:35:52.986 184-184/? I/DEBUG: signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0
11-14 10:35:53.000 184-184/? I/DEBUG:     r0 00000000  r1 b91265e4  r2 00000210  r3 00000000
11-14 10:35:53.000 184-184/? I/DEBUG:     r4 0000029c  r5 00000001  r6 b9126598  r7 00000290
11-14 10:35:53.000 184-184/? I/DEBUG:     r8 b8d22c30  r9 b8d222c8  sl b9126599  fp 9f51ead8
11-14 10:35:53.000 184-184/? I/DEBUG:     ip b6e33648  sp 9f51ea88  lr a0eac560  pc b6dee9f0  cpsr 200f0010
11-14 10:35:53.001 184-184/? I/DEBUG: backtrace:
11-14 10:35:53.001 184-184/? I/DEBUG:     #00 pc 000129f0  /system/lib/libc.so (__memcpy_base+104)
11-14 10:35:53.001 184-184/? I/DEBUG:     #01 pc 0001455c  /data/app/com.example.paycamlivedetect-1/lib/arm/libuvc.so
11-14 10:35:53.001 184-184/? I/DEBUG:     #02 pc 0000b03c  /data/app/com.example.paycamlivedetect-1/lib/arm/libusb100.so (usbi_handle_transfer_completion+324)
11-14 10:35:53.001 184-184/? I/DEBUG:     #03 pc 00010a20  /data/app/com.example.paycamlivedetect-1/lib/arm/libusb100.so
11-14 10:35:53.001 184-184/? I/DEBUG:     #04 pc 0000bcac  /data/app/com.example.paycamlivedetect-1/lib/arm/libusb100.so
11-14 10:35:53.001 184-184/? I/DEBUG:     #05 pc 0000b70c  /data/app/com.example.paycamlivedetect-1/lib/arm/libusb100.so (libusb_handle_events_timeout_completed+540)
11-14 10:35:53.001 184-184/? I/DEBUG:     #06 pc 0000be78  /data/app/com.example.paycamlivedetect-1/lib/arm/libusb100.so (libusb_handle_events+52)
11-14 10:35:53.001 184-184/? I/DEBUG:     #07 pc 000124f4  /data/app/com.example.paycamlivedetect-1/lib/arm/libuvc.so (_uvc_handle_events+136)
11-14 10:35:53.001 184-184/? I/DEBUG:     #08 pc 0001659b  /system/lib/libc.so (__pthread_start(void*)+30)
11-14 10:35:53.001 184-184/? I/DEBUG:     #09 pc 000144c3  /system/lib/libc.so (__start_thread+6)
11-14 10:35:54.992 504-626/system_process E/NativeCrashListener: Exception dealing with report
                                                                 android.system.ErrnoException: read failed: EAGAIN (Try again)
                                                                     at libcore.io.Posix.readBytes(Native Method)
                                                                     at libcore.io.Posix.read(Posix.java:165)
                                                                     at libcore.io.BlockGuardOs.read(BlockGuardOs.java:230)
                                                                     at android.system.Os.read(Os.java:350)
                                                                     at com.android.server.am.NativeCrashListener.consumeNativeCrashData(NativeCrashListener.java:240)
                                                                     at com.android.server.am.NativeCrashListener.run(NativeCrashListener.java:138)

hugoleong, 2019-11-15:
UVCcamera的内存泄漏问题，以前有朋友遇到过，看到你（xiaozhi003 ）的帖子，完美解决了我的问题，感谢！！！！

修改UVCPreview.cpp 如下即可
void UVCPreview::addCaptureFrame(uvc_frame_t *frame) {
	pthread_mutex_lock(&capture_mutex);
	if (LIKELY(isRunning())) {
		// keep only latest one
		if (captureQueu) {
			recycle_frame(captureQueu);
		}
		captureQueu = frame;
		pthread_cond_broadcast(&capture_sync);
	}
    else
    {
        recycle_frame(frame);
    }
	pthread_mutex_unlock(&capture_mutex);
}

closed 2019-11-15 by hugoleong

leach-chen, 2021-07-03:
多谢，最近也在用uvccamera，我试试。另外还会概率性出现闪退的问题，我也找不到一篇帖子。

gotoorder, 2021-10-12:
这样改了也还是会