saki4510t/UVCCamera #533: stream.c bulk mode

jskimn, 2019-12-10:
When UVC is bulk mode, I got fail from "libusb_submit_transfer" in uvc_stream_start_bandwidth.
Is there anybody using bulk mode?

tangqingcai, 2020-04-21:
i am developing a project used bulk mode transfer ;
and i found it could not work too;
someone said it should update libuvc to the lastest version;

jskimn, 2020-04-23:
yes,
after updating libuvc, I could play bulk mode with no problem.

dazzhe, 2020-06-10:
@jskimn 
How did you update libuvc?
I replaced libuvc directory with a master branch of [libuvc](https://github.com/libuvc/libuvc) but I got many errors.
This project modifies many parts of libuvc so that I guess simple replacement won't work.

kros235, 2020-08-05:
@jskimn May I ask how did you update libuvc?
I encountered exactly the same issue
(got fail from "libusb_submit_transfer" in uvc_stream_start_bandwidth.) 
and stuck, facing this issue more than 2 months now....
just replacing libuvc doesn't work and causing massive errors

Please share some tips if possible...:S