saki4510t/UVCCamera #534: 拔掉摄像后崩溃

along1991, 2019-12-12:
摄像头插拔几次后,直接崩溃,预览画面录制都没有问题,这个是什么情况

2019-11-29 16:36:15.309 8138-8202/com.wr.medical I/[Gralloc]: leave: prime_fd=108,share_attr_fd=110
2019-11-29 16:36:24.183 8138-8201/com.wr.medical A/libc: Fatal signal 7 (SIGBUS), code 2, fault addr 0x7199000000 in tid 8201 (CameraThread)
2019-11-29 16:36:24.221 8205-8205/? A/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
2019-11-29 16:36:24.221 8205-8205/? A/DEBUG: Build fingerprint: 'Android/rk3399_all/rk3399_all:7.1.2/NHG47K/zhangj09121201:userdebug/test-keys'
2019-11-29 16:36:24.221 8205-8205/? A/DEBUG: Revision: '0'
2019-11-29 16:36:24.221 8205-8205/? A/DEBUG: ABI: 'arm64'
2019-11-29 16:36:24.222 8205-8205/? A/DEBUG: pid: 8138, tid: 8201, name: CameraThread  >>> com.wr.medical <<<
2019-11-29 16:36:24.222 8205-8205/? A/DEBUG: signal 7 (SIGBUS), code 2 (BUS_ADRERR), fault addr 0x7199000000
2019-11-29 16:36:24.222 8205-8205/? A/DEBUG:     x0   0000007198fffcc0  x1   000000719a90e170  x2   0000000000000280  x3   000000719d0a63c1
2019-11-29 16:36:24.222 8205-8205/? A/DEBUG:     x4   0000000000000000  x5   0000000000000047  x6   0000007198fffff0  x7   0000000000000000
2019-11-29 16:36:24.222 8205-8205/? A/DEBUG:     x8   0000000000000000  x9   0000000000000000  x10  0000000000000000  x11  0000000000000000
2019-11-29 16:36:24.222 8205-8205/? A/DEBUG:     x12  0000000000000000  x13  0000000000000000  x14  0000000000000000  x15  00004143d8c3802d
2019-11-29 16:36:24.222 8205-8205/? A/DEBUG:     x16  000000719d08ef58  x17  00000071c4bd3908  x18  0000000000ffffeb  x19  00000071c46d9ee0
2019-11-29 16:36:24.222 8205-8205/? A/DEBUG:     x20  000000719c2d1ab0  x21  0000000000000000  x22  000000719a90de00  x23  0000000000000640
2019-11-29 16:36:24.222 8205-8205/? A/DEBUG:     x24  000000719c2d1a00  x25  0000000a00000000  x26  000000000000000a  x27  00000071c46d9f9c
2019-11-29 16:36:24.222 8205-8205/? A/DEBUG:     x28  0000000100000000  x29  00000071a6fe7130  x30  000000719d07ac50
2019-11-29 16:36:24.222 8205-8205/? A/DEBUG:     sp   00000071a6fe70c0  pc   00000071c4bd3a54  pstate 0000000020000000
2019-11-29 16:36:24.227 8205-8205/? A/DEBUG: backtrace:
2019-11-29 16:36:24.227 8205-8205/? A/DEBUG:     #00 pc 000000000001aa54  /system/lib64/libc.so (memcpy+332)
2019-11-29 16:36:24.227 8205-8205/? A/DEBUG:     #01 pc 0000000000013c4c  /data/app/com.wr.medical-2/lib/arm64/libuvc.so
2019-11-29 16:36:24.227 8205-8205/? A/DEBUG:     #02 pc 000000000000b5d4  /data/app/com.wr.medical-2/lib/arm64/libusb100.so (usbi_handle_transfer_completion+336)
2019-11-29 16:36:24.227 8205-8205/? A/DEBUG:     #03 pc 0000000000010884  /data/app/com.wr.medical-2/lib/arm64/libusb100.so
2019-11-29 16:36:24.227 8205-8205/? A/DEBUG:     #04 pc 000000000000c28c  /data/app/com.wr.medical-2/lib/arm64/libusb100.so
2019-11-29 16:36:24.227 8205-8205/? A/DEBUG:     #05 pc 000000000000bcf4  /data/app/com.wr.medical-2/lib/arm64/libusb100.so (libusb_handle_events_timeout_completed+748)
2019-11-29 16:36:24.227 8205-8205/? A/DEBUG:     #06 pc 000000000000c414  /data/app/com.wr.medical-2/lib/arm64/libusb100.so (libusb_handle_events+44)
2019-11-29 16:36:24.227 8205-8205/? A/DEBUG:     #07 pc 0000000000011e24  /data/app/com.wr.medical-2/lib/arm64/libuvc.so (_uvc_handle_events+132)
2019-11-29 16:36:24.228 8205-8205/? A/DEBUG:     #08 pc 0000000000068748  /system/lib64/libc.so (_ZL15__pthread_startPv+208)
2019-11-29 16:36:24.228 8205-8205/? A/DEBUG:     #09 pc 000000000001da7c  /system/lib64/libc.so (__start_thread+16)

ajavafans, 2020-03-11:
解决了没？遇到了同样的问题

along1991, 2020-03-11:
解决了,底层c的问题,可以加我vx1012388357,但不免费帮忙

along1991, 2020-03-11:
> 解决了没？遇到了同样的问题

解决了，充满c的问题，可以加我vx1012388357，但不免费帮忙

fanyang365, 2020-06-30:
> 解决了没？遇到了同样的问题

https://blog.csdn.net/vspiders/article/details/90201818

febflyer, 2021-05-18:
拜谢大佬