saki4510t/UVCCamera #535: Crash on Android 10

tamasberesoebb, 2019-12-12:
I have just updated my S10 to Android 10, and now this library no longer works (it used to work fine on Android 9, on my S10).

```
2019-12-12 14:38:17.276 1475-2635/? D/UsbHostManager: USB device attached: vidpid 046d:0892 mfg/product/ver/serial null/HD Pro Webcam C920/0.19/ABBAEA2F hasAudio/HID/Storage: true/false/false
2019-12-12 14:38:17.278 1475-2635/? D/UsbHostManager: Added device UsbDevice[mName=/dev/bus/usb/001/002,mVendorId=1133,mProductId=2194,mClass=239,mSubclass=2,mProtocol=1,mManufacturerName=null,mProductName=HD Pro Webcam C920,mVersion=0.19,mSerialNumberReader=com.android.server.usb.UsbSerialReader@957154c,mConfigurations=[
    UsbConfiguration[mId=1,mName=null,mAttributes=128,mMaxPower=250,mInterfaces=[
    UsbInterface[mId=0,mAlternateSetting=0,mName=null,mClass=14,mSubclass=1,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=134,mAttributes=3,mMaxPacketSize=64,mInterval=8]]
    UsbInterface[mId=1,mAlternateSetting=0,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[]
    UsbInterface[mId=1,mAlternateSetting=1,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=192,mInterval=1]]
    UsbInterface[mId=1,mAlternateSetting=2,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=384,mInterval=1]]
    UsbInterface[mId=1,mAlternateSetting=3,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=512,mInterval=1]]
    UsbInterface[mId=1,mAlternateSetting=4,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=640,mInterval=1]]
    UsbInterface[mId=1,mAlternateSetting=5,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=800,mInterval=1]]
    UsbInterface[mId=1,mAlternateSetting=6,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=944,mInterval=1]]
    UsbInterface[mId=1,mAlternateSetting=7,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=2688,mInterval=1]]
    UsbInterface[mId=1,mAlternateSetting=8,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=2848,mInterval=1]]
    UsbInterface[mId=1,mAlternateSetting=9,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=3040,mInterval=1]]
    UsbInterface[mId=1,mAlternateSetting=10,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=4992,mInterval=1]]
    UsbInterface[mId=1,mAlternateSetting=11,mName=null,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=5,mMaxPacketSize=5120,mInterval=1]]
    UsbInterface[mId=2,mAlternateSetting=0,mName=null,mClass=1,mSubclass=1,mProtocol=0,mEndpoints=[]
    UsbInterface[mId=3,mAlternateSetting=0,mName=null,mClass=1,mSubclass=2,mProtocol=0,mEndpoints=[]
    UsbInterface[mId=3,mAlternateSetting=1,mName=null,mClass=1,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=131,mAttributes=5,mMaxPacketSize=68,mInterval=4]]
    UsbInterface[mId=3,mAlternateSetting=2,mName=null,mClass=1,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=131,mAttributes=5,mMaxPacketSize=100,mInterval=4]]
    UsbInterface[mId=3,mAlternateSetting=3,mName=null,mClass=1,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=131,mAttributes=5,mMaxPacketSize=132,mInterval=4]]]]
2019-12-12 14:38:17.300 1475-2635/? E/DeviceFilter: invalid number for field exclude
    java.lang.NumberFormatException: For input string: "true"
        at java.lang.Integer.parseInt(Integer.java:615)
        at android.hardware.usb.DeviceFilter.read(DeviceFilter.java:115)
        at com.android.server.usb.UsbProfileGroupSettingsManager.getDeviceFilters(UsbProfileGroupSettingsManager.java:486)
        at com.android.server.usb.UsbProfileGroupSettingsManager.packageMatchesLocked(UsbProfileGroupSettingsManager.java:549)
        at com.android.server.usb.UsbProfileGroupSettingsManager.getDeviceMatchesLocked(UsbProfileGroupSettingsManager.java:717)
        at com.android.server.usb.UsbProfileGroupSettingsManager.resolveActivity(UsbProfileGroupSettingsManager.java:753)
        at com.android.server.usb.UsbProfileGroupSettingsManager.deviceAttached(UsbProfileGroupSettingsManager.java:746)
        at com.android.server.usb.UsbHostManager.usbDeviceAdded(UsbHostManager.java:408)
        at com.android.server.usb.UsbHostManager.monitorUsbHostBus(Native Method)
        at com.android.server.usb.UsbHostManager.lambda$XT3F5aQci4H6VWSBYBQQNSzpnvs(Unknown Source:0)
        at com.android.server.usb.-$$Lambda$UsbHostManager$XT3F5aQci4H6VWSBYBQQNSzpnvs.run(Unknown Source:2)
        at java.lang.Thread.run(Thread.java:919)
2019-12-12 14:38:17.314 1475-2635/? I/chatty: uid=1000(system) UsbService host identical 118 lines
2019-12-12 14:38:17.314 1475-2635/? E/DeviceFilter: invalid number for field exclude
    java.lang.NumberFormatException: For input string: "true"
        at java.lang.Integer.parseInt(Integer.java:615)
        at android.hardware.usb.DeviceFilter.read(DeviceFilter.java:115)
        at com.android.server.usb.UsbProfileGroupSettingsManager.getDeviceFilters(UsbProfileGroupSettingsManager.java:486)
        at com.android.server.usb.UsbProfileGroupSettingsManager.packageMatchesLocked(UsbProfileGroupSettingsManager.java:549)
        at com.android.server.usb.UsbProfileGroupSettingsManager.getDeviceMatchesLocked(UsbProfileGroupSettingsManager.java:717)
        at com.android.server.usb.UsbProfileGroupSettingsManager.resolveActivity(UsbProfileGroupSettingsManager.java:753)
        at com.android.server.usb.UsbProfileGroupSettingsManager.deviceAttached(UsbProfileGroupSettingsManager.java:746)
        at com.android.server.usb.UsbHostManager.usbDeviceAdded(UsbHostManager.java:408)
        at com.android.server.usb.UsbHostManager.monitorUsbHostBus(Native Method)
        at com.android.server.usb.UsbHostManager.lambda$XT3F5aQci4H6VWSBYBQQNSzpnvs(Unknown Source:0)
        at com.android.server.usb.-$$Lambda$UsbHostManager$XT3F5aQci4H6VWSBYBQQNSzpnvs.run(Unknown Source:2)
        at java.lang.Thread.run(Thread.java:919)
```

kriass, 2019-12-20:
Maybe you can taking a try to build your software on API 27.

hendrep, 2020-01-15:
I can confirm that I am also having issues with the Samsung S10 on Android 10. @tamasberesoebb  were you able to find a solution?

tamasberesoebb, 2020-01-15:
With this app from the play store, the camera works. They seem to be doing things differently.
https://play.google.com/store/apps/details?id=infinitegra.app.usbcamera&hl=en

hendrep, 2020-01-16:
Setting target sdk to 27 seems to do the trick - could not find a way to get it to work on 28

hendrep, 2020-01-16:
Unfortunately Play store does not allow you to upload an app with target sdk below 28. 

This is an open issue: https://issuetracker.google.com/issues/145082934

imad-123, 2020-01-16:
on devices_filter xml file do this :
<resources>

    <usb-device
        />

</resources>

falll2000, 2020-02-26:
I build my uvc app in build version 27. But not luck, I got this crash.
Any other solution?

tianshanaoxue, 2020-04-07:
原因如下，还未找到解决办法：
 系统类UsbUserSettingsManager 中的如下方法中的
mUserContext.checkCallingPermission(android.Manifest.permission.CAMERA);
永远返回-1，还未找到解决办法

 private boolean isCameraPermissionGranted(String packageName, int uid) {
        int targetSdkVersion = android.os.Build.VERSION_CODES.P;
        try {
            ApplicationInfo aInfo = mPackageManager.getApplicationInfo(packageName, 0);
            // compare uid with packageName to foil apps pretending to be someone else
            if (aInfo.uid != uid) {
                Slog.i(TAG, "Package " + packageName + " does not match caller's uid " + uid);
                return false;
            }
            targetSdkVersion = aInfo.targetSdkVersion;
        } catch (PackageManager.NameNotFoundException e) {
            Slog.i(TAG, "Package not found, likely due to invalid package name!");
            return false;
        }

        if (targetSdkVersion >= android.os.Build.VERSION_CODES.P) {
    // 升级到android10 之后，checkCallingPermission 永远返回-1，即PackageManager.PERMISSION_DENIED，即使已经有了CAMERA权限。
            int allowed = mUserContext.checkCallingPermission(android.Manifest.permission.CAMERA);
            if (android.content.pm.PackageManager.PERMISSION_DENIED == allowed) {
                Slog.i(TAG, "Camera permission required for USB video class devices");
                return false;
            }
        }

        return true;
    }

41108871, 2020-06-01:
@tianshanaoxue 
Coulde you can preview the usb camera on your Android 10？

designerMichael, 2020-08-17:
> 原因如下，还未找到解决办法：
> 系统类UsbUserSettingsManager 中的如下方法中的
> mUserContext.checkCallingPermission(android.Manifest.permission.CAMERA);
> 永远返回-1，还未找到解决办法
>
> private boolean isCameraPermissionGranted(String [...]

请问找到解决方法了吗？遇到同样的问题

41108871, 2020-08-17:
if (targetSdkVersion &gt;= android.os.Build.VERSION_CODES.P) {
            int allowed = mUserContext.checkCallingPermission(android.Manifest.permission.CAMERA);
            //if (android.content.pm.PackageManager.PERMISSION_DENIED == allowed) {
            //    Slog.i(TAG, "Camera permission required for USB video class devices");
            //    return false;
            //}
        }

把这段代码直接屏蔽掉

Send time: Monday, Aug 17, 2020 3:05 PM
To: "saki4510t/UVCCamera"<UVCCamera@noreply.github.com&gt;; 
Cc: "深圳市壹品房产有限公司"<41108871@qq.com&gt;; "Comment"<comment@noreply.github.com&gt;; 

  

 
     if (targetSdkVersion &gt;= android.os.Build.VERSION_CODES.P) { // 升级到android10 之后，checkCallingPermission 永远返回-1，即PackageManager.PERMISSION_DENIED，即使已经有了CAMERA权限。         int allowed = mUserContext.checkCallingPermission(android.Manifest.permission.CAMERA);         if (android.content.pm.PackageManager.PERMISSION_DENIED == allowed) {             Slog.i(TAG, "Camera permission required for USB video class devices");             return false;         }     }     return true; }   
请问找到解决方法了吗？遇到同样的问题

designerMichael, 2020-08-17:
> if (targetSdkVersion &gt;= android.os.Build.VERSION_CODES.P) {             int allowed = mUserContext.checkCallingPermission(android.Manifest.permission.CAMERA);             //if (android.content.pm.PackageManager.PERMISSION_DENIED == allowed) {             //    Slog.i(TAG, "Camera permission required for USB video class devices");             //    return false;             //}         } 把这段代码直接屏蔽掉
> […](#)

请问不是定制系统，第三方品牌手机，如何屏蔽吖？

rajan7685, 2022-05-03:
In Android version 12 the preview is not showing. Can you please help me on this