saki4510t/UVCCamera #538: When I set the high resolution, I Can't preview,

GQii, 2020-01-03:
Hi saki,
    When I set the high resolution,When I set the high resolution,The screen cannot be previewed。
but when i set the low resolution , There's a screen on the screen.
    I don't know how to solve this problem, I hope you can help me, thank you。

GQii, 2020-01-08:
[image]
[image]

@saki4510t ,I don't know how to solve this problem.Can you help me, thank you very much?

jangho0311, 2020-02-14:
Hi, GQii
I know what kind of problem you are talking about.
but 640*480 is not high resolution.
because i use 1920*1080 resolution.
maybe don't use 4K.
you should check external camera spec and modify default (resolution, frame rate).
The default resolution are modified jni and java files.
Thank you.

GQii, 2020-02-23:
@Jade-source First of all, thank you for your reply.
I tried to modify resolution and frame rate in the past, but the effect is still as shown above.
I'm not a solution to this problem right now.

Jade-source, 2020-02-26:
@GQii 
you should use UVCCameraHandler.createHandler(Activity, CameraViewInterface, encoderType, 1920, 1080, format) when preview is FHD. 

I think the external camera does not compress video format well.

GQii, 2020-02-28:
@Jade-source  Thank you very much, I've solved this problem.