saki4510t/UVCCamera #547: 从昨天下午开始我的as怎么没法加载common 2.14.4的aar了

PingWei1314, 2020-02-26:
Unable to resolve dependency for ':app@debug/compileClasspath': Could not download common.aar (com.serenegiant:common:2.12.4)
Show Details
Affected Modules: app

ERROR: Unable to resolve dependency for ':app@debugAndroidTest/compileClasspath': Could not download common.aar (com.serenegiant:common:2.12.4)
Show Details
Affected Modules: app

ERROR: Unable to resolve dependency for ':app@debugUnitTest/compileClasspath': Could not download common.aar (com.serenegiant:common:2.12.4)
Show Details
Affected Modules: app

ERROR: Unable to resolve dependency for ':app@release/compileClasspath': Could not download common.aar (com.serenegiant:common:2.12.4)
Show Details
Affected Modules: app

ERROR: Unable to resolve dependency for ':app@releaseUnitTest/compileClasspath': Could not download common.aar (com.serenegiant:common:2.12.4)
Show Details
Affected Modules: app

ERROR: Unable to resolve dependency for ':libusbcamera@debug/compileClasspath': Could not download common.aar (com.serenegiant:common:2.12.4)
Show Details
Affected Modules: libusbcamera

ERROR: Unable to resolve dependency for ':libusbcamera@debugAndroidTest/compileClasspath': Could not download common.aar (com.serenegiant:common:2.12.4)
Show Details
Affected Modules: libusbcamera

ERROR: Unable to resolve dependency for ':libusbcamera@debugUnitTest/compileClasspath': Could not download common.aar (com.serenegiant:common:2.12.4)
Show Details
Affected Modules: libusbcamera

ERROR: Unable to resolve dependency for ':libusbcamera@release/compileClasspath': Could not download common.aar (com.serenegiant:common:2.12.4)
Show Details
Affected Modules: libusbcamera

ERROR: Unable to resolve dependency for ':libusbcamera@releaseUnitTest/compileClasspath': Could not download common.aar (com.serenegiant:common:2.12.4)
Show Details
Affected Modules: libusbcamera

vip553944427, 2020-03-09:
maven { url 'https://raw.githubusercontent.com/saki4510t/libcommon/master/repository/' }

SoftwareME, 2020-04-22:
maven { url 'https://raw.githubusercontent.com/saki4510t/libcommon/master/repository/' }
无法访问 需要翻墙么

Clark-caipeiyuan, 2020-04-27:
Finally,how did you solve this problem?

hushengjun, 2020-05-25:
> Finally,how did you solve this problem?

You can download from the author's common library under github

vip553944427, 2020-05-26:
> > Finally,how did you solve this problem?
> 
> You can download from the author's common library under github

maven { url 'https://raw.githubusercontent.com/saki4510t/libcommon/master/repository/' }
就是添加这个地址挂代理就能加载了

WestbrookZero, 2020-09-09:
@vip553944427 请问要挂什么代理？

Clark-caipeiyuan, 2020-11-18:
> > Finally,how did you solve this problem?
> 
> You can download from the author's common library under github

Thank you so much.I have solved this problem.