saki4510t/UVCCamera #553: How to get audio data from USB

huyubao, 2020-04-03:


imprezakd, 2020-07-14:
Can you tell me , how to get audio data from usb camera?

huyubao, 2020-07-16:
You can use UAC

imprezakd, 2020-07-16:
Thanks , I need to play the sound while previewing extremecap UVC , do you have any example of UAC ?

huyubao, 2020-07-16:
I will debug again, so there are no examples

imprezakd, 2020-07-16:
I am testing usbcameratest0 , preview is ok but no sound , Hope you can help , thanks

imprezakd, 2020-10-16:
@butterfly0Dream  No，Could you help me ?

jinlongxu2018, 2021-06-29:
找到对应的音频端点即可录音

imprezakd, 2021-06-29:
@jinlongxu2018 錄音沒有問題，可以取得音頻，但我希望是預覽時並同時撥放音頻，像是EasyCap application

jinlongxu2018, 2021-06-29:
> @jinlongxu2018 錄音沒有問題，可以取得音頻，但我希望是預覽時並同時撥放音頻，像是EasyCap application
我是得到了录音的callback之后,创建了一个AudioTrack.然后callback和AudioTrack之间通过Ringbuffer交换数据,实现了音频的loopback的

imprezakd, 2021-06-29:
@jinlongxu2018 您好，有完整範例嗎? 如果有的話就太感謝了~