saki4510t/UVCCamera #556: Problem taking photos with multiple cameras

lgstud-io, 2020-04-24:
I have an issue to take pictures simultaneously with multiple cameras.
Most of the cases (99% of time) only a single picture is taken

I have a class that wraps every connected camera to the device. 
The cameraNumber is a unique number of the camera (1 to n)
I call this method for every single camera wrapper object

It does not matter if I call them at once, all one by one waiting for the previous camera to finish and call onPhotoSaved in listener, Only a single photo is taken by camera 1. And the result of the .stillCapture callback returns the path to the image "..._1.jpg" for every other still capture call also.

So:
I call takePicture for camera No1.
It retuns me picPath "...._1.jpg"
Then I call takePicture for camera No2.
It retuns me picPath" ..._1.jpg" also. and the picture named "...._2.jpg" is never taken.

What might be the problem?

my photo taking code is the following:

`void takePicture(String folderPath, String photoTime) {
       if (cameraHandler != null && cameraHandler.isOpened()) {
                String newImagePath = folderPath + "/" + photoTime+ "_" + cameraNumber + ".jpg";
                cameraHandler.captureStill(newImagePath, picPath -> {
                        listener.onPhotoSaved(new CapturedPhoto(cameraNumber, picPath));
                    }
                });
            }
            else {
                listener.onPhotoSaved(new CapturedPhoto(cameraNumber, null));
            }
        }
    }`

quocnhat7, 2020-06-03:
Did you increase "cameraNumber" after capture picture?

heaveninfosoft, 2020-07-09:
@lgstud-io  can you send me code for multiple camera taking picture as well as recording video??