saki4510t/UVCCamera #557: Could not download common.aar (com.serenegiant:common:2.12.4)

Clark-caipeiyuan, 2020-04-27:
ERROR: Unable to resolve dependency for ':app@debug/compileClasspath': Could not download common.aar (com.serenegiant:common:2.12.4)
Show Details
Affected Modules: app

ERROR: Unable to resolve dependency for ':app@debugAndroidTest/compileClasspath': Could not download common.aar (com.serenegiant:common:2.12.4)
Show Details
Affected Modules: app

ERROR: Unable to resolve dependency for ':app@debugUnitTest/compileClasspath': Could not download common.aar (com.serenegiant:common:2.12.4)
Show Details
Affected Modules: app

ERROR: Unable to resolve dependency for ':app@release/compileClasspath': Could not download common.aar (com.serenegiant:common:2.12.4)
Show Details
Affected Modules: app

ERROR: Unable to resolve dependency for ':app@releaseUnitTest/compileClasspath': Could not download common.aar (com.serenegiant:common:2.12.4)
Show Details
Affected Modules: app

I cannot sync or build my peoject from yesterday。
Then I replaced he maven{url 'http://raw.github.com/saki4510t/libcommon/master/repository/'} 
by maven { url 'https://raw.githubusercontent.com/saki4510t/libcommon/master/repository/' }
 ，but it cannot work.

wnpllrzodiac, 2020-05-07:
the same problem

wangruoyudev, 2020-05-14:
> ERROR: Unable to resolve dependency for ':app@debug/compileClasspath': Could not download common.aar (com.serenegiant:common:2.12.4)
> Show Details
> Affected Modules: app
>
> ERROR: Unable to resolve [...]

maven {
            url 'https://gitee.com/liuchaoya/libcommon/raw/master/repository/'
        }

this can help you resolve the problem.