saki4510t/UVCCamera #560: How to record video without preview?

MatthewDevelop, 2020-06-10:
Is it possible to record video without preview?

hushengjun, 2020-06-10:
It should be fine, you don’t need to call setPreviewDisplay(Surface surface)

MatthewDevelop, 2020-06-11:
有联系方式吗？可以联系大佬请教下吗？

shengjunhu, 2020-07-10:
may be u need to modify UVCPreview.cpp startPreview(); do_preview()

shengjunhu, 2020-07-12:
> 有联系方式吗？可以联系大佬请教下吗？

https://github.com/shengjunhu/android_sample_uvccamera/blob/master/sdk_uvc_camera/src/main/jni/UVCCamera/UVCPreview.cpp
Search：  //Add by shengjunhu

shengjunhu, 2020-07-12:
> Hello, I would like to run the frame callback without preview, but I am not able to make it work. Do you know how to do so?

https://github.com/shengjunhu/android_sample_uvccamera/blob/master/sdk_uvc_camera/src/main/jni/UVCCamera/UVCPreview.cpp
Search： //Add by shengjunhu

EarvinChiang, 2022-08-18:
> > Hello, I would like to run the frame callback without preview, but I am not able to make it work. Do you know how to do so?
> 
> https://github.com/shengjunhu/android_sample_uvccamera/blob/master/sdk_uvc_camera/src/main/jni/UVCCamera/UVCPreview.cpp Search： //Add by shengjunhu

This link no longer exists, can you provide the new link again, thank you.

shengjunhu, 2022-09-14:
> This link no longer exists, can you provide the new link again, thank you.

https://github.com/shengjunhu/android_uvc_camera/blob/master/sdk_uvc_camera/src/main/jni/libcamera/source/UVCPreview.cpp

oleszczyk, 2024-11-14:
Hello @shengjunhu , grabbing frames (RGB565 in my case) into a callback without any need of preview is working perfectly for me, but I would like to refer for original question from this post - is it possible to record a video from those frames (still without preview)?

Or maybe do you know any external library (javacv or ffmpeg) which would allow me to record a video from received frames?