saki4510t/UVCCamera #565: Resolution issue?

rgm-22, 2020-07-16:
**Great work! Thank you!**

I would like to modify the libraries because I am having this excepition:

I/AbsUVCCameraHandler: supportedSize:{"formats":[{"index":1,"type":4,"default":1,"size":["160x120"]},{"index":2,"type":4,"default":1,"size":["160x120","160x122"]},{"index":3,"type":4,"default":1,"size":["160x120"]},{"index":4,"type":4,"default":1,"size":["160x120"]},{"index":5,"type":4,"default":1,"size":["160x120"]}]}
V/zhf_debug: handleSetPreviewCallback: frameCallback=com.example.o0orick.camera.MainActivity$5@d9370d9
I/libUVCCamera: [14743*UVCPreview.cpp:266:callbackPixelFormatChanged]:PIXEL_FORMAT_RGB565:
V/CameraThread: handleStartPreview:
W/System.err: java.lang.IllegalArgumentException: Failed to set preview size
W/System.err: at com.serenegiant.usb.UVCCamera.setPreviewSize(UVCCamera.java:328)

With these values:

**private static final int PREVIEW_WIDTH = 160;
private static final int PREVIEW_HEIGHT = 120;
private static final int PREVIEW_MODE = 0; // YUV**

My camera has the following features:

160 x 120
9 fps
UYVY format
Any suggestion to solve it or build the libraries?

Thank you very much in advance

closed 2020-10-23 by rgm-22

feng612266, 2024-05-23:
> **Great work! Thank you!**
> 
> I would like to modify the libraries because I am having this excepition:
> 
> I/AbsUVCCameraHandler: supportedSize:{"formats":[{"index":1,"type":4,"default":1,"size":["160x120"]},{"index":2,"type":4,"default":1,"size":["160x120","160x122"]},{"index":3,"type":4,"default":1,"size":["160x120"]},{"index":4,"type":4,"default":1,"size":["160x120"]},{"index":5,"type":4,"default":1,"size":["160x120"]}]} V/zhf_debug: handleSetPreviewCallback: frameCallback=com.example.o0orick.camera.MainActivity$5@d9370d9 I/libUVCCamera: [14743*UVCPreview.cpp:266:callbackPixelFormatChanged]:PIXEL_FORMAT_RGB565: V/CameraThread: handleStartPreview: W/System.err: java.lang.IllegalArgumentException: Failed to set preview size W/System.err: at com.serenegiant.usb.UVCCamera.setPreviewSize(UVCCamera.java:328)
> 
> With these values:
> 
> **private static final int PREVIEW_WIDTH = 160; private static final int PREVIEW_HEIGHT = 120; private static final int PREVIEW_MODE = 0; // YUV**
> 
> Any suggestion to solve it?
> 
> Thank you very much in advance

hi,how resolve this question

rgm-22, 2024-05-23:
Hi @feng612266,

What camera (specs) are you using?

It depends on your resolution and your video format.