saki4510t/UVCCamera #569: Access multiple USB Cameras at same time

vidhyagithub, 2020-08-18:
Can the library access frames two USB Cameras concurrently?

MauriDalga, 2020-09-10:
Yes u can, i do that.

EjazAmir, 2020-10-28:
hey, can you share your project code where you use mulitple usb cameras? I'm really stuck in this

MauriDalga, 2020-10-28:
I can't share the project, sorry because it belongs to the company where I work. But really, i only did something very similar to 'usbCameraTest7'. Do you already see that example?

EjazAmir, 2020-10-29:
> I only did something very similar to 'usbCameraTest7'. Do you already see that example

Yeah, i tried but unable to compile the application. The error says, "No such class found". 
WIll you kindly share the piece of code that actually connects the mulitple cameras and captures images, not asking for whole project bcx i understand the confidentiality matters.
Regards

EjazAmir, 2020-11-02:
> I only did something very similar to 'usbCameraTest7'. Do you already see that example

hey, i tried 'usbCameraTest7' but unable to capture images from multiple cameras, can you kindly help me that